ev3dev/ev3dev #615: how to get TL-WN725N wifi dongle working?

kortschak, 2016-04-26:
I have look through the issue and found what comes close to a recommendation for the TP-Link TL-WN725N wifi dongle, and so got one.

I have not been able to get it to work out of the box. Is there a set of instructions (other than the dkms instruction on the X page - it looks from some of the issues that it is built into recent kernels, though I may be reading wrong - building on an ev3 seems like a pathway to pain).

I'm using the 3.16.7-ckt26-10-ev3dev-ev3 kernel and lsusb gives the following:

```
Bus 001 Device 002: ID 0bda:8179 Realtek Semiconductor Corp. 
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
```

In brickman the Wi-Fi screen just says 'Not available' and the command line approach leads me to this:

```
$ sudo connmanctl
connmanctl> enable wifi
Error wifi: Method "SetProperty" with signature "sv" on interface "net.connman.Technology" doesn't exist
```

which doesn't look promising.

kortschak, 2016-04-26:
Found solution (needed the module package installed). In this case (found by `apt-cache search 8188eu`):

```
apt-get install rtl8188eu-modules-3.16.7-ckt26-10-ev3dev-ev3
modprobe 8188eu
```

The rest follows.

closed 2016-04-26 by kortschak

dlech, 2016-04-26:
This package should be automatically installed by the `linux-image-ev3dev-ev3` meta package. If it was not, there is a bug.

kortschak, 2016-04-26:
That makes sense. When I installed the kernel I used the kernel package directly instead of thr meta package (see [here](https://github.com/ev3dev/ev3dev/issues/610#issuecomment-211140771) and following comments)) and did not install the meta package afterwards. So no bug in the package. I'll install it to keep up to date.

Thanks.