ev3dev/ev3dev #618: ev3dev-lang-python Motors no longer working

psychemedia, 2016-04-29:
### System information:

* I am using: LEGO MINDSTORMS EV3

* I installed ev3dev using this image file: ev3dev-jessie-2015-12-30

* My kernel version is: 3.16.7-ckt26-10-ev3dev-ev3

* My host computer is running: OS/X

* I am connecting using: Wi-Fi

### About my issue:

#### What led up to the situation?
I updated all packages, tried running demos from rhempel/ev3dev-lang-python and motor demos did not work. Speaker and LED/touch demos work correctly.

#### What exactly did you do (or not do) that was effective (or ineffective)?
Tried to get motors to work; through EV3 screen, I can see motors are there labeled as expected.

#### What was the outcome of this action?
Motors don't respond

#### What outcome did you expect instead?
Motors respond

dlech, 2016-04-29:
This is expected. Please read the announcement for kernel release 10.

http://www.ev3dev.org/news/2016/04/11/Kernel-Release-Cycle-10/

psychemedia, 2016-04-29:
Ah - ok - thanks... tells me there are breaking changes but doesn't actually help me get the motors working...

I also just tried updating ev3dev-lang-python from github to same effect - so breaking changes not reflected there. What setting do I need where to make the things work?!

dlech, 2016-04-29:
Try https://github.com/rhempel/ev3dev-lang-python/commits/feature-new-tacho-driver-attributes. If you have questions about ev3dev-lang-python, it has it's own issue tracker.

Or you can [downgrade your kernel](https://github.com/ev3dev/ev3dev/issues/609).

mob41, 2016-04-29:
I think that the kernel can be released for development use first next time. 

And tell the developers to update their own library before a due date.

If the library is bundled with the OS together (like ev3dev-lang-python), the kernel should not be released as normal release.

Otherwise, the library will be broken and the users will blame about it.

rhempel, 2016-04-29:
I have been swamped in the last month with some issues outside of ev3dev - I do have a new version of the binding with much faster access to device attributes working, now I need to update my `spec.json` and regenerate the bindings to follow the latest kernel changes before releasing 0.7.0 of the Python binding.

Sorry for the slow progress on this end.

dlech, 2016-05-08:
I'm going to call this a duplicate of #608.

closed 2016-05-08 by dlech