ev3dev/ev3dev #619: Having issue to connect Pycharm to the Brick

Snooone, 2016-04-29:
### System information:

* I am using: LEGO MINDSTORMS EV3

* I installed ev3dev using this image file: ev3-ev3dev-jessie-2015-12-30.img

* My kernel version is: 3.16.7-ckt26-10-ev3dev-ev3

* My host computer is running: Windows 10 Home edition

* I am connecting using: Wifi

### About my issue:

#### What led up to the situation?
Trying to connect the Lego Brick with Pycharm I am getting the following error
     ``` Can't login: Remote host closed connnection during hand shake ```
I am using following details:
```Host: robot@192.168.0.47/myproject.git
login: robot
password: maker```

#### What exactly did you do (or not do) that was effective (or ineffective)?
I followed the pycharm Tuto: http://www.ev3dev.org/docs/tutorials/setting-up-python-pycharm/
then I followed the Get help tuto:  http://www.ev3dev.org/support/

#### What was the outcome of this action?
error before kernel upgrade and after upgrade is the same. it seems to me (but I am not expert) that there is a certificate issue, if this is the case i don't know how to fix it...)

#### What outcome did you expect instead?
Pycharm connection to the brick.

Thanks in advance for your help

dlech, 2016-04-29:
> robot@192.168.0.47/myproject.git

This should have a colon, not a slash

    robot@192.168.0.47:myproject.git

Snooone, 2016-04-29:
Hi dlech,

thanks for your answer, I have tried with a colon with slash, with .git, without .git.... 

I desactivated the firewall...and got always the same  answer:

[image]

dlech, 2016-04-29:
Can you connect to the EV3 using PuTTY? Can you browse to `/home/robot/myproject.git` using Filezilla?

Snooone, 2016-04-29:
Yep I can connect through Putty, and use command shell as well.

I cannot connect using filezilla anonymous account . it seems ftp is not launched. is it the issue?

/home/robot/myproject.git contains:
```
robot@ev3dev:~/myproject.git/hooks$ ls /home/robot/myproject.git -ln
total 32
-rw-r--r-- 1 1000 1000   23 Apr 28 20:33 HEAD
drwxr-xr-x 2 1000 1000 4096 Apr 28 20:33 branches
-rw-r--r-- 1 1000 1000   66 Apr 29 17:28 config
-rw-r--r-- 1 1000 1000   73 Apr 28 20:33 description
drwxr-xr-x 2 1000 1000 4096 Apr 28 20:37 hooks
drwxr-xr-x 2 1000 1000 4096 Apr 28 20:33 info
drwxr-xr-x 4 1000 1000 4096 Apr 28 20:33 objects
drwxr-xr-x 4 1000 1000 4096 Apr 28 20:33 refs
```

dlech, 2016-04-29:
Use sftp in filezilla (port 22, same as SSH)

Also, I noticed you included the password in `robot:maker@192.168.0.47:myproject.git`, which does not match the tutorial. Also, confirm that `192.168.0.47` is the IP address on your EV3.

Basically, `robot:maker@192.168.0.47` should be replaced with whatever you are using in PuTTY to connect.

Snooone, 2016-04-29:
yes the EV3 is ```192.168.0.47```. And it works when I connect with Filezilla Sftp on the EV3 IP with robot/maker login. I am able to see in filezilla the home repository of ```robot``` login.

PyCharm still does not connect.

dlech, 2016-04-29:
`robot@ev3dev:myproject.git` works for me. I guess you will just have to google the error message and see what you can come up with. I am afraid the problem is with your system setup somehow. I'm out of ideas.

Snooone, 2016-04-29:
Thanks for your help David!
What is your system version and what about ev3dev, is it a name declared in Hosts file?

dlech, 2016-04-29:
> What is your system version

Same as what you posted in the original comment with all packages updated.

> and what about ev3dev, is it a name declared in Hosts file?

ev3dev runs `nmbd` by default for Windows name resolution

Snooone, 2016-04-29:
what is your your windows system ad PyCharm version?

I am struggling with google and just not seeing what is going on....

Nevertheless thx.

WasabiFan, 2016-04-29:
Can you try disabling WiFi and connecting over USB tethering instead? That should eliminate the possibility of firewalls or other devices getting in the way.

dlech, 2016-04-29:
> what is your your windows system ad PyCharm version?

Windows 10 Pro as Virtualbox guest on Linux host. Pycharm Version: 2016.1.2 Build: 145.844 Released: April 7, 2016

Snooone, 2016-05-01:
Hi Dlech and Was,

I tried already over USB connection, and I got the same issue.

I configured from scratch a new Win 10 PC, without AVAST, and I have the same issue ``` Repository test as failed``` ; as well Filezilla and PuTTY are still working from this PC.

Do someone know how to get the exact error related to ``` Repository test as failed``` ?

Is there some log on PC or on brick to help determining, if it s a Network issue, an authentification issue, a Brick git issue..... 

Is there a way to activate some process log connection on the brick ? (sorry for this question, I did not practice Linux for more than 20 years, and it seems to me I had used command-tracking on shells but I do not remember how....)

dlech, 2016-05-01:
You could try running `sudo systemctl status sshd.service` on the EV3 to see the SSH server log.

I would ask PyCharm support about the `Repository test has failed` error.

Does PyCharm work with a remote repository somewhere else, such as a github repository? Perhaps the git repository is not setup correctly on your EV3 (although from your earlier comment, it does look setup correctly to me).

dlech, 2017-10-24:
FYI, Visual Studio Code will be the "official" ev3dev IDE in ev3dev-stretch. It should be easier to setup.

closed 2017-10-24 by dlech