ev3dev/ev3dev #621: run-direct in Kernel Release Cycle 10

laurensvalk, 2016-05-01:
If I understand correctly from the release notes, the following should still work on 3.16.7-ckt26-10-ev3dev-ev3:

>  If you need unregulated control of motors (speed_regulation = off) use the run-direct command

```
echo run-direct > command
echo 30 > duty_cycle_sp
```

This does not work (the motors do not start moving). Are any additional steps required to start the motor?

mob41, 2016-05-01:
What motor are you using? Normal EV3 motors like ```legoev3-large-motor``` etc.?

Have you tried to reset your motor properties before trying this? By sending a command to ```command```:
```echo reset > command```

Have you also tried to set ```duty_cycle_sp``` before sending ```run-direct``` to ```command```?

It works on the old kernel ```3.16.7-ckt21-9-ev3dev-ev3```. I am now install kernel 10 to test it out.

dlech, 2016-05-01:
No, this should work, but it is not working for me either with the official packaged kernel.

dlech, 2016-05-01:
Haven't tested it, but pretty sure I found/fixed the bug already. I guess kernel cycle 11 will be coming sooner than expected since this is a critical bug.

mob41, 2016-05-02:
I just tested it, it doesn't work. I just did these steps:
```cd /sys/class/tacho-motor/motor0```
```echo 50 > duty_cycle_sp```
```echo run-direct > command``` (Not running)
```echo 50 > duty_cycle_sp``` (Still not running)
```cat state``` (Show "running stalled")
```echo reset > command```

rhempel, 2016-05-02:
Thks @dlech - it's about time I got the kernel updated on my EV3 development brick so I can fininsh the Python binding anyways :-)

dlech, 2016-05-08:
Fix has been released in [kernel v4.4.9-11-ev3dev-ev3](http://www.ev3dev.org/news/2016/05/08/Kernel-Release-Cycle-11/).

closed 2016-05-08 by dlech