ev3dev/ev3dev #622: loading lm75 does not allow LEGO 9749 temperature sensor to auto-detect

kortschak, 2016-05-03:
### System information:

* I am using: LEGO MINDSTORMS EV3

* I installed ev3dev using this image file: ev3-ev3dev-jessie-2015-12-30.img

* My kernel version is: 3.16.7-ckt26-10-ev3dev-ev3

* My host computer is running: Linux

* I am connecting using: Wi-Fi

### About my issue:

#### What led up to the situation?
Connect LEGO 9749 and `modprobe lm75`

#### What exactly did you do (or not do) that was effective (or ineffective)?
Attempt to insert module to allow auto-detection of the temperature sensor as described [here](http://www.ev3dev.org/docs/sensors/#fn:lm75). Also try `modprobe lm75` and then attach sensor.

#### What was the outcome of this action?
lsmod show lm75 to be loaded, but nothing in the way of sensor auto-detection.

#### What outcome did you expect instead?
An temperature sensor device in brickman device browser or a sensor directory in `/sys/class/lego-sensor`.

dlech, 2016-05-03:
The I2C adapter drivers on the EV3 are set to [I2C_CLASS_LEGOEV3](https://github.com/ev3dev/lego-linux-drivers/blob/02f3d89bb1fb348c5adde854f9da3c08f83772e5/ev3/legoev3_i2c.c#L101).

We could add a new `i2c-hwmon` mode to the input ports that sets the class to `I2C_CLASS_HWMON` instead, then it would try to detect `hwmon` class devices instead of `lego-sensor` class devices.

For now, you will have to manually set the driver using the `set_device` attribute of the port to get it to work.

kortschak, 2016-05-03:
At the moment the docs say it should be auto-detected. Maybe this could be clarified.

dlech, 2016-05-03:
https://github.com/ev3dev/ev3dev.github.io/commit/af2d65dca7e8944c0bf0acc5138924601166a8bf

kortschak, 2016-05-04:
Setting the device does not appear to work.

With the temperature sensor attached to port in4:
```
robot@ev3dev:/sys/class/lego-port/port3$ sudo modprobe lm75
robot@ev3dev:/sys/class/lego-port/port3$ lsmod | grep lm75
lm75                    4007  0 
hwmon                   2268  1 lm75
robot@ev3dev:/sys/class/lego-port/port3$ ls
address  device  driver_name  i2c-legoev3.6  in4:nxt-i2c-host  mode  modes  power  set_device  status  subsystem  uevent
robot@ev3dev:/sys/class/lego-port/port3$ echo tmp275 >set_device 
-bash: echo: write error: Operation not supported
```

Is there something I'm missing? It seems that this device requires the [Other/I2C sensor approach](http://www.ev3dev.org/docs/sensors/using-i2c-sensors/#using-otheri2c-sensors).

kortschak, 2016-05-04:
No, that appears not to work either (again device in port in4 lm75 still loaded):

```
robot@ev3dev:~$ sudo echo tmp275 0x4C >/sys/bus/i2c/devices/i2c-6/new_device 
-bash: /sys/bus/i2c/devices/i2c-6/new_device: Permission denied
```

The contents of `/sys/bus/i2c/devices` and the last two lines of dmesg show there is a device there:

```
robot@ev3dev:~$ ls /sys/bus/i2c/devices/                 
1-0050  i2c-1  i2c-6
```
```
[ 4765.905475] i2c-legoev3 i2c-legoev3.6: registered on input port 4
[ 4765.921950] lego-port port3: Added new device 'in4:nxt-i2c-host'
```

dlech, 2016-05-04:
> sudo echo tmp275 0x4C >/sys/bus/i2c/devices/i2c-6/new_device

This is wrong. It runs `echo` with sudo, but `>` redirection is part of bash, which is running as a normal user, hence the permission issue. Here are some alternatives:

    sudo sh -c 'echo tmp275 0x4C >/sys/bus/i2c/devices/i2c-6/new_device'
    echo tmp275 0x4C | sudo tee /sys/bus/i2c/devices/i2c-6/new_device

> The contents of /sys/bus/i2c/devices and the last two lines of dmesg show there is a device there:

This is the I2C adapter. The actual hwmon device will look like `6-004c`

kortschak, 2016-05-04:
Thanks that has got it working. This bites me more often that I care to admit.

>This is the I2C adapter. The actual hwmon device will look like `6-004c`

Thanks for the clarification. I was really meaning that it is being seen as physically present by the ev3 - always good to let people know the device is plugged in when reporting an issue :)

If I were to express an interest in getting this to work in the `lego-sensor` class how foolish would I be being? and if not too foolish, where would I start looking to get it working there? (It seems sad that a LEGO device that is a common part of Mindstorms-based teaching is not something that works as easily as it does with the Mindstorms software where it just works).

dlech, 2016-05-04:
This is an unfortunate case where an official LEGO product doesn't follow the quasi-official LEGO specification for I2C sensors. I think what I would like to see is a `lego-sensor` wrapper around the `hwmon` subsystem. If we don't turn on too many hwmon drivers, I think it would be OK to scan for both `lego-sensor` devices and `hwmon` devices when we plug in a sensor.

On the other hand, you are the only person that has asked about the LEGO temperature sensor or any other hwmon sensor in 2.5 years, so maybe we leave it as-is and let advanced users figure it out.

And on the 3rd hand, since this is an official LEGO device, maybe we add a hack to make it work with the `lego-sensor` class. This is essentially what the official LEGO firmware is doing.

At the moment, I am leaning towards continuing to use existing Linux drivers and maybe having a better way to auto-detect them. There are a number of 3rd party sensors that fit this category as well. A couple examples...

* http://www.dexterindustries.com/shop/dpressure-sensors-lego-mindstorms-ev3-nxt/
* http://www.ev3dev.org/docs/sensors/mindsensors.com-sensor-building-kit-for-nxt-with-pcf8591-ic/

kortschak, 2016-05-04:
Yes, it's unfortunate. The reason I'm interested in this is that I would like to use some of the LEGO science curriculum material. LEGO officially support this device for that use, so making the libraries I am writing work with the hardware already available makes it more likely that my code has the potential for adoption (this is the 3rd hand argument). The 'other hand' argument makes this use very difficult - I'm looking at using this kind of thing as extension for kids that find LabVIEW too trivial, but expecting them to sort out drivers might be a step too far.

At the moment my next major task is to build something so I can have proper automated testing of ev3go, but after that I'd be very happy to try my hand at helping out with this.

dlech, 2016-05-04:
For the specific case of the temperature sensor, I'm sure I could be persuaded any which way if you are willing to make it work.

So, going with the assumption that we are making it fit the `lego-sensor` class, Here are some useful bits of information.

* [Datasheet](http://www.ti.com/lit/ds/symlink/tmp275.pdf)
* [Modes, scaling and I2C address/registers]( https://github.com/mindboards/ev3sources/blob/master/lms2012/lms2012/Linux_AM1808/sys/settings/typedata.rcf#L32). 
    * Two modes, Celsius and Fahrenheit.
    * Vendor needs to be faked as `LEGO`.
    * Product needs to be faked as `Temp.`
    * I2C address is 0x4C
    * It looks like they are writing 0x60 to register 0x01 at startup. i.e. setting to 12-bit mode
    * Read temperature from register 0x00 (two bytes)
* The way the official LEGO firmware confirms the presence of this sensor is if it [reads back 0x60 from register 0x01] (https://github.com/mindboards/ev3sources/blob/master/lms2012/d_iic/Linuxmod_AM1808/d_iic.c#L762).

kortschak, 2016-05-05:
Thanks. I think the only other thing I need is a pointer to where I hook into to get the started - I can dig to find out, but a pointer would save a huge amount of time.

dlech, 2016-05-05:
We need to add the address 0x4C [here](https://github.com/ev3dev/lego-linux-drivers/blob/master/sensors/nxt_i2c_sensor_core.c#L429) so that it is probed.

Then in [nxt_i2c_sensor_detect](https://github.com/ev3dev/lego-linux-drivers/blob/master/sensors/nxt_i2c_sensor_core.c#L359), we need to add a special case for `if (client->addr == 0x4c)`.

I would like to add the temperature sensor detection as a separate function to keep things clean. Maybe something like this just before `while (tries--) {`:

    if (client->addr == 0x4c) {
        err = nxt_i2c_sensor_detect_lego_temp(client);
        if (err)
            return err;
        snprintf(info->type, I2C_NAME_SIZE, "%s", LEGO_NXT_TEMPERATURE_NAME);
        return 0;
    }
            
The rest should be very similar to the last patch that you fixed up.

gquintanac, 2016-05-05:
Hi diech and kortschak. I am also working with the temperature sensor on the ev3. I have followed your recommendations and now I can read the temperature from the file /sys/class/hwmon/hwmon0/temp1_input  
I am wondering how the polling rate can be modified. I would like to sample faster than with the default rate. Do you have any suggestions?

dlech, 2016-05-05:
What sort of polling rate are you getting? What sort of polling rate do you want?

Looking at the [driver](https://github.com/ev3dev/ev3-kernel/blob/ev3dev-jessie/drivers/hwmon/lm75.c#L260), it looks like this rate is fixed. The hwmon subsystem is not designed for fast polling.

kortschak, 2016-05-05:
https://www.kernel.org/doc/Documentation/hwmon/lm75

dlech, 2016-05-12:
I'm going to leave this issue open for automatic detection of hwmon sensors in general.

The NXT temperature sensor is now a `lego-sensor` device instead of using `hwmon`. 
https://github.com/ev3dev/lego-linux-drivers/commit/7da135613d82fc4b556c04673d5c8b95f37d1cad

dlech, 2016-06-17:
For reference, if we decide we need hwmon sensors to be detected, we can add an `hwmon-i2c` mode for input ports. See [this commit](https://github.com/ev3dev/lego-linux-drivers/commit/2831b1dba46b8521c5dd9f590658aacd53030a36) for how `other-i2c` mode was added.

dlech, 2018-07-20:
There doesn't seem to be any interest in hwmon stuff, so calling this wontfix for now. We can always open the issue again if needed.

closed 2018-07-20 by dlech