ev3dev/ev3dev #624: 【Q】connot open img file in windows vitraul drive

LEXUGE, 2016-05-08:
First,I'm Chinese. I can't speak English very well.Sorry.
I try to download the zip or xz file again.
but not have fixed question.
connot open img file in windows vitraul drive.
And use the win32 disk imager ,it only can write two file:????.scr and ul?????
how to begin to setup EV3DEV.

LEXUGE, 2016-05-08:
help!

ensonic, 2016-05-08:
Have you followed these instructions?
http://www.ev3dev.org/docs/tutorials/writing-sd-card-image-win32diskimager/

dlech, 2016-05-08:
> it only can write two file:boot.scr and ulmage

That sounds correct. You can't see the rest of the files using Windows because they are on an ext4 partition.

Just put the SD card in your EV3 and turn on the EV3.

LEXUGE, 2016-05-08:
@dlech But the win32 disk imager will create a new part of SD card.Size is 48 MB

LEXUGE, 2016-05-08:
@dlech 48MB Would not be able to store the inside of the Linux Image

LEXUGE, 2016-05-08:
@ensonic yes,I followed with this doc

mob41, 2016-05-08:
實際上，那個image是一個partitioned image. Win32Imager會直接把那image連partition table燒上SD card。你現在所看到的partition我相信是一個boot partition 48 MB。而其他partitions是Windows看不見的。因為是ext4，多數是linux/unix用的

mob41, 2016-05-08:
Partition = 分區

如果你直的要看ext4的partition. 可以安裝一個ubuntu/Linux VM在你Windows.
或Windows ext4 driver
[https://www.google.com.hk/search?q=windows+ext4&oq=windows+e&aqs=chrome.0.35i39j69i57j0l2j69i60l2.4037j0j4&sourceid=chrome-mobile&ie=UTF-8](https://www.google.com.hk/search?q=windows+ext4&oq=windows+e&aqs=chrome.0.35i39j69i57j0l2j69i60l2.4037j0j4&sourceid=chrome-mobile&ie=UTF-8)

markyi370, 2016-05-08:
@mob41 how does an ext4 driver help here? The person asked how to setup ev3. You don't need an ext4 driver for that.

dlech, 2016-05-08:
@LEXUGE, you must use the whole `ev3-ev3dev-jessie-2015-12-30.img` file when writing the SD card. It will create two partitions. Just put the SD card in your EV3 and it will boot.

Do not try to open the `ev3-ev3dev-jessie-2015-12-30.img.zip` with windows virtual drive. There is no reason to do this. It will not help you.

mob41, 2016-05-08:
@markyi370 In general, you don't need an ext4 driver. And you can just burn the image to the SD.

In his case, somehow I don't know why he is trying to read the partiton of the SD, he said he doesn't know why the Win32Imager is generating a new partition of 48 mb, and he is asking that that partition is not enough to place a Linux sys.

LEXUGE, 2016-05-09:
@mob41 @markyi370 @dlech Thank you very much,I will try it again!

markyi370, 2016-05-09:
I wonder if an installer like leJOS has would help people like @LEXUGE...

mob41, 2016-05-09:
@markyi370 Actually... leJOS installer also needs the user to write the image to the SD card, the same as ev3dev.

mob41, 2016-05-09:
@markyi370 I think that he doesn't understand about the "ev3dev startup guide", but he did try to understand it.

mob41, 2016-05-09:
I was wondering to add translations in ev3dev.org

markyi370, 2016-05-09:
@mob41 for lejos you just extract lejosimage.zip to sd card. easy

markyi370, 2016-05-09:
@mob41 He could just use Google Translate.

ensonic, 2016-05-10:
@markyi370 that won't work for our partition layout :/ Also for lejos, you need to make sure the sdcard is marked as bootable.

LEXUGE, 2016-05-10:
@dlech And if I run the C program(for example write"Hello EV3"),Can it display on EV3？

markyi370, 2016-05-10:
> And if I run the C program(for example write"Hello EV3"),Can it display on EV3？

@LEXUGE Please open another issue for that question.

LEXUGE, 2016-05-10:
@markyi370 OK

mob41, 2016-05-10:
@LEXUGE Take a look from [here](http://www.ev3dev.org/docs/tutorials/using-ev3-lcd/). I haven't that yet. But it is sure that you need to write pixel data to the framebuffer in order to display on EV3.

markyi370, 2016-05-13:
this should be closed

closed 2017-10-24 by dlech