ev3dev/ev3dev #626: Consider making LEGO Power Storage Sensor explicitly store device name

kortschak, 2016-05-09:
The LEGO_POWER_STORAGE_SENSOR driver does not explicitly store the device name ("Store."). This needs to be investigated further.

See https://github.com/ev3dev/lego-linux-drivers/pull/32/files/272d30e5b70cb0d8210c3ff9dd404617671ee4a0#r62440495

dlech, 2016-05-09:
Does anyone actually have one of these?

kortschak, 2016-05-09:
Yes. See linked comment responses.

dlech, 2016-05-09:
Great. Hadn't seen that yet. First thing to do is run 

    i2c-dump -y -r 0x00-0x07 6 0x02
    i2c-dump -y -r 0x08-0x0f 6 0x02
    i2c-dump -y -r 0x10-0x17 6 0x02

and see what we get. `6` is assuming it is plugged into input port 4.

kortschak, 2016-05-09:
```
robot@ev3dev:~$ /usr/sbin/i2cdump -y -r 0x00-0x07 6 0x02
No size specified (using byte-data access)
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
00: 00 00 00 00 00 00 00 00                            ........        
robot@ev3dev:~$ /usr/sbin/i2cdump -y -r 0x08-0x0f 6 0x02
No size specified (using byte-data access)
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
00:                         00 00 f1 1b 00 00 79 26            ..??..y&
robot@ev3dev:~$ /usr/sbin/i2cdump -y -r 0x10-0x17 6 0x02
No size specified (using byte-data access)
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
10: 00 00 64 00 00 00 00 00                            ..d.....        
```

dlech, 2016-05-09:
OK. I was thinking that from the official LEGO firmware source code I somehow deduced that it returned something useful here, but I guess not. So, we will have to handle it much like the NXT Temperature sensor.

As far as I know, this is the only device at address 0x02, so it may be good enough to just test for anything at all responding on 0x02 and say that it is the Storage sensor. I don't know if any of these registers will be consistent across all devices. Technically, the official firmware is looking that register 0x08 == 0x00 and register 0x10 == 0x00, so I suppose these can be relied upon to always be 0x00.

You could try doing a full i2c dump (`i2cdump -y 6 0x02`) and see if there is anything that looks useful for confirming identity, but if not, I'm not too worried about it. 0x02 is actually an illegal I2C address, so there aren't going to be any "real" I2C devices that would use this address.

kortschak, 2016-05-12:
Nothing. What follows is the dump from a freshly attached, uncharged energy metre. Charging it a little changes the expected bytes.

```
# i2cdump -y 6 0x02
No size specified (using byte-data access)
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
00: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
20: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
60: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
70: 00 XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    .XXXXXXXXXXXXXXX
80: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
90: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
a0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
b0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
c0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
d0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
e0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
f0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
```

dlech, 2016-05-12:
OK. I still think we can use register 0x08 and/or register 0x10 == 0 to test "is there something there?"

kortschak, 2016-05-12:
I have that working, but the scaling is off. I'll look into that tonight.

kortschak, 2016-05-12:
The initial story (only for energy stored value at this stage since it's the only static value without attaching a battery/panel) is that the value is 8 bits too far to the right; when the device's LCD report 1J, I read 256, 2J 512, etc. The scaling looks correct as far as the LMS doc goes, so I'm thinking it has to do with the data type that's specifies in the defs struct.

kortschak, 2016-05-13:
Looking at the scaling and address data I suspect it is LE rather than BE (the set-up sequence in the last column has a negative value at the end - this is not defined anywhere, but I think it may mean that).

dlech, 2016-05-13:
I think you are right. Comes from [here](https://github.com/mindboards/ev3sources/blob/master/lms2012/d_iic/Linuxmod_AM1808/d_iic.c#L1259). And I think [this](https://github.com/mindboards/ev3sources/blob/master/lms2012/d_iic/Linuxmod_AM1808/d_iic.c#L1067) is why I got it wrong in the first place. LEGO is reversing the byte order when `Reverse` is false.

kortschak, 2016-05-13:
> LEGO is reversing the byte order when Reverse is false. 

That's horrible.

dlech, 2016-06-15:
This fix has been released in the [12-ev3dev](http://www.ev3dev.org/news/2016/06/14/Kernel-Release-Cycle-12/) kernel.

closed 2016-06-15 by dlech