ev3dev/ev3dev #627: Installer for ev3dev

markyi370, 2016-05-09:
I would like to suggest a Windows/Linux installer for ev3.

dlech, 2016-05-09:
Installation only consists of downloading an image file and writing it to an SD card. There are already programs that do this. What would be the advantage of writing our own version of one of these and maintaining it?

mob41, 2016-05-09:
@markyi370 leJos uses an installer because it needs to install the leJOS system directly to the Lego firmware. After the installation on the ev3, the SD card is not with the OS. leJOS is on the EV3 firmware.

However, ev3dev does not. ev3dev just requires you to burn a whole image to the SD card. And actually the image is a Linux file-system. ev3dev does not touch the firmware of the ev3. It just works like "dual boot" that if you remove the SD card, the original EV3 firmware will boot. If plug back in the SD card, ev3dev will boot.

Programming a new installer for writing the image to the SD card conveniently is possible, but unnecessary.

WasabiFan, 2016-05-09:
> What would be the advantage of writing our own version of one of these and maintaining it?

One advantage I see is that it can give you an easy way to set the proper options in `config.txt` on RasPi. Overall, I could imagine it doing the following:
- Guide the use through choosing the right image file
- Download the image file for the selected target
- Confirm that the SD card is valid (< 32GB)
- Image the card
- Set configuration options in `config.txt` with descriptions
- Give the user "next steps" (links to website, information on writing actual code, etc).
- Potentially configure networking
  - Report the IP address (for platforms without a display)

Now, whether it's worth the effort right now is a different question. Definitely, for people who are complete novices, this would be helpful. And even for more advanced users, it would be nice to have everything "just work" without manually downloading imaging applications or image files. But the minimum work required for the most basic feature set would be at least a few hours, and likely days to make it properly cross-platform and functional.

Overall, I don't think that it would be a worthwhile investment right now, given that any beginner that got past the installation would still be stuck before they got to running actual code. But once we have guides for use and have demonstrated that the platform is ready, creating an installer would be significantly more attractive.

WasabiFan, 2016-06-01:
We've discovered a new imaging application called [Etcher](http://www.etcher.io) (see #660) that is pretty and mostly functional, although there are some kinks to be worked out. If we want to build an installer-type application at some point, I think our best option would be to fork Etcher and add our own content on top. Using it as a base would significantly reduce the work required to get an installer up-and-running, because it's already cross-platform and written using Web technologies so it's easily accessible to devs.

rhempel, 2016-06-02:
What about forking, adding a feature that allows the user to select an image from a URL, and automagically downloading (or just download and burn incrementally) - I'll be interested to see what you can do with this :-)

WasabiFan, 2016-06-02:
Yeah, I think that should be doable. It should also be fairly easy to ask the GitHub API for a release list and display a real list of releases. Or, even easier, just display a frame with the GitHub website in it :wink: Either way, if we are going to make an installer in the future, I think this is the way to do it.

mob41, 2016-06-02:
I am writing a IDE that can program the ev3dev graphically via putting blocks together, just like the official LEGO one. Would it be great if I implement the "ev3dev" installer in it?

ensonic, 2016-06-03:
Just fro reference - we have a ticket on the openroberta side for such a feature too:
https://github.com/OpenRoberta/robertalab/issues/30

WasabiFan, 2016-06-03:
Ah, interesting -- I didn't consider the "mass imaging" perspective for classrooms. We should make sure to consolidate our efforts so we don't duplicate work if/when we decide to go forward.

dlech, 2016-09-23:
It looks like Etcher has support for image metadata, so we should be able to get most if not all of the suggested ideas implemented in upstream Etcher without having to fork it. See https://github.com/resin-io/etcher/blob/bdf920f86459d4ab63e62bd3b380f6c2d336ceac/docs/EXTENDED-ARCHIVES.md for the current state of this in Etcher.

WasabiFan, 2016-09-23:
I've seen mentions of those features, but never managed to find a definitive description... that's promising! The color and logo customization is also very nice. I know they have those style features implemented (as of a few weeks ago) so at least some of it is implemented, and there are open PRs as well. I'm going to try to dig into this a bit more at some point (probably via their Gitter room) and see what the state of this is overall.

dlech, 2019-01-28:
It would be interesting to try https://www.pibakery.org/ to see if it is possible to to customize ev3dev images.