ev3dev/ev3dev #628: mindsensors.com pressure sensor docs missing ranges

dlech, 2016-05-09:
See <http://www.ev3dev.org/docs/sensors/mindsensors.com-digital-pneumatic-pressure-sensor/>. Use the "Edit on GitHub" button to find the source.

The "Values" column in the "Modes" table should have ranges of possible values in parentheses after "Pressure". For example, `Pressure (0 to 58 PSI)`. Relative modes might need a foot note to explain that the range is 0 - offset to 58 - offset rather than 0 to 58.

WasabiFan, 2016-05-09:
To be clear, you're saying that you want to modify the documentation pipeline to add this info (this is not a bug)?

dlech, 2016-05-09:
If incomplete documentation is a bug, the sure, it's a bug. Someone just needs to do the math and add the correct values for each of the different units of measurement.

Also, see the LEGO NXT Ultrasonic sensor in the same file for an example on how to do value footnotes.

kortschak, 2016-05-10:
The user documentation for the device is sort of lacking. The device pressure range is only specified in PSI. From that we can get rough ranges for the other units. If that is good enough, I'll send a PR.

dlech, 2016-05-10:
Good enough for me.

closed 2016-05-10 by dlech