ev3dev/ev3dev #629: Permission denied with  /sys/devices/platform/snd-legoev3/volume

pambrose, 2016-05-10:
### System information:

* I am using: LEGO MINDSTORMS EV3

* I installed ev3dev using this image file: ev3-ev3dev-jessie-2015-12-30.img

* My kernel version is: 4.4.9-11-ev3dev-ev3

* My host computer is running: OSX

* I am connecting using: Wi-Fi

### About my issue:
I am able to read values from /sys/devices/platform/snd-legoev3/volume
but I am unable to write values to the same location:

```bash
robot@ev3dev:~$ cat /sys/devices/platform/snd-legoev3/volume
100
robot@ev3dev:~$ 
robot@ev3dev:~$ echo 90 > /sys/devices/platform/snd-legoev3/volume
-bash: /sys/devices/platform/snd-legoev3/volume: Permission denied
robot@ev3dev:~$ 
```

I am able to change the volume using **amixer**:

```bash
robot@ev3dev:~$ amixer set Playback,0 80%
Simple mixer control 'Playback',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 256
  Mono: 205 [80%]
robot@ev3dev:~$ !ca
cat /sys/devices/platform/snd-legoev3/volume
80
```

Any help would be appreciated,
Paul

dlech, 2016-05-10:
The file permissions were changed in the `11-ev3dev` kernel to fix some compiler warnings (the kernel used to not be so strict about this).

It would be preferred if you used alsa commands or libraries to change the volume. The `volume` attribute probably won't be around forever. For example, to set the volume in a non-interactive way...

    amixer set "Playback Volume" 50%

If you *really* want to keep using the `volume` attribute, you can setup a udev rule like our other [ev3dev udev rules](https://github.com/ev3dev/ev3dev-rules/blob/ev3dev-jessie/debian/ev3dev-rules.ev3dev.udev) to change the permissions on the attributes in `/sys/devices/platform/snd-legoev3`

dlech, 2016-05-10:
Oops, I see you already figured out the `amixer` part better than I. :smiley:

pambrose, 2016-05-10:
Thanks David.

closed 2016-05-10 by pambrose

cho934, 2016-05-10:
Perfect, I searched this information 2 weeks ago... to be documented in the wiki...