ev3dev/ev3dev #630: Better beginner documentation

markyi370, 2016-05-10:
I am interested in writing a better beginner documentation, can I have a guide?

WasabiFan, 2016-05-10:
I think this is a duplicate of #477 and #492. Please read through those issues, and feel free to comment on them if you want to.

markyi370, 2016-05-10:
I mean for installing ev3dev.

MolarAmbiguity, 2016-05-10:
Which step are you having trouble with? I think the documentation explains the process reasonably well.

dlech, 2016-05-10:
Our website source is at https://github.com/ev3dev/ev3dev.github.io. Feel free to make a pull request for any improvements.

rhempel, 2016-05-10:
Agreed, @markyi370, please comment here on the places where the documentation is unclear so that we can better understand where you think the documentation needs work. We're always happy to have pull requests for improvements to the project.

psychemedia, 2016-05-12:
FWIW, I posted a walkthough of getting ev3 up and running with ev3dev and IPython - https://blog.ouseful.info/2016/05/12/using-ipython-on-lego-ev3-robots-running-ev3dev/ Feel free to take any of that as CC0 licensed for use in ev3dev docs.

laurensvalk, 2016-05-13:
> I posted a walkthough of getting ev3 up and running with ev3dev and IPython

Nice! So since the web server is not running on the ev3 in your case, how do you find the responsiveness of your setup?

(Running [everything on the EV3] (https://www.youtube.com/watch?v=vJHr6vIZk0k) is quite slow and sometimes it gets stuck. I'm thinking your approach might run much faster.) 

I'll see if I can try this for a robot with a relatively tight control loop.

psychemedia, 2016-05-13:
@laurensvalk it takes a fair time for the first command to fire (I've gotten into the habit of running `print('hello world') as the first line, but then it's pretty good when you're doing some simple line at a time stuff.

I've only just started working with the brick, trying to document working examples of all the python functions myself, so I haven't actually got as far as controlling anything yet!

One thing I noticed using the `apt-get` version of `ipython` is that it's pretty old. If I can get a newer version installed, it would be interesting to see if the `interact()` function in `ipywidgets` works... (`easy_install` breaks with trying to install a recent version of `ipython` as a dependency, but I have remember having success using `pip` before?)

psychemedia, 2016-05-14:
Just tried getting `ipywidgets` into the brick (there's a load of Jupyter stuff pulled in as requirements, much of which I suspect is superfluous given I'm running a remote IPython kernel on the brick and Jupyter on a laptop host), and they seem to work:

[image]

So - what I'd like now is to log a light sensor value on the brick, for example, and stream the data as it's collected into an IPython widget. Anyone know how I can do that?;-)

rhempel, 2016-05-14:
Oooh, that looks really interesting! Still working on getting the Python binding properly assembled - sorry for the delay

psychemedia, 2016-05-14:
[Example notebooks here](https://gist.github.com/psychemedia/1e2282149a17f5c16252c1ed7c8c6ce6) (using old bindings).

If you like notebook style, line at a time coding (which I do), it seems to work quite well. As a model for presenting instructional / tutorial material, it's also quite powerful, I think?

To make things even easier, I think it might be useful to have a Docker container with the Jupyter notebook and *irkernel* package installed, along with a script/dashboard to help set up the connection definition file the first time. Two issues with this: a) I'm not sure how this would work in terms of setting up ports for the Jupyter server to tunnel to the remote ev3 IPython kernel? b) the brick would have to have  a public / internet observable IP address if the container was running in the cloud, for example?

psychemedia, 2016-06-01:
We've bailed on running ev3dev for a day long residential school activity on introductory robotics because we've run out of development time, so I've posted the code for the activities we were developing at https://github.com/psychemedia/ev3robotics 

I'll add other materials as I can.

markyi370, 2017-05-16:
I'm going to close this issue now (because I think the documentation is sufficient now).

closed 2017-05-16 by markyi370