ev3dev/ev3dev #631: Does rpi2-ev3dev-jessie-2015-12-30.img can be used to boot rpi2+BiickPi (not BrickPi+) ?

yang2218, 2016-05-11:
### System information:

* I am using: *PASTE DEVICES HERE*

* I installed ev3dev using this image file: *IMAGE NAME GOES HERE*

* My kernel version is: *KERNEL VERSION GOES HERE*

* My host computer is running: *OS GOES HERE*

* I am connecting using: *PASTE METHOD HERE*

### About my issue:

#### What led up to the situation?

#### What exactly did you do (or not do) that was effective (or ineffective)?

#### What was the outcome of this action?

#### What outcome did you expect instead?

dlech, 2016-05-11:
Yes. Be sure to edit `config.txt` before booting. It explains what to turn on if you are using BrickPi.

yang2218, 2016-05-11:
But I get the following Error:

Power_supply brickpi-battery driver failed to report voltage_now property -5

yang2218, 2016-05-11:
Sorry, I make a mistake to comment and close. It sure be just comment.

dlech, 2016-05-11:
It sounds like you did not follow the directions:

> # uncomment only if you are using BrickPi+
> #dtparam=brickpi_battery=okay

The BrickPi does not have the capability to monitor battery voltage. So *do not* uncomment this line.

closed 2017-08-12 by dlech