ev3dev/ev3dev #633: Cannot read LEGO IR sensor with BrickPi

JorgePe, 2016-05-11:
### System information:

    Dexter Industries BrickPi (HW v1.7.3)
    Rapsberry Pi Model 3B

* I am using: LEGO EV3 Infra-red sensor

* I installed ev3dev using this image file: rpi2-ev3dev-jessie-2015-12-30.img

* My kernel version is: 4.4.9-11-ev3dev-ev3

* My host computer is running: Ubuntu 16.04

* I am connecting using: Wi-Fi

### About my issue:

I'm trying to use LEGO EV3 IR sensor with BrickPi.
I boot the RPi3+BrickPi with no sensors or motors connected.
dmesg shows BrickPi is registered
I initialize port0 (i.e. S1):
```
echo ev3-uart > /sys/class/lego-port/port0/mode
echo lego-ev3-ir > /sys/class/lego-port/port0/set_device
```

I connect the IR sensor and check mode
```
cat /sys/class/lego-sensor/sensor0/mode
IR-PROX
```

I read the value
```
cat /sys/class/lego-sensor/sensor0/value0
0
```

It's always 0.

The sensor is OK, it works with my EV3 (running same kernel version).

Strange fact:
If I boot the RPi3/BrickPi with sensor already connected to S1, I cannot set port mode:
```
cat /sys/class/lego-port/port0/modes
none nxt-analog nxt-color nxt-i2c
```

both port0 and port1 modes show the same BUT port4 and port5 modes show the expected modes:
```
none nxt-analog nxt-color nxt-i2c ev3-analog ev3-uart
```

First I suspected not enough power, as I was booting only with USB power from the RPi3. But it also happens with power only from the BrickPi (8xAA pack) and power from both sources simultaneously.

JorgePe, 2016-05-11:
Problem is not with LEGO EV3 IR sensor.
Same problem with LEGO EV3 Gyro (45505) - also reads very well with EV3 (ev3dev, same kernel).
Some kind of problem with uart readings?

I checked /boot/flash/config.txt,
```
init_uart_clock=32000000
```

dlech, 2016-05-11:
I'm going to say that we can probably call this a duplicate of #542. The BrickPi firmware has some issues with EV3/UART sensors.

I have noticed some of the same issues, such as the BrickPi locking up if you have the sensor plugged in before you change the mode.

> Strange fact:
> If I boot the RPi3/BrickPi with sensor already connected to S1, I cannot set port mode:
> 
>    cat /sys/class/lego-port/port0/modes
>    none nxt-analog nxt-color nxt-i2c

This means for some reason the microprocessor on the BrickPi is reporting firmware version 1 instead of version 2. Probably the microprocessor has got into a bad state.

> First I suspected not enough power, as I was booting only with USB power from the RPi3.

We do not support back-powering like this on any device. Always use the power input connector on the BrickPi. (Same goes for PiStorms and EVB).

JorgePe, 2016-05-11:
> This means for some reason the microprocessor on the BrickPi is reporting firmware version 1 instead of version 2.

How do I check my BrickPi firmware? It was bought almost 2 years ago and never upgraded.

dlech, 2016-05-11:
Dexter Industries has never released any updated firmware, so you have the latest. You obviously have version 2 since EV3 sensors are at least sort of working.

JorgePe, 2016-05-12:
OK, BrickPi firmware doesn't support EV3 sensors, just EV3 motors. My bad, should have read that at Dexter Industries BrickPi forum.
Closing this, thank you.
If i can get a firmware upgrade for my BrickPi I will report if I get success.

closed 2016-05-12 by JorgePe

dlech, 2016-05-12:
BrickPi firmware does support EV3 sensors, just not very well. There is no such thing as a firmware upgrade for BrickPi. None have ever been released.

JorgePe, 2016-05-12:
Their github has some non-final firmwares, they say one need an Arduino and a modded cable, I have an Arduino UNO somewhere and I can make the cable so I was thinking doing it one of those days (more likely never ;)

dlech, 2016-05-12:
In the very early days of BrickPi, they did try to attempt to do firmware updates, but if you bought a BrickPi in the last 2 years, it will have the latest and greatest firmware already. If you do try some other firmwares, it will be interesting to see if there is a difference, but I suspect that they will only make things worse and not better. Also, be aware that the 2.5 firmware is for BrickPi+ (hw v2.8) only because it has different motor controller chips that are wired differently. It could damage your BrickPi if you use this firmware.

JorgePe, 2016-05-12:
Hmm... I read that 2.5/2.8 somewhere here in ev3dev github. I think I leave this firmware upgrade for thee greek calends, will probably get a BrickPi+ first.
My BrickPi or BrickPi Advanced (Dexter's naming is confusing) was bought 2 years ago but I don't know how much time it spent at my local distributor warehouse, probably alone. Portugal is a very small country and not geeky enough to sustain local distributors.