ev3dev/ev3dev #634: BrickPi How to use LEGO NXT Ultrasonic Sensor?

JorgePe, 2016-05-12:
### System information:

Dexter Industries BrickPi (HW v1.7.3)
Rapsberry Pi Model 3B

* I am using: LEGO NXT Ultrasonic sensor (9846)

* I installed ev3dev using this image file: rpi2-ev3dev-jessie-2015-12-30.img

* My kernel version is: 4.4.9-11-ev3dev-ev3

* My host computer is running: Ubuntu 16.04

* I am connecting using: Wi-Fi

### About my issue:

This is not a bug report, maybe a documentation change request or just a case of not searching enough.

Since my BrickPi firmware doesn't support EV3 sensors, I decided to try the few NXT sensors I have.
So, boot RPi3/BrickPi with external power source connected to BrickPi and USB power source connected to RPi3 and no devices connected to BrickPi ports.

```
cat /sys/class/lego-port/port0/modes
none nxt-analog nxt-color nxt-i2c ev3-analog ev3-uart
```
Now I connect the LEGO NXT Ultrasonic sensor. According to Docs,  mode is "nxt-i2c" and device is "lego-nxt-us", so
```
echo nxt-i2c > /sys/class/lego-port/port0/mode
echo lego-nxt-us > /sys/class/lego-port/port0/set_device
-bash: echo: write error: Invalid argument
```
dmsg:
```
lego-port port0: set_device: Missing parameters -requires both driver name and I2C address
```

Never needed to give I2C address of LEGO sensors, not sure but I think never saw anything in ev3dev site about that but if others have same "problem" this was how I got it working:

The same Docs say I2C address is 0x01 so why not?
echo "lego-nxt-us 0x01" > /sys/class/lego-port/port0/set_device

dmesg:
```
lego-port port0: Added new device 'ttyAMA0:S1:lego-nxt-us'
lego-sensor sensor0: Registered 'lego-nxt-us' on 'ttyAMA0:S1:i2c1'.
```

Sensor seems to work OK so I finally can use my BrickPi with LEGO sensors (was using just EV3 motors).
And although I know it's not supported, after using it this way I disconnected external power source to the  BrickPi (kept USB power to the RPi3) and it still works, but a reading of 14 cm is now 16 cm - not good for precision but perhaps good enough for fuzzy decisions.

dlech, 2016-05-12:
> Never needed to give I2C address of LEGO sensors

This is because EV3 can automatically detect this sensor and BrickPi cannot automatically detect any sensors. See http://www.ev3dev.org/docs/sensors/#automatic-detection

JorgePe, 2016-05-12:
Yes, I was aware of that. Documentation is getting better every month but it's still hard to read everything and find everything (at least for me, sorry).

The link you send has a note "3" that links to
http://www.ev3dev.org/docs/drivers/lego-port-class/

but although it explains "set_device" attribute it doesn't states that the I2C address is also part of the attribute.

dlech, 2016-05-12:
Pull requests are welcome to make the documentation better!

Most pages have an "Edit on GitHub" button to get you to the correct source.

WasabiFan, 2016-05-12:
@JorgePe What do you think would make it easier to find information in the docs? We'd love to hear feedback if you have it.

JorgePe, 2016-05-12:
My problem is mostly with details. I know that most users, probably also you guys, don't have all devices to test. And a NXT sensor with BrickPi is also far away from the "core", using an EV3 with EV3 devices.

So proposing a configuration example for every device is crazy. Perhaps a FAQ? Or a script for users to run to contribute to a database of configurations used around thw world?

(talking about scripts... a map of ev3dev around the world, somehwere in ev3dev site, like the Raspberry Pi  at http://rastrack.co.uk/ would seem to nerd?)

As I said, I find documentation is getting better almost every day. That list of sensors (and the list of motors/actuators) is a great starting point for me, it just needs more details like these set_config attribute.

I had another issue today, after the NXT US, I connected an EV3 medium motor to the BrickPi and tried to use my [short] python MQTT scripts to control it from my EV3. Had to read the python github to find that I have to use " Motor('ttyAMA0:MA')" instead of "MediumMotor(OUTPUT_A)" in the BrickPy. I knew that MediumMotor and LargeMotor are for EV3 motors only with the EV3 but didn't understand that Motor was the proper type to use with the BrickPi.

I got there, that's what matters :)

JorgePe, 2016-05-12:
@dlech I think i did just that. I find awkward to see code inside - I added some words to the comments, I hope that's the proper way.

dlech, 2016-05-12:
You need to click the "Create Pull Request" button.

JorgePe, 2016-05-12:
Uff... to much information for one day :)

JorgePe, 2016-05-13:
OK. I made my first documentation contribution (if one can call it that).
Case closed, will try to see what I can do to help you [and me] further.

closed 2016-05-13 by JorgePe