ev3dev/ev3dev #635: 【Q】Can't Upgrade

LEXUGE, 2016-05-12:
Help!
It said:
robot@ev3dev:~$ apt-get update
E: Could not open lock file /var/lib/apt/lists/lock - open (13: Permission denied)
E: Unable to lock directory /var/lib/apt/lists/
E: Could not open lock file /var/lib/dpkg/lock - open (13: Permission denied)
E: Unable to lock the administration directory (/var/lib/dpkg/), are you root?

LEXUGE, 2016-05-12:
@dlech

ddemidov, 2016-05-12:
try `sudo apt-get update`

LEXUGE, 2016-05-12:
robot@ev3dev:~$ sudo apt-get update

We trust you have received the usual lecture from the local System
Administrator. It usually boils down to these three things:

    #1) Respect the privacy of others.
    #2) Think before you type.
    #3) With great power comes great responsibility.
@ddemidov

ddemidov, 2016-05-12:
That is normal. `sudo` means 'do something with a superuser rights'. So it gives you this lecture (first time) and asks for your password.

LEXUGE, 2016-05-12:
But,what is the password of [sudo]@ddemidov

ddemidov, 2016-05-12:
This is *your* password, meaning the password of the current user. Its `maker` by default for the `robot` user.

LEXUGE, 2016-05-12:
Thank you very much

LEXUGE, 2016-05-12:
But:
[sudo] password for robot:
Err http://ftp.debian.org jessie InRelease

Err http://ev3dev.org jessie InRelease

Err http://ftp.debian.org jessie Release.gpg
  Temporary failure resolving 'ftp.debian.org'
Err http://ev3dev.org jessie Release.gpg
  Temporary failure resolving 'ev3dev.org'
Reading package lists... Done
W: Failed to fetch http://ftp.debian.org/debian/dists/jessie/InRelease

W: Failed to fetch http://ev3dev.org/debian/dists/jessie/InRelease

W: Failed to fetch http://ftp.debian.org/debian/dists/jessie/Release.gpg  Temporary failure resolving 'ftp.debian.org'

W: Failed to fetch http://ev3dev.org/debian/dists/jessie/Release.gpg  Temporary failure resolving 'ev3dev.org'

W: Some index files failed to download. They have been ignored, or old ones used instead.
@ddemidov

LEXUGE, 2016-05-12:
Help!

ddemidov, 2016-05-12:
That means your ev3 is not connected to the internet.

LEXUGE, 2016-05-12:
What?
Oh!In the All network->IPv4
I load the default windows,but it do not back to All Network

LEXUGE, 2016-05-12:
I only click the connect

LEXUGE, 2016-05-12:
How can connect the internet!I followed the dochttp://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-usb/

LEXUGE, 2016-05-12:
@ddemidov

ddemidov, 2016-05-12:
I am not sure how to do that on widows, but I guess the important steps in the tutorial you linked to are 7-9. Simple test: `ping google.com` should work on the ev3.

LEXUGE, 2016-05-12:
Oh,But I followed the doc
http://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-usb/
@dlech

dlech, 2016-05-12:
At this point in time, I highly recommend *not* upgrading any packages unless you want a broken system that you are planning on fixing yourself.

If you are not connected to the Internet, then you probably did not follow the tutorial exactly. Also, Windows Internet Connection Sharing breaks easily. It is usually fixed by disabling it and then setting it up again.

dlech, 2016-05-12:
Also, many of these questions have been asked and answered before. Please search issues (open and closed) before asking a question.

LEXUGE, 2016-05-12:
Oh,But I didn't faild.I always do by docs
Yes,I changed it,but it didn't back.But ,I update faild,haha!
[image]

They are all good,I didn't setting faild

LEXUGE, 2016-05-12:
What a magical question

LEXUGE, 2016-05-12:
But the other questions can't help me.:-(
Can it work on buletooth?(windows)
@dlech

dlech, 2016-05-12:
Bluetooth works for me on Window.

LEXUGE, 2016-05-12:
@dlech I can ping the website,Am I connect the Internet?

LEXUGE, 2016-05-12:
Yes,it is succesful!

LEXUGE, 2016-05-12:
So,you said,do not upgrade the sys.If there is a new version,how can I upgrade the new version?Resetup?
@dlech

dlech, 2016-05-12:
When we release a new image file, you can install the new image file to the SD card.

You could also use `sudo apt-get update && sudo apt-get upgrade && sudo apt-get dist-upgrade` at that time, but it will take a very long time. It will be much faster to backup any files you want and write the new image file.

LEXUGE, 2016-05-12:
OK,thank you very much,and what time in your country?
"sudo apt-get update && sudo apt-get upgrade && sudo apt-get dist-upgrade"is this one command?

dlech, 2016-05-12:
11 AM.

> is this one command?

It can be. `&&` means if the previous part was successful, then do the next part. You could also run them as separate commands.

    sudo apt-get update
    sudo apt-get upgrade
    sudo apt-get dist-upgrade

LEXUGE, 2016-05-12:
You are so great,I am only a pupil in China

LEXUGE, 2016-05-12:
Ask a very easy question:Can the computer give the electric to EV3?

dlech, 2016-05-12:
No. It requires at least a 1A power supply. 3A is better if you are running more than 1 motor.

LEXUGE, 2016-05-12:
Oh,it only have 7.86 betty now
the EV3DEV can't install many software(for example:w3m)

dlech, 2016-05-12:
I don't understand, what is "7.86 betty"

Also, you need to run `sudo apt-get update` before you can install any software with `sudo apt-get install`.

LEXUGE, 2016-05-12:
But,I input sudo apt.......gcc,it said:
robot@ev3dev:~$ sudo apt-get install gcc
Reading package lists... Done
Building dependency tree... Done
E: Unable to locate package gcc

dlech, 2016-05-12:
Yes. You must run `sudo apt-get update` first, otherwise the package list is empty.

LEXUGE, 2016-05-12:
原来如此,Thanks!

LEXUGE, 2016-05-12:
Are you working now?have I bother you yet?

dlech, 2016-05-12:
If you bothered me too much, I would not answer your questions so much. :wink:

ev3dev is my work most of the time.

LEXUGE, 2016-05-12:
haha,it can visit my website:

dlech, 2016-05-12:
The link is missing.

LEXUGE, 2016-05-12:
lexuge.github.io

LEXUGE, 2016-05-12:
But it isn't English

LEXUGE, 2016-05-12:
@dlech How to uninstal softwarel?Help!

dlech, 2016-05-12:
`sudo apt-get remove <package>`

dlech, 2016-05-12:
Yes, that also deletes any configuration files (the ones in `/etc/`)

LEXUGE, 2016-05-12:
Thanks.How can I visit my FTP

dlech, 2016-05-12:
Install an ftp client.

LEXUGE, 2016-05-12:
oh,I thought it have one client.But I'm error.

LEXUGE, 2016-05-12:
This is a very good sys.I like it!

closed 2017-10-24 by dlech