ev3dev/ev3dev #637: Connect to another EV3 or another device by i2c (arduino or mbed) ?

cho934, 2016-05-13:
Hi,

What's your recommandations to connect several EV3 ?

What's your recommandations to connect another custom device by i2c (arduino or mbed) ?

BR,

dlech, 2016-05-13:
> What's your recommandations to connect several EV3 ?

TCP/IP. With Wi-Fi, they are already connected. With Bluetooth and USB, you can use the tethering feature in Wireless and Networks. Enable tethering on the controlling EV3 but not on the remote one.

> What's your recommandations to connect another custom device by i2c (arduino or mbed) ?

Force an input port into I2C mode with `echo nxt-i2c > mode`. Pin 5 is SCL, pin 6 is SDA and pin 3 is GND. Use 82k pullup resistors on the two signal lines and connect to your Arduino's I2C pins. 3.3V or 5V is OK.

cho934, 2016-05-13:
Several EV3:
And we can use sockets for example to send/receive command ?

I2C:
And then we can use some generic i2C functions using ev3devlang (in cpp) ? Which ones ?

dlech, 2016-05-13:
> And we can use sockets for example to send/receive command ?

Yes. I don't know if there is anything existing in C++ for this, so you will probably have to write your own.

> And then we can use some generic i2C functions using ev3devlang (in cpp) ? Which ones ?

Generic functions: yes. In ev3devlang: no - it's just a standard Linux library.

This should get you started: http://www.elinux.org/Interfacing_with_I2C_Devices

dlech, 2016-05-13:
I tend to use these functions though:

```
i2c_smbus_read_byte_data — SMBus “read byte” protocol
i2c_smbus_write_byte_data — SMBus “write byte” protocol
i2c_smbus_read_word_data — SMBus “read word” protocol
i2c_smbus_write_word_data — SMBus “write word” protocol
i2c_smbus_read_block_data — SMBus “block read” protocol
i2c_smbus_write_block_data — SMBus “block write” protocol 
```

See https://www.kernel.org/doc/htmldocs/device-drivers/i2c.html

dlech, 2016-05-13:
And of course you should read this: http://www.ev3dev.org/docs/sensors/using-i2c-sensors/

closed 2017-10-24 by dlech