ev3dev/ev3dev #638: 【Q】A question about gcc and build-essential

LEXUGE, 2016-05-14:
today,I install the gcc by `sudo apt-get install gcc`
Now,it can compile and run the C program.
But,the person from Internet said:"You must install build-essential"
Must I install it?If I install it.What can it do?

mob41, 2016-05-14:
@LEXUGE Actually, these questions about Linux should not be asked at here, as they are not related to ev3dev. You should ask these questions like on [stackexchange.com](stackexchange.com).

I actually don't install ```gcc``` and ```build-essential``` on my ev3dev environment. You should install it to a development machine (A machine for developing programs)

LEXUGE, 2016-05-14:
@dlech

dlech, 2016-05-14:
> Must I install it?If I install it.What can it do?

You can do whatever you want to. `build-essential` also installs `make` which is useful when compiling more than one file. Beyond that, I don't really know what else it provides. This is not my area of expertise.

LEXUGE, 2016-05-14:
@dlech So,I only install gcc.Can it work to compile one file?I think I don't dev big project/

dlech, 2016-05-14:
yes

LEXUGE, 2016-05-14:
thank you.the C program can run on the EV3

closed 2017-10-24 by dlech