ev3dev/ev3dev #640: brickpi: sensor registration not working for nxt-i2c devices

kortschak, 2016-05-14:
### System information:

* I am using: Dexter Industries [BrickPi+ (HW v2.8)  (2.7 according to etched V# on case)].

* I installed ev3dev using this image file: rpi2-ev3dev-jessie-2015-12-30.img

* My kernel version is: 4.4.9-11-ev3dev-rpi2

* My host computer is running: Linux

* I am connecting using: Wi-Fi

### About my issue:

#### What led up to the situation?
Build [sensor registration package for ev3go](https://github.com/ev3go/brickpi/blob/master/sensors.go) and test with [example code](https://github.com/ev3go/brickpi/blob/master/examples/sensor/sensor.go).

Run example with an i2c device and get EINVAL error returned. e.g.:

```
$ go install github.com/ev3go/brickpi/examples/sensor
$ sensor -driver ms-pps58-nx
2016/05/14 23:36:44 failed to find sensor: ev3dev: failed to set attribute /sys/class/lego-port/port0/set_device: write /sys/class/lego-port/port0/set_device: invalid argument
```

Shell equivalent:

```
$ echo nxt-i2c >/sys/class/lego-port/port0/mode
$ echo mx-pps58-nx >/sys/class/lego-port/port0/set_device
-bash: echo: write error: Invalid argument
```

The same error is returned for all I2C/NXT devices I have tested.

#### What exactly did you do (or not do) that was effective (or ineffective)?
Above.

#### What was the outcome of this action?
Above.

#### What outcome did you expect instead?
Registration of device. In the case of the sensor Go code, then also emitting values (all working for `ev3-analog`/`lego-ev3-touch`).

The docs at http://www.ev3dev.org/docs/ports/brickpi-in-port/ indicate that this should work - indeed that it is required.

dlech, 2016-05-15:
You are forgetting the I2C address for `set_device`

    $ echo mx-pps58-nx 0x0C >/sys/class/lego-port/port0/set_device

dlech, 2016-05-15:
If you check `dmesg`, I think it should tell you this.

dlech, 2016-05-15:
Duplicate of #634. Maybe we should open a new issue to discuss using the default I2C address if no address is specified.

kortschak, 2016-05-15:
Cool. I'll send a PR later today clarifying that (should just be a few words in the I2C subscript saying that the address is needed and linking to the using I2C page).

It looks like there are a number of devices that don't work on the BrickPi - testing this to make sure I understood, the two mindsensors devices I tried did not work, but the LEGO energy store did:

```
[  217.572720] lego-port port0: Added new device 'ttyAMA0:S1:ms-absolute-imu'
[  217.603425] brickpi-i2c-sensor ttyAMA0:S1:ms-absolute-imu: The 'ms-absolute-imu' driver requires special operations that are not supported in the 'brickpi-i2c-sensor' module.
[  217.610343] brickpi-i2c-sensor: probe of ttyAMA0:S1:ms-absolute-imu failed with error -22
[  295.027155] lego-port port0: Removed device 'ttyAMA0:S1:ms-absolute-imu'
[  420.346624] brickpi-i2c-sensor ttyAMA0:S1:ms-pps58-nx: The 'ms-pps58-nx' driver requires special operations that are not supported in the 'brickpi-i2c-sensor' module.
[  420.354526] brickpi-i2c-sensor: probe of ttyAMA0:S1:ms-pps58-nx failed with error -22
[  420.360071] lego-port port0: Added new device 'ttyAMA0:S1:ms-pps58-nx'
[  468.377730] lego-port port0: Removed device 'ttyAMA0:S1:ms-pps58-nx'
[  741.318525] lego-sensor sensor0: Registered 'lego-power-storage' on 'ttyAMA0:S1:i2c2'.
[  741.329334] lego-port port0: Added new device 'ttyAMA0:S1:lego-power-storage'
[  804.348557] lego-port port0: Removed device 'ttyAMA0:S1:lego-power-storage'
[  804.351862] lego-sensor sensor0: Unregistered 'lego-power-storage' on 'ttyAMA0:S1:i2c2'.
```

Is there a principle that I can use to figure out which will and will not work? (other than trying).

kortschak, 2016-05-15:
Also, it looks like the sensor address has ":i2c2" appended. Is this documented somewhere (really just is it guaranteed to stay consistent)?

```
$ cat /sys/class/lego-port/port0/address 
ttyAMA0:S1
$ cat /sys/class/lego-sensor/sensor5/address 
ttyAMA0:S1:i2c2
```

dlech, 2016-05-15:
> Is there a principle that I can use to figure out which will and will not work?

This is probably something we could improve. You can see [where the error comes from](https://github.com/ev3dev/lego-linux-drivers/blob/fc9c1cc896d013f520c5d7a47414bb07f0a9cb5f/brickpi/brickpi_i2c_sensor.c#L124). If the sensor has special ops, the we say you can't use it on BrickPi.

This is because the ops assume you have a Linux I2C adapter available. But the BrickPi doesn't work this way. It's been a while since I looked at it, so I don't remember exactly how generic I2C ops work with it - or if that is even a possibility. If it is, we could write a Linux I2C adapter driver for it and then those sensors could work seamlessly.

If you search [nxt_i2c_sensor_defs.c](https://github.com/ev3dev/lego-linux-drivers/blob/master/sensors/nxt_i2c_sensor_defs.c) for `.ops`, you will see that there are currently 8 sensors that have ops.

dlech, 2016-05-15:
> Also, it looks like the sensor address has ":i2c2" appended. Is this documented somewhere (really just is it guaranteed to stay consistent)?

It is possible to have multiple I2C devices connected to a single port (maybe not on the BrickPi, but in general). So having `i2c<addr>` tacked on the end is necessary to prevent name collisions.

Of course, `<addr>` will usually be the default address, but some sensors (e.g. many mindesensors.com sensor) it is possible to change the address.

kortschak, 2016-05-15:
The `<addr>` there is decimal or hex (or link to source)?

dlech, 2016-05-15:
[decimal](https://github.com/ev3dev/lego-linux-drivers/blob/master/sensors/nxt_i2c_sensor_core.c#L259)

kortschak, 2016-05-16:
>If you search `nxt_i2c_sensor_defs.c` for `.ops`, you will see that there are currently 8 sensors that have ops.

I count 7: "ht-nxt-smux", "mi-xg1300l", "ms-absolute-imu", "ms-ev3-smux", "ms-nxtmmx", "ms-8ch-servo", "ms-pps58-nx".

dlech, 2016-05-16:
I must have counted the one in the comments.

bmegli, 2016-06-02:
> It is possible to have multiple I2C devices connected to a single port (maybe not on the BrickPi, but in general). So having  i2c<addr>  tacked on the end is necessary to prevent name collisions.

BrickPi should support up to 8 (eight)  I2C devices on each of its regular sensor ports, at least in theory. 

Excerpt from https://github.com/DexterInd/BrickPi_C/blob/master/Drivers/BrickPi.h:

``` C
/* 
172   I2C 
173 */ 
174   unsigned char SensorI2CDevices[4];        // How many I2C devices are on each bus (1 - 8). 
175   unsigned char SensorI2CSpeed  [4];        // The I2C speed. 
176   unsigned char SensorI2CAddr   [4][8];     // The I2C address of each device on each bus.   
177   unsigned char SensorI2CWrite  [4][8];     // How many bytes to write 
178   unsigned char SensorI2CRead   [4][8];     // How many bytes to read 
179   unsigned char SensorI2COut    [4][8][16]; // The I2C bytes to write 
180   unsigned char SensorI2CIn     [4][8][16]; // The I2C input buffers 
181 }; 
```

From what I have seen it is set in `brickpi_in_port_set_i2c_mode` in `brickpi_ports_in.c` to use only the first (of eight possible) devices. This seems reasonable but possibility for improvement exists (yeah, why not attach 32 sensors...)

bmegli, 2016-06-02:
> I count 7: "ht-nxt-smux", "mi-xg1300l", "ms-absolute-imu", "ms-ev3-smux", "ms-nxtmmx", "ms-8ch-servo", "ms-pps58-nx".

Of those 7, 2 (ms_imu, mi_xg1300l) don't even touch smbus so they could be made to work even without I2C adapter driver (they just make some post processing or keep some data).

But adapter would simplify a lot. I am also tempted to write it.
Not that I know how... ;-)

dlech, 2016-10-11:
Is there any action we should be taking here or can we close this issue?

kortschak, 2016-10-12:
I have nothing, but @bmegli may.

bmegli, 2016-10-12:
- documentation which i2c sensors don't work on BrickPi
 - at some point i2c adapter but gain is little so this is low priority

Those should be separate issues

dlech, 2016-10-12:
@bmegli, could you please open new issues for these?

bmegli, 2016-10-12:
@dlech 

Ok, I have added both in #757

This issue is solved now.

closed 2016-10-12 by dlech