ev3dev/ev3dev #641: PyBluez for connecting to a wii remote? Wiimote not working ...

rpgit12, 2016-05-15:
(Hope this is he right place to inquire.. otherwise please let me know.)

Does anyone have a word of advice before I got attempt to install PyBluez for accessing a wiimote?

Following one of the demos on ev3dev.org, I'm trying to connect to a wiimote for remote control. The device seems to work fine for bluez - I was able to find & pair using bluetoothctl - but the python wiimote package doesn't seem to work. I believe it's not recognizing the remote model; Nintendo RVL-CNT-01-TR, class 0x000508. (The wiimote packages looks abandoned since 2010.)

So now I'm looking PyBluez, but that requires pit for install, which in turn requires all of the dev tools on the brick.

JorgePe, 2016-05-15:
I once used python-cwiid to connect to the Wiimote.
http://ofalcao.pt/blog/en/2014/controlling-the-sbrick-with-a-wiimote

I did not use the EV3, just my Ubuntu laptop but I think it is possible to do it.

rpgit12, 2016-05-15:
Thanks; I actually found many references to working examples. I believe it's the more recent wii devices that wiimote doesn't recognize (and there's no contributions to that package since around 2010.) Is it possible you used an older wiimote remote? (Mine are about 2 year old.)

JorgePe, 2016-05-15:
Yes, quite possible. Mine is also about 2 years old.

dlech, 2016-05-15:
how about `sudo apt-get get install python-bluez`? It gets you PyBluez 0.20.

rpgit12, 2016-05-15:
That worked, thank you.

Btw, the issue seems to be with cwiid; it identifies the device through class 0x002504 while newer models of the wiimote use 0x000508 (also see wikipedia.) Bluez works fine. Fixing cwiid might still end up being the cheapest route.

closed 2016-05-15 by rpgit12