ev3dev/ev3dev #642: mindsensors light sensor array RAW mode gives unexpected results

kortschak, 2016-05-17:
### System information:

* I am using: LEGO MINDSTORMS EV3

* I installed ev3dev using this image file: ev3-ev3dev-jessie-2015-12-30.img

* My kernel version is: 4.4.9-11-ev3dev-ev3

* My host computer is running: Linux

* I am connecting using: Wi-Fi

### About my issue:

#### What led up to the situation?
I am reasonably confident this is a regression, though I have not had a chance to check this yet.

#### What exactly did you do (or not do) that was effective (or ineffective)?
Plug in a mindsensors light sensor array and set to RAW mode, then get values[0-7] over varying positions over a line.

No black or white calibration was done prior to the following.

#### What was the outcome of this action?
No change in values corresponding to position of sensor relative to line.

```
$ echo CAL >/sys/class/lego-sensor/sensor0/mode; \
> cat /sys/class/lego-sensor/sensor0/value[0-7]; \
> echo; \
> echo RAW >/sys/class/lego-sensor/sensor0/mode; \
> cat /sys/class/lego-sensor/sensor0/value[0-7]
100
100
100
100
0
0
0
0

69
57
83
87
70
78
62
59
$ echo CAL >/sys/class/lego-sensor/sensor0/mode; \
> cat /sys/class/lego-sensor/sensor0/value[0-7]; \
> echo; \
> echo RAW >/sys/class/lego-sensor/sensor0/mode; \
> cat /sys/class/lego-sensor/sensor0/value[0-7]
0
0
0
0
100
100
100
100

58
48
70
87
78
86
66
84
```

#### What outcome did you expect instead?
Correlation between position and output.

CAL mode does behave as expected, and I recall that RAW mode worked with release 10.

I noticed this originally on a BrickPi which gives even more weird results:

```
$ echo CAL >/sys/class/lego-sensor/sensor0/mode; \
> cat /sys/class/lego-sensor/sensor0/value[0-7]; \
> echo; \
> echo RAW >/sys/class/lego-sensor/sensor0/mode; \
> cat /sys/class/lego-sensor/sensor0/value[0-7]
0
0
0
0
100
100
100
100

2
25696
25700
0
2
0
0
0
$ echo CAL >/sys/class/lego-sensor/sensor0/mode; \
> cat /sys/class/lego-sensor/sensor0/value[0-7]; \
> echo; \
> echo RAW >/sys/class/lego-sensor/sensor0/mode; \
> cat /sys/class/lego-sensor/sensor0/value[0-7]
100
100
100
100
0
0
0
0

2
25696
12
0
2
0
0
0
```

dlech, 2016-05-17:
It looks like the data type does not match the scaling (mismatch of signed vs. unsigned 16-bit). https://github.com/ev3dev/lego-linux-drivers/blob/master/sensors/nxt_i2c_sensor_defs.c#L2092

Also, the question marks in the value range lead me to believe that this mode was never fully tested.

dlech, 2016-05-17:
I'm guessing that this is a 10 (or 12) bit analog value. Could be an endianness problem too.

Try looking at the raw data using `i2cdump -y -f -r 0x6A-0x7A 6 0x0A`

Also, make sure you aren't mixing up LineLeader and LightArray, these are two different sensors.

Also also, it could be that the raw data address has moved with different firmware versions of the sensor.

dlech, 2016-05-17:
Link on mindsensors.com website currently points to the wrong doc.

http://www.mindsensors.com/pdfs/LightSensorArray-User-Guide.pdf

kortschak, 2016-05-17:
>Also, the question marks in the value range lead me to believe that this mode was never fully tested.

OK. I was reading that as "the max value is from an unscaled range".

>Link on mindsensors.com website currently points to the wrong doc.

Yeah, I noticed that. I've been thinking about sending a note to them but they haven't yet responded to comments about using http for pages holding customer-identifying data, so I'm not sure it's worth while. They also have a different address for the sensor in that documents, 0x14, which does not appear to be correct.

```
$ i2cdump -y -f 3 0x0A
No size specified (using byte-data access)
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
00: 56 32 2e 33 30 00 00 00 6d 6e 64 73 6e 73 72 73    V2.30...mndsnsrs
10: 4c 53 41 72 72 61 79 00 53 70 31 20 30 33 00 00    LSArray.Sp1 03..
20: 53 65 70 20 31 32 20 32 30 31 33 00 31 36 46 31    Sep 12 2013.16F1
30: 39 33 38 00 00 00 00 00 00 00 00 00 00 00 00 00    938.............
40: 00 99 64 64 64 64 64 64 64 64 41 35 4e 54 4b 53    .?ddddddddA5NTKS
50: 41 4a 3d 33 4a 50 49 51 3f 44 45 39 53 57 4e 56    AJ=3JPIQ?DE9SWNV
60: 42 54 3a 30 46 4d 46 4e 3e 3b 45 00 39 00 53 00    BT:0FMFN>;E.9.S.
70: 57 00 4e 00 56 00 42 00 54 00 00 00 00 01 01 01    W.N.V.B.T....???
80: 01 01 01 01 00 00 00 00 00 00 00 00 00 00 00 00    ????............
90: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
c0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
e0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
f0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
```

From what I can see the 0x6a-0x79 range always has the second bytes set to zero.

dlech, 2016-05-17:
> They also have a different address for the sensor in that documents, 0x14, which does not appear to be correct.

It's correct in that 0x14 is what you use in other programming environments. See http://www.ev3dev.org/docs/sensors/i2c-sensor-addressing/

dlech, 2016-05-17:
For comparison, here is what I have:

```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
00: 56 32 2e 32 30 00 00 00 6d 6e 64 73 6e 73 72 73    V2.20...mndsnsrs
10: 4c 53 41 72 72 61 79 00 53 70 30 20 30 33 00 00    LSArray.Sp0 03..
20: 53 65 70 20 30 34 20 32 30 31 33 00 31 36 46 31    Sep 04 2013.16F1
30: 39 33 38 00 00 00 00 00 00 00 00 00 00 00 00 00    938.............
40: 00 00 00 00 00 00 00 00 00 00 42 4f 66 55 57 58    ..........BOfUWX
50: 67 51 36 43 58 47 49 4a 57 45 55 67 82 6d 6f 72    gQ6CXGIJWEUg?mor
60: 82 68 23 2c 3d 30 32 30 3c 2e 23 00 2c 00 3d 00    ?h#,=020<.#.,.=.
70: 30 00 32 00 30 00 3c 00 2e 00 00 00 00 01 01 01    0.2.0.<......???
80: 01 01 01 01 00 00 00 00 00 00 00 00 00 00 00 00    ????............
90: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
c0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
e0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
f0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
```

dlech, 2016-05-17:
It looks like your firmware is slightly new than mind and that we both upgraded firmware for EV3 compatibility, judging by the date. I wonder if they broke the raw data readings when they implemented EV3 compatibility.

kortschak, 2016-05-17:
>It's correct in that 0x14 is what you use in other programming environments. See http://www.ev3dev.org/docs/sensors/i2c-sensor-addressing/

Ah, yes.

>I wonder if they broke the raw data readings when they implemented EV3 compatibility.

Do they have a bug reporting mechanism?

dlech, 2016-05-18:
I've been playing with this some more and and mine seems to be working as expected, so perhaps the firmware version does have something to do with it.

I think support@mindsensors.com is best. I think they have been very busy lately with the FIRST competitions. They haven't responded to my last couple inquires yet either.

kortschak, 2016-05-18:
I've sent an email linking to this issue.

micgil, 2016-05-18:
> No black or white calibration was done prior to the following.

You have to calibrate the device to compensate for changes in distance and ambient light or your values will be wrong.
0x14 is the 8-bit address, 0x0A is the 7-bit address.
The link on the product page has now been corrected.

dlech, 2016-05-18:
> You have to calibrate the device to compensate for changes in distance and ambient light or your values will be wrong.

We are not talking about the calibrated values here. They are working. We are talking about the "raw voltage" values at register 0x6A-0x79. They way I understand it, these values should not be affected by calibration.

They are working for me (fw v2.20) but not for @kortschak (fw v2.30).

kortschak, 2016-05-18:
@micgil as @dlech says, the issue here is the raw values which AFAICS should not be affected by calibration, and the reproducers above were designed to show that where calibrated reading works, raw readings do not.

But to address it directly, here is the result of doing calibration:

```
$ echo CAL >/sys/class/lego-sensor/sensor0/mode
$ echo CAL-WHITE >/sys/class/lego-sensor/sensor0/command
$ echo CAL-BLACK >/sys/class/lego-sensor/sensor0/command
$ cat /sys/class/lego-sensor/sensor0/value[0-7]; \
> echo; \
> echo RAW >/sys/class/lego-sensor/sensor0/mode; \
> cat /sys/class/lego-sensor/sensor0/value[0-7]
92
89
93
84
12
0
0
0

40
34
49
52
45
46
40
37
```

You can see we have exactly the same effect (in fact calibration appears to worsen the calibrated readings in as much as they now never reach the maximum value - this is not an issue).

After having done this (just now), the behaviour of the sensor has now reverted to the expected and I can no longer repeat the behaviour in the report even over power cycling or reboots. The only think that I can think is that the sensor somehow got into a state that has now resolved `</hand-waving>`. However I have now found another issue that may be worth looking into.

The calibrated values appear to latch for an unreasonable amount of time. To illustrate we take two measurements, the first has one half black and the other half right, the second has the colour orientation swapped, reading after ~5 seconds.

```
$ echo CAL >/sys/class/lego-sensor/sensor2/mode; cat /sys/class/lego-sensor/sensor2/value[0-7]; echo; echo RAW >/sys/class/lego-sensor/sensor2/mode; cat /sys/class/lego-sensor/sensor2/value[0-7]
0
0
0
0
34
89
92
96

40
35
49
52
70
152
129
137
$ echo CAL >/sys/class/lego-sensor/sensor2/mode; cat /sys/class/lego-sensor/sensor2/value[0-7]; echo; echo RAW >/sys/class/lego-sensor/sensor2/mode; cat /sys/class/lego-sensor/sensor2/value[0-7]
0
0
0
0
25
90
92
96

135
114
160
171
67
46
40
37
```

What you notice is that the readings of the two orientations are approximately the same for calibrated, but not for raw which behave (now!) as expected.

kortschak, 2016-05-18:
@dlech The above was done on an EV3. When I get a chance I'll repeat with a BrickPi to see if that can recover the incorrect behaviour since it gave very strange results. Maybe the weird state was caused by the BrickPi?

dlech, 2016-05-18:
> Maybe the weird state was caused by the BrickPi?

I suppose it is possible. I would think any bad state would be reset by removing power from the sensor though.

kortschak, 2016-05-18:
I would have thought so too, but I need to check to nail it down.

kortschak, 2016-05-19:
So, I've been able to get the sensor to reproduce the behaviour on the EV3 that is in the original comment though not after using it with a BrickPi and I don't know what exactly brought it back - and now depowering it *does* fix the issue.

In addition to the issues already mentioned, I have found something that I suspect is due to either BrickPi firmware or hardware. The sensor cannot accept commands or report available commands (making calibration impossible), and the raw values are not meaningful at all - this has not changed since the original report.

```
$ echo nxt-i2c >/sys/class/lego-port/port0/mode
$ echo "ms-light-array 0x0a" >/sys/class/lego-port/port0/set_device 
$ cat /sys/class/lego-sensor/sensor0/address 
ttyAMA0:S1:i2c10
$ cat /sys/class/lego-sensor/sensor0/mode
CAL
$ echo CAL-WHITE >/sys/class/lego-sensor/sensor0/command
-bash: echo: write error: Invalid argument
$ cat /sys/class/lego-sensor/sensor0/commands
cat: /sys/class/lego-sensor/sensor0/commands: Operation not supported
$ cat /sys/class/lego-sensor/sensor0/value[0-7]
75
75
76
58
0
0
0
0
$ echo RAW >/sys/class/lego-sensor/sensor0/mode
$ cat /sys/class/lego-sensor/sensor0/value[0-7]
2
0
0
0
2
0
0
0
```
Note that the absence of high values in the 1-3 values for RAW don't always happen. There are never non-zero values in the values 5-7.

I'm unable to get an i2c dump on the BrickPi for this sensor to see what is going on here.

```
$ i2cdump -y -f 3 0x0A
No size specified (using byte-data access)
Error: Could not open file `/dev/i2c-3' or `/dev/i2c/3': No such file or directory
$ echo nxt-i2c >/sys/class/lego-port/port0/mode
$ echo "ms-light-array 0x0a" >/sys/class/lego-port/port0/set_device 
$ i2cdump -y -f 3 0x0A
No size specified (using byte-data access)
Error: Could not open file `/dev/i2c-3' or `/dev/i2c/3': No such file or directory
$ cat /sys/class/lego-port/port0/address 
ttyAMA0:S1
$ cat /sys/class/lego-sensor/sensor0/address 
ttyAMA0:S1:i2c10
```

dlech, 2016-05-19:
Yes, BrickPi does not have "real" Linux I2C. This is why commands don't work and why you can't use i2c-tools.

See https://github.com/ev3dev/ev3dev/issues/640#issuecomment-219319374

kortschak, 2016-05-19:
Yeah, I figured that was the reason. It seems that it might be worth adding a note to the [Using I2C Sensors](http://www.ev3dev.org/docs/sensors/using-i2c-sensors/) page with caveat about BrickPi I2C.

kortschak, 2016-05-20:
Instead I've added a link to [/docs/drivers/brickpi-i2c-sensor](http://www.ev3dev.org/docs/drivers/brickpi-i2c-sensor/) in https://github.com/ev3dev/lego-linux-drivers/pull/38. This allows details of what works/not at that page. Is it just commands and i2c-tools that won't work with their I2C implementation?

dlech, 2016-07-24:
Is this still an issue?

kortschak, 2016-07-24:
I will check over the next couple of days.

dlech, 2017-10-02:
Please re-open if this is still a problem

closed 2017-10-02 by dlech