ev3dev/ev3dev #643: Out of memory when updating package lists

WasabiFan, 2016-05-20:
### System information:

* I am using: LEGO MINDSTORMS EV3

* I installed ev3dev using this image file: ev3-ev3dev-jessie-2015-12-30.img

* My kernel version is: `3.16.7-ckt21-9-ev3dev-ev3`

* My host computer is running: Windows

* I am connecting using: Wi-Fi (Netgear WNA1100 - N150 Wireless USB Adapter)

### About my issue:

#### What led up to the situation?

I wanted to upgrade my kernel, but first I needed to update package lists.

#### What exactly did you do (or not do) that was effective (or ineffective)?

I ran `apt-get update` from a superuser prompt.

#### What was the outcome of this action?

5+ hours later...

```
root@ev3dev:/home/robot# apt-get update
Ign http://ftp.debian.org jessie InRelease
Hit http://ftp.debian.org jessie Release.gpg
Hit http://ev3dev.org jessie InRelease
Hit http://ftp.debian.org jessie Release
Hit http://ev3dev.org jessie/main armel Packages
Hit http://ftp.debian.org jessie/main armel Packages
Hit http://ftp.debian.org jessie/contrib armel Packages
Hit http://ftp.debian.org jessie/non-free armel Packages
Ign http://ev3dev.org jessie/main Translation-en_US
Ign http://ev3dev.org jessie/main Translation-en
Hit http://ftp.debian.org jessie/contrib Translation-en
Hit http://ftp.debian.org jessie/main Translation-en
Hit http://ftp.debian.org jessie/non-free Translation-en
Reading package lists... Error!
Segmentation fault
```

#### What outcome did you expect instead?

The update operation to finish without hitting a segfault.

#### My notes

My guess is that this is an issue of running out of memory. If it is of any use:

```
root@ev3dev:/home/robot# swapon -s
Filename                                Type            Size    Used    Priority
/dev/zram0                              partition       65532   65532   16383
root@ev3dev:/home/robot# free
             total       used       free     shared    buffers     cached
Mem:         59436      57820       1616         56        444       9796
-/+ buffers/cache:      47580      11856
Swap:        65532      65532          0
```

dlech, 2016-05-20:
https://github.com/ev3dev/ev3dev/blob/master/release-notes/ev3-ev3dev-jessie-2015-12-30.img-release-notes.md#known-issues

dlech, 2016-05-20:
Although I guess that doesn't help if you can `apt-get update` first.

WasabiFan, 2016-05-20:
I'm not looking for a fix for the speed, I want to fix the segfault. Are you saying that those hints will work without updating and have the potential to fix the error?

dlech, 2016-05-20:
try #520

dlech, 2016-05-20:
```
fallocate -l 512M /swapfile
chmod 600 /swapfile
mkswap /swapfile
swapon /swapfile
```

WasabiFan, 2016-05-20:
Hmmm... I was pretty sure I had already added a custom swapfile (I know I did at some point). Must've been on another SD card or EV3. Should I try the extra zram thing or an additional swapfile first?

dlech, 2016-05-20:
You have to run `apt-get update` before you can `apt-get upgrade zram-init`, so manual swapfile is pretty much your only option.

WasabiFan, 2016-05-20:
Just added a swapfile and re-started the update. Will report back with the results.

WasabiFan, 2016-05-20:
After just over 10 minutes, it's done! Is this theoretically fixed by that updated package for future releases?

Also, why would adding swap space make the operation faster? In situations like this I'd normally guess that the problem was that is was using slow swap space, but in this case it is the opposite problem... there must be something I'm missing. Is compressed RAM slower than SD swap?

closed 2016-05-20 by WasabiFan

dlech, 2016-05-20:
> Is this theoretically fixed by that updated package for future releases?

yes

> Also, why would adding swap space make the operation faster?

Because you were totally out of memory and it was spending all of it's time looking for nooks and crannies where there were a few bytes of memory free and shuffling things around to try to free up a few bytes of memory so it could continue.

> Is compressed RAM slower than SD swap?

From my unscientific observation, compressed ram is slightly faster.

WasabiFan, 2016-05-20:
> Because you were totally out of memory and it was spending all of it's time looking for nooks and crannies where there were a few bytes of memory free and shuffling things around to try to free up a few bytes of memory so it could continue.

Ah, that makes sense. So it was just shuffling around fragmented memory to get enough space, and at some point it couldn't get any more and segfaulted. That makes sense.