ev3dev/ev3dev #645: How to use infrared mode on EV3 color sensor

g0rp, 2016-05-24:
Image used on EV3 : ev3dev-jessie-2015-02-24.img
Kernel version : 4.4.9-11-ev3dev-ev3

Hi everyone,
Me and my projec group are still rookies at EV3, so we'll try to be as clear as possible.
We have been told that the EV3 color sensor ([http://shop.lego.com/en-US/EV3-Color-Sensor-45506]) was also able to work on infrared mode. A friend of ours succeeded in making the color sensor works as infrared through the Mindstorm EV3 software, so we figured we could, with ev3dev, use this same color sensor as infrared.

We really need to make it work since our ultrasonic sensor is.... tired. So we just want to know if it's indeed possible to set the mode of the color sensor on IR_PROX, or any idea you guys can have. We also want to know, of course, if we should forget about this idea.
Time is limited for the project, we can't offer time to order new things, so we got to do with what we have.

So far, we tried to plug the color sensor on different EV3 inputs (2 and 4 if I remember well) and then we try to "instantiate" this sensor that way :  (sorry i couldn't get the indentation right on Github, but the indentation IS right in my code)

`from ev3dev import core`
`import time`

`iSens = core.InfraredSensor(address='in4') # we also tried with in2`
`while True:`
 `    print iSens.proximity()`
 `    time.sleep(2)`

but we get an attribute error : 
`AttributeError : 'InfraredSensor' object has no attribute _path`

Thanks all for reading, if you have any question I'm available.

Regards

WasabiFan, 2016-05-24:
The sensor that you linked to is a "Color sensor", which is not the same as an "Infrared sensor". If you add `print iSens.connected()` before the loop, my bet is that it will print false, meaning that the code couldn't connect to a device because it couldn't find any Infrared sensors.

So, your problem is that the sensor that you're plugging in is not the same as the sensor that you are instantiating. I am not aware of any infrared distance mode on the normal color sensor, so you should probably check with the person who told you that and confirm that they were indeed talking about the color sensor.

P.S. @rhempel doesn't the python lib check if the sensor is connected before attempting to read from it? From this it looks like it doesn't, so you might want to check that. Statically-typed languages FTW! :wink:

g0rp, 2016-05-24:
Ok WasabiFan,

I'm probably gonna yell at my friend right now for making us believe we could make the color sensor work as infrared !

I'm closing the issue since there is no way to do this. Thanks again for that fast reply !

closed 2016-05-24 by g0rp