ev3dev/ev3dev #646: NXTCam and ev3dev

qboisseau, 2016-05-25:
Image used on EV3 : ev3dev-jessie-2015-02-24.img
Kernel version : 4.4.9-11-ev3dev-ev3

Hi again,

We are trying to make NXTCam v4 work with ev3dev. For now, we calibrated the colours on NXTCam through the dedicated software, and we connected through I2C the NXTCam to the ev3 brick, on the input number 1.

What we achieved for now : 
`from ev3dev import core`
`nxtcam = core.Sensor(address="in1")`
`print nxtcam.value()`

So this works, it prints 0 if nothing is detected, 1 if it is. But can't go further. We want to retrieve the coordinates of the detoured box of the object, we have no idea how to do so...

Any thoughts? Thanks in advance

WasabiFan, 2016-05-25:
See the docs page [here](http://www.ev3dev.org/docs/sensors/mindsensors.com-vision-subsystem-v4-for-nxt-or-ev3/). There is a table that says that there are 6 values that the driver gives you (value0 through value5). If you want to get value 2, for example, which is the upper-left X coordinate, you would call `nxtcam.value(2)`. You do that for each of the four coordinate values to get your points.

dlech, 2016-05-25:
The nxtcam is a very advanced sensor, so not all functionality is available through the lego-sensor interface. If the docs that @WasabiFan linked get you what you were looking for, great. If not, have a look at <http://www.ev3dev.org/docs/sensors/using-i2c-sensors/> for ways to get access to the advanced features.

closed 2019-03-25 by dlech