ev3dev/ev3dev #647: port5 on BrickPi+?

kortschak, 2016-05-25:
The [BrickPi sensor I2C page](http://www.ev3dev.org/docs/drivers/brickpi-i2c-sensor/) mentions port 5. This is available on the BrickPi, but because of the power socket on the BrickPi+, does not appear to be available on that device. Is this the case? If so there should probably be a note on that page that the plus does not have that option.

bmegli, 2016-06-02:
Hi @kortschak,

Port 5 in BrickPi has little to do with BrickPi really. It's just wired to Raspberry Pi I2C interface.

You can see the schema here (this is for BrickPi, not BrickPi+, top middle):

https://github.com/DexterInd/BrickPi/blob/karan259-master/Hardware%20Design/BrickPi%201.7.3.pdf

In BrickPi+  this is still just Raspberry Pi I2C connected to the battery driver (MCP3031), you can see it here (top left):

https://github.com/DexterInd/BrickPi/blob/master/Hardware%20Design/BrickPi_2_8_1.pdf

From what I know there is no ev3dev port driver for port 5 even for BrickPi (not +)

See here: http://www.ev3dev.org/docs/ports/

The brickpi-in-port is the "regular" brickpi port not port 5 (https://github.com/ev3dev/lego-linux-drivers/blob/master/brickpi/brickpi_ports_in.c)

Sensor port 5 in BrickPi is supported by linux I2C driver, you can get to the bus through:

I have successfully tested sensor port 5 with Microinfinity CruizCore using i2c-tools. Note that CruizCore and some other i2c sensors are currently not supported by regular BrickPi sensor port due to missing I2C adapter driver). Also you would not currently get the fancy lego-sensor-class support for port 5.

If you are desparate and need *now* to use some I2C sensor not supported by regular BrickPi+ ports (with optional functions) you can still do it.

I2C is a bus and supports multiple devices. You should be able to solder port 5 back to BrickPi+ and connect additional devices to the bus. You would have to use generic I2C programming interface however not that wonderful one ev3dev has.

To use port 5 in BrickPi I had to lower its baudrate (was too fast for CruizCore which was designed for slow 9600), so it's possible that you have to tweek it for other sensors as well.

I can elaborate futher if somebody needs.

Kind regards

dlech, 2016-06-02:
> Also you would not currently get the fancy lego-sensor-class support for port 5.

Sure you would. Just not automatic detection.

kortschak, 2016-06-02:
Thanks @bmegli, yes that was pretty much what I had assumed - the details are helpful though. The matter is not pressing - I'm just interested in finding out what I can do at the moment.

kortschak, 2016-06-02:
>I can elaborate futher if somebody needs.

It sounds like you are in a good position to send a PR to add some docs ;)

bmegli, 2016-06-03:
@dlech 

>> Also you would not currently get the fancy lego-sensor-class support for port 5.

> Sure you would. Just not automatic detection.

As far as I remember I was unable to do it (use lego-sensor-class on port-5, loading the device for I2C manually). I will check today. 

@kortschak it seems that I don't know enough yet ;-)

bmegli, 2016-06-03:
Ok, as usual @dlech was right.

>> Also you would not currently get the fancy lego-sensor-class support for port 5.
    > Sure you would. Just not automatic detection.

I was testing with ev3dev 4.4.9-11-ev3dev-rpi2 and BrickPi (not BrickPi+) and Microinfinity CruizCore XG1300L I2C sensor. I am using low I2C speed (9600) as recommended in CruizCore docs, this may be different for other sensor.

So to get lego-sensor device instantiated on sensor port 5 in BrickPI (I2C) one has to:

1. Set correct I2C baudrate for the sensor
a) either in boot parameters /boot/flash/config.txt (add `dtparam=i2c_baudrate=9600` at the end)
b) or by unloading and loading i2c_bcm2708 kernel module with params
``` bash
modprobe -r i2c_bcm2708
modprobe i2c_bcm2708 baudrate=9600
```
Note: CruizCore for example doesn't work with default 100 000 baudrate, the lego-sensor class loads but the values are stuck at -1
2. Plug in the sensor - changing the baudrate when the sensor is plugged doesn't usually work for me so change it and then plug the sensor.
3. Load the ev3dev i2c-nxt driver manually
```bash
echo mi-xg1300l 0x01 > /sys/bus/i2c/devices/i2c-1/new_device
```
4. Use the sensor as usual (through /sys/class/lego-sensor/sensorX/... interface)

bmegli, 2016-06-10:
Maybe baudrate should be set by default or maybe there should be commented line in /boot/flash/config like it is for BrickPi PowerSupply

kortschak, 2016-06-10:
>Mention and link here: http://www.ev3dev.org/docs/ports/ (but this is probably not the best place, maybe at the end?)

Yes, end looks better.

But this LGTM

closed 2016-06-10 by kortschak

dlech, 2016-06-11:
> Maybe baudrate should be set by default or maybe there should be commented line in /boot/flash/config like it is for BrickPi PowerSupply

open a new issue or PR to https://github.com/ev3dev/brickstrap/blob/master/projects/ev3dev-jessie/rpi-base/root/boot/flash/config.txt