ev3dev/ev3dev #650: gitter

dlech, 2016-05-26:
Continuing the conversation from #648 where @WasabiFan said:

> Also, while we're on the topic, I think you should consider setting up a Gitter room for ev3dev: it's like IRC but built in the last decade :wink: I like it because it has all the features of a modern chat system in a format that facilitates open discussion, like IRC does. I, for one, would be a lot more involved if we had something like that.

dlech, 2016-05-26:
Anyone interested in using gitter in addition to/instead of irc?

[Try it out](https://gitter.im/ev3dev/chat?utm_source=share-link&utm_medium=link&utm_campaign=share-link)

WasabiFan, 2016-05-26:
Cool! It will be interesting to see if this attracts more people looking for help.

dwalton76, 2016-05-26:
+1 

I did not even realize there was an IRC channel :(

closed 2016-06-17 by dlech