ev3dev/ev3dev #651: EV3 large motor fine control issue

Fabcha44, 2016-05-26:
### System information:

* I am using: *    LEGO MINDSTORMS EV3*

* I installed ev3dev using this image file: *ev3-ev3dev-jessie-2015-12-30.img*

* My kernel version is: *3.16.7-ckt21-9-ev3dev-ev3*

* My host computer is running: *Linux*

* I am connecting using: *Wi-Fi USB dongle*

### About my issue:

#### What led up to the situation?
Trying to have the large motors running at the specified speed, in particular at low speed, to a specific position.

#### What exactly did you do (or not do) that was effective (or ineffective)?
Having being surprise by the outcome of a C program, I wrote the script below in order the behavior of the motors :
```
#/bin/sh
cd /sys/class/tacho-motor
SPEED=$1
POS=$2

echo reset > motor0/command ; echo reset > motor1/command
echo brake > motor1/stop_command ; echo brake > motor0/stop_command
echo ${POS} > motor1/position_sp ; echo ${POS} > motor0/position_sp
echo ${SPEED} > motor0/duty_cycle_sp ; echo ${SPEED} > motor1/duty_cycle_sp
echo run-to-rel-pos > motor1/command ; echo run-to-rel-pos > motor0/command

sleep 0.25

sp0=`cat motor0/speed`
sp1=`cat motor1/speed`

while [ $(( sp0 + sp1 )) -ne "0" ] ; do
	sp0=`cat motor0/speed`
	sp1=`cat motor1/speed`
	echo -e motor0 speed : ${sp0}'\t\tmotor1 speed '${sp1}
done

pos0=`cat motor0/position`
pos1=`cat motor1/position`
echo -e motor0 position : ${pos0}'\t\tmotor1 position '${pos1}

cd -
```
I called the script 3 times, at speed 50, 15 and 10, with a goal of 720 degrees. You can see the result below :
**[robot@robot ~ ]$ ./tst_motor.sh  50 720**
motor0 speed : 292		motor1 speed 290
motor0 speed : 295		motor1 speed 285
motor0 speed : 291		motor1 speed 285
motor0 speed : 298		motor1 speed 286
motor0 speed : 290		motor1 speed 286
motor0 speed : 285		motor1 speed 292
motor0 speed : 287		motor1 speed 287
motor0 speed : 293		motor1 speed 287
motor0 speed : 293		motor1 speed 287
motor0 speed : 296		motor1 speed 296
motor0 speed : 296		motor1 speed 293
motor0 speed : 290		motor1 speed 290
motor0 speed : 293		motor1 speed 286
motor0 speed : 294		motor1 speed 284
motor0 speed : 292		motor1 speed 291
motor0 speed : 290		motor1 speed 285
motor0 speed : 285		motor1 speed 285
motor0 speed : 287		motor1 speed 284
motor0 speed : 292		motor1 speed 284
motor0 speed : 86		motor1 speed 98
motor0 speed : 86		motor1 speed 0
motor0 speed : 0		motor1 speed 0
motor0 position : 731		motor1 position 730
/home/robot
**[robot@robot ~ ]$ ./tst_motor.sh 15 720**
motor0 speed : 69		motor1 speed 59
motor0 speed : 71		motor1 speed 56
motor0 speed : 68		motor1 speed 57
motor0 speed : 67		motor1 speed 56
...
motor0 speed : 65		motor1 speed 56
motor0 speed : 66		motor1 speed 56
motor0 speed : 68		motor1 speed 56
motor0 speed : 68		motor1 speed 55
motor0 speed : 66		motor1 speed 56
motor0 speed : 65		motor1 speed 55
motor0 speed : 61		motor1 speed 54
motor0 speed : 61		motor1 speed 51
motor0 speed : 61		motor1 speed 50
motor0 speed : 63		motor1 speed 52
motor0 speed : 64		motor1 speed 54
motor0 speed : 67		motor1 speed 57
motor0 speed : 65		motor1 speed 51
motor0 speed : 0		motor1 speed 49
motor0 speed : 0		motor1 speed 49
motor0 speed : 0		motor1 speed 48
motor0 speed : 0		motor1 speed 51
motor0 speed : 0		motor1 speed 49
motor0 speed : 0		motor1 speed 49
motor0 speed : 0		motor1 speed 52
motor0 speed : 0		motor1 speed 0
motor0 position : 721		motor1 position 720
/home/robot
**[robot@robot ~ ]$ ./tst_motor.sh 10 720**
motor0 speed : 29		motor1 speed 0
motor0 speed : 29		motor1 speed 0
motor0 speed : 31		motor1 speed 0
motor0 speed : 29		motor1 speed 0
motor0 speed : 29		motor1 speed 0
motor0 speed : 27		motor1 speed 0
motor0 speed : 26		motor1 speed 0
motor0 speed : 28		motor1 speed 0
motor0 speed : 24		motor1 speed 0
motor0 speed : 24		motor1 speed 0
motor0 speed : 22		motor1 speed 0
motor0 speed : 19		motor1 speed 0
motor0 speed : 18		motor1 speed 0
motor0 speed : 18		motor1 speed 0
motor0 speed : 0		motor1 speed 0
motor0 position : 49		motor1 position 1
/home/robot

Even if bash is not very effective, I think the above results show :
1) With the same duty_cycle_sp, the actual speed is not the same for the two motors
2) The brake stop command is not effective, especially at relatively high speed (first run at 50% duty)
3) At low duty (10%), it seems that there is no torque at all.

With the same robot and the LEGO software, the two motors run at the same speed and stop precisely at the requested position, at low duty as well at high duty.

Can you help me understand what's going on?

Beside, I read somewhere that someone was requesting a mean to ask the motors to start together. I have to say that I am eager to see that too, because even with a C program, at relatively low speed, I see my robot starting with a small angle because of the delay on the actual start of the motors).

dwalton76, 2016-05-26:
I am having issues also. I want to rotate 1080 but goes past that by ~120 degrees but position says it rotated 1080 so I can't even back up to correct the overshoot. I'll post more details later.

rhempel, 2016-05-27:
You should be:

- Writing the desired speed to the `speed_sp` not the `duty_cycle_sp`
- Using the `ramp_up_sp` and `ramp_down_sp` (200-500 msec) to give the motor driver a chance to slow the motor
- Using 'hold` as the `stop_command` to encourage the motor to hold to the position setpoint

dlech, 2016-05-27:
also there were significant motor changes in the `10-ev3dev` kernel release. We can't support `9-ev3dev`.

dlech, 2016-05-27:
and yeah, `brake` is passive brake, `hold` is active brake. `hold` in ev3dev ~= brake in EV3-G.

WasabiFan, 2016-05-27:
To explain a bit more, there was recently a kernel release that included a fairly major rewrite of some of the motor drivers. Because ev3dev is still in relatively early stages, the maintainers can't support things like this on older kernels. You are running `-9-ev3dev` while the most recent is `-11-ev3dev`. Be aware of the fact that updating the kernel will mean that most existing code won't work.

Separately, with ev3dev's driver model, `break` does not do anything to actively stop the motor. What it does is essentially attempt to prevent the motor from turning; it does not go backwards, or even attempt to correct for error. `hold`, meanwhile, will use PID to move the motor to the position that you set and keep it there until you send another command. This is essentially what EV3-G does.

The lack of "push back" that you are referring to is likely due to the fact that you are using `duty_cycle`. The duty cycle is just the amount of power being sent to the motor (in essence), so it does not attempt to maintain a specific speed: it just applies power until it hits the target. What I suspect you are looking for is the `speed`: that is the measured speed in tacho counts per second (on some motor models that also means degrees per second). If you use a speed setpoint instead of a duty cycle setpoint, it will actively attempt to maintain the speed that you request. In your kernel, you'll need to set `speed_regulation` to enable regulation and then set `speed_sp` instead of `duty_cycle_sp`. In newer kernels the duty cycle and speed regulation properties have been removed and everything is speed-based.

Fabcha44, 2016-05-27:
Thank you  for your replies.
I have upgraded the kernel and using the speed solved the issue.

closed 2016-05-27 by Fabcha44

WasabiFan, 2016-05-27:
Great! Glad to hear it.