ev3dev/ev3dev #652: Pixy COL-CODE mode

pambrose, 2016-05-27:
### System information:

* I am using: LEGO MINDSTORMS EV3
* I installed ev3dev using this image file: ev3-ev3dev-jessie-2015-12-30.img
* My kernel version is: 4.4.9-11-ev3dev-ev3
* My host computer is running: OSX
* I am connecting using: Wi-Fi

I am having mixed luck using a Pixy(CMUcam5) camera with an EV3 running ev3dev. Single color signature modes are working well, but the color code mode is not. I loaded color codes on the camera with the pixymon app, but all the values are zero after enabling the COL-CODE mode.

I am confused with the way the [ev3dev Pixy doc](http://www.ev3dev.org/docs/sensors/charmed-labs-pixy-cmucam5-for-lego/)  describe the data returned. In the [Pixy doc](http://cmucam.org/projects/cmucam5/wiki/Using_Color_Codes) under “**How color codes are reported**” it says a 2 byte color code number is returned, which encodes a series of octal digits representing the color signatures recognized in the color code match.  In the ev3dev doc describing the Pixy modes, the COL-CODE values includes a COUNT value instead of  the matching octal values. 

So my questions are:
1.  Are you aware of anyone having luck with the COL-CODE mode?
2.  Is the COUNT value in the COL-CODE mode a mistake?

Thanks,
Paul

dlech, 2016-05-27:
> Are you aware of anyone having luck with the COL-CODE mode?

Nope. I'm not aware of anyone using the LEGO compatible version of the Pixy cam with ev3dev at all.

> Is the COUNT value in the COL-CODE mode a mistake?

Quite possibly. I made my best guess based on the docs and source code when I wrote the driver, but I don't actually own the device, so this driver is totally untested.

pambrose, 2016-05-27:
It is an interesting device that makes some pretty cool apps for kids.  Can I send you one via Amazon to assist with your device development?

dlech, 2016-05-27:
Yes, that would be great! You can find my email address on my GitHub profile or PM me on irc or gitter to get my shipping info.

Another thing I forgot to mention is that for advanced uses of the Pixy cam like tracking multiple objects, it is probably best to not use the lego-sensor class device. Here is a [recent irc chat about it](http://ev3dev.org/irclog/#id-47592).

pambrose, 2016-05-27:
Excellent. Thanks.

rhempel, 2016-05-27:
I have one here @dlech - so I might be able to help with testing as well.

dlech, 2016-06-15:
I'm looking into this and it turns out that we have to write the color code we are looking for when reading the data. There are no other sensors that work like this, so we will have to modify the i2c sensor driver to be able to handle.

What this means is that we will have to have a color mode for each combination of color codes. It wouldn't be practical to implement a large number of them. I think the first three sounds reasonable though, so

    COL-CODE-12
    COL-CODE-13
    COL-CODE-23

For advanced uses though, I think we should advocate using the PixyCam via USB. Today, I've just packaged up PixyMon for Ubuntu 16.04, so it shouldn't be too hard to package up libpixyusb and friends too.

Thoughts?

dlech, 2016-06-15:
I'm watching the video now and it is much more complex than that. We might just have to say that color codes must be handled via USB or direct I2C commands rather than the lego-sensor driver.

dlech, 2016-06-16:
I see now what pixy is doing and it is an ugly hack. We could actually make this work, but the color code modes would come with the caveat that they won't work with polling enabled. You would have to set `poll_ms` to `0` and set the mode each time you want to update the value. And we would have to pick which combinations of color codes we would want to support.

For example, with just 2 colors, you can have:

```
12
112
121
122
212
222
...
```

it gets out of hand with too many modes very fast

pambrose, 2016-06-16:
Is the permutation problem addressed on [this wiki page](http://cmucam.org/projects/cmucam5/wiki/Using_Color_Codes): "The smaller numerical representation (123) is always reported for consistency"?

dlech, 2016-06-16:
I have already taken that into account. I'm just not seeing a way to do this without adding a ridiculous number of modes or a special driver feature that is only used by this sensor.

pambrose, 2016-06-17:
One-offs and a ridiculous number of modes sound unappealing.  Can you describe a preferrable approach that would work better with ev3dev that the Charmed Labs guys might consider doing?

dlech, 2016-06-17:
I can't think of a way that would work better with our current driver model. Part of the difficulty comes from the fact that the I2C driver can only be polled - that is the nature of I2C.

I have just packaged and compiled some of the official pixy software for EV3 so that we can try using the USB interface instead of I2C.

    sudo apt-get update
    sudo apt-get install python3-libpixyusb

Then try out the [demo program](https://github.com/ev3dev/pixy/blob/ev3dev-jessie/src/host/libpixyusb_swig/get_blocks.py). You are much less likely to miss blocks this way.