ev3dev/ev3dev #653: Disabling Brickman while app is running

pambrose, 2016-05-27:
I am able to manipulate the EV3 LCD display as described [here](http://www.ev3dev.org/docs/tutorials/using-ev3-lcd/) without a problem.  However, the image written to /dev/fb0 is overwritten by Brickman as it periodically refreshes the display.

Is there any way of disabling Brickman screen refreshes while an app is running?

Thanks

dlech, 2016-05-27:
When you run a program from the file browser in brickman, this is done automatically.

When you run a program remotely, you need to manually switch to a different console. `chvt`, `conspy` and `openvt` are tools for this. Also see [this](https://github.com/ev3dev/grx/wiki/Running-Example-Programs). It's for the GRX graphics library that Brickman is based on, so is mostly applicable.

And just like a Linux desktop, you can plug a keyboard into the EV3 and use <key>ALT</key>+<key>CTRL</key>+<key>Fx</key> to switch between consoles.

dlech, 2017-07-25:
ev3dev stretch has a new command for dealing with this. Run your programs with `conrun` and it will take care of switching consoles for you.

https://github.com/ev3dev/console-runner/blob/ev3dev-stretch/doc/console-runner.rst

closed 2017-07-25 by dlech