ev3dev/ev3dev #654: EVB FATCATLAB sensor support

kortschak, 2016-05-28:
I've just done a survey of FATCATLAB sensors work (none is mentioned in the supported sensors list, so I wasn't expecting great success).

Install from evb-ev3dev-jessie-2016-01-25.img, kernal at 4.4.9-ti-rt-r25-11-ev3dev-bb.org.

Three sensors work out of the box: altitude, gesture and humidity.

Two sensors are not detected: light sensor and IR receiver.

Two sensors appear to crash Brickman: 9DOF sensor and ADC adapter. They can be seen via the sysfs, but only the first device attached will register, subsequent devices fail to load and devices that are removed are not unloaded. The device cannot be shutdown from the command line after either of these devices has been attached.

The behaviour is pretty much the same when these sensors are attached to an EV3 with the exception that Brickman is not crashed (I need to look at `/sys/class/lego-sensor` in this case - not yet done).

I'm happy to try to pick some of these up, but again I'm likely to need some help/pointers, since the last drivers I worked on were I2C not UART.

dlech, 2016-05-28:
I don't think most people realize this, but this but LEGO did a decent job designing the UART sensors. The sensors send all the information that you need to use the sensors, so you don't have to write a driver for each sensor. You just plug them in and they work.

I'm guessing that the problems you are seeing are some deviations from the official LEGO firmware. Unfortunately, Andy Lau and his fat cat have been MIA since January 2015 and no one has been able to get their hands on the EVB firmware source code. So, we will have to do some reverse engineering.

This will be easiest if you have a logic analyzer. By attaching it to pins 5 and 6, we can monitor what the sensors is sending and see if it is something the driver knows how to handle or not.

If you don't have a logic analyzers, you can at least check `dmesg` for related messages. You can also do some `printk` debugging, but I have found this troublesome with UART sensors because they receive data much faster than it can be printed out. The driver in question is <https://github.com/ev3dev/lego-linux-drivers/blob/master/sensors/ev3_uart_sensor_ld.c>.

> subsequent devices fail to load and devices that are removed are not unloaded.

I have actually already fixed this problem, but have not released an updated package. It also looks like I forgot to push the change to GitHub and I'm away from my main dev computer for the weekend, so this will have to wait. Ping me on Tuesday if I haven't done anything on this yet.

And finally, when we get these working, we will need to put a big disclaimer on the sensors page on the website that "these sensors do not work with BrickPi, PiStorms or the mindsensors.com EV3 sensor adapter". These 3rd party devices have EV3 sensors hard coded and only know about the four LEGO brand sensors. It is not technically possible to get them working with these devices without a firmware update. These devices use <https://github.com/ev3dev/lego-linux-drivers/blob/master/sensors/ev3_uart_sensor_core.c> and <https://github.com/ev3dev/lego-linux-drivers/blob/master/sensors/ev3_uart_sensor_defs.c>. Do not add the FatcatLab sensors to these files. They will not work.

Instead, we should add them to <https://github.com/ev3dev/lego-linux-drivers/blob/master/sensors/other_sensor_defs.c> or a new `*_defs.c` file to generate documentation for the website.

kortschak, 2016-05-28:
Tangentially related, how do I attach a serial interface to the EVB?

dlech, 2016-05-31:
http://botbench.com/blog/2013/08/15/ev3-creating-console-cable/

dlech, 2016-05-31:
You have to solder on some male (3x1) and female (6x1) header. I recommend low-profile female header. I hacked an old DIP socket to use a female header on mine.

[image]
[image]

dlech, 2016-05-31:
If you don't crash the kernel, `dmesg -w` in a second terminal window works just as well as a serial console.

dlech, 2016-05-31:
> subsequent devices fail to load and devices that are removed are not unloaded.

This is fixed in `evb-systemd 3`. Please upgrade that package.

dlech, 2016-05-31:
> Two sensors appear to crash Brickman

Please open a new issue for this and post the output of `sudo systemctl status -l brickman.service`

kortschak, 2016-05-31:
>If you don't crash the kernel, `dmesg -w` in a second terminal window works just as well as a serial console.

Yes, I think I'll do that for now.

kortschak, 2016-06-03:
```
robot@ev3dev:~$ dpkg -l evb-systemd
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version      Architecture Description
+++-==============-============-============-=================================
ii  evb-systemd    3            armhf        Systemd unit files and udev rules
```

Same outcomes. I'll file another couple of issue.

kortschak, 2016-06-03:
Sorry, it seems to be sporadic now.

dlech, 2016-07-24:
I think we have most of these working now. Can this issue be closed?

kortschak, 2016-07-24:
Yes. Thanks.

closed 2016-07-24 by dlech