ev3dev/ev3dev #655: Stuck at ev3dev booting flash...

chris-xstem, 2016-05-28:
### System information:

* I am using: *LEGO MINDSTORMS EV3*

* I installed ev3dev using this image file: *ev3-ev3dev-jessie-2015-12-30.img*

* My kernel version is: *...DO NOT KNOW...*

* My host computer is running: *OS X*

* I am connecting using: *USB*

### About my issue:

While booting, I can see the "ev3dev booting" flash, however, it stays there... for hours ....

Not sure what I can do to trouble shoot, I have tried these tips:
[image]

But it still stuck at the booting flash screen.

dlech, 2016-05-28:
Did you actually try a different SD card?

The only way to actually figure out where it is getting stuck it so attach a USB serial adapter to input port 1 and look at the output there.

JorgePe, 2016-05-28:
This also happens frequently with one of my EV3, usually the two-button reset doesn't work and I need to take LiPo battery out, then it always boots immediately after.
Since this is a SD card I've been using at least more than an year (apt-get update + upgrade + dist-upgrade) and the other EV3 is OK with a fresh SD card with the latest (December?) image I never cared much, thinking it  maybe just a bug from previous versions.

chris-xstem, 2016-05-29:
OK…

Going to try the "USB serial adapter”.

Chris.Chen @ XSTEM EdFun
4 ~ 6, ZHENHUA ROAD, BAOSHAN DISTRICT, SHANGHAI, CHINA
Tel: +8621.5620.6698 <>
Phone: +86138.1783.4205 <>

dlech, 2016-07-13:
If this is still not working, please try the latest "nightly" build.

[ [image] ](https://bintray.com/ev3dev/nightly/ev3dev-jessie/_latestVersion)

Fuzzyma, 2016-09-01:
@dlech the link you provided is not valid. I would like to try the nightly build. Where can I get it?

dlech, 2016-10-11:
Please re-open if this is still an issue with the latest image.

closed 2016-10-11 by dlech