ev3dev/ev3dev #656: EVB fbdev pixel memory layout

kortschak, 2016-05-28:
I'd like to generalise my fbdev code in [ev3go/ev3dev](https://github.com/ev3go/ev3dev/blob/master/lcd.go) to work with the EVB screen. To this end, can someone point me to the pixel memory layout for the EVB screen memory?

dlech, 2016-05-31:
You can get all of the framebuffer information through `ioctl`s. 

Examples:
* https://github.com/dlech/ev3dev-photo-booth/blob/master/ev3dev-photo-booth.py
* https://github.com/ev3dev/grx/blob/815309e95e89b12ed0f9d924e197a26d7ad2f7e2/src/vdrivers/vd_lnxfb.c24640b/ev3dev/core.py#L2446

dlech, 2016-05-31:
If you must resort to hard-coding (it would be *much* better if you don't - for example, if someone decides to rotate the screen 90 degrees then your library won't work), you can get the info using `fbset`. I don't have my EVB fired up at the moment, but here is the output for the Adafruit display on my BrickPi:

```
fbset -fb /dev/fb1 --info

mode "320x240"
    geometry 320 240 320 240 16
    timings 0 0 0 0 0 0 0
    nonstd 1
    rgba 5/11,6/5,5/0,0/0
endmode

Frame buffer device information:
    Name        : fb_ili9340
    Address     : 0
    Size        : 153600
    Type        : PACKED PIXELS
    Visual      : TRUECOLOR
    XPanStep    : 0
    YPanStep    : 0
    YWrapStep   : 0
    LineLength  : 640
    Accelerator : No
```

From `geometry`, we get:
* width: 320 pixels
* height: 240 pixels
* virtual width: 320 pixels
* virtual height: 240 pixels
* color depth: 16 bits per pixel

from `rgba`, we get:
* red: 5 bits `<< 11`
* green: 6 bits `<< 5`
* blue: 5 bits `<< 0`
* alpha: none

from `LineLength`, we get:
* row offset: 640 bytes

kortschak, 2016-06-01:
Yeah, I've done that and have the pixel packing as described (I get RGB565 as above), but fiddling bits from the command line does not give me the results I expect (writing two bytes affects more than a single pixel). I'll have another play and see if I can figure that out.

kortschak, 2016-06-12:
I've sorted this out, it was bash weirdness - I should just write things properly rather than fiddling in the shell.

closed 2016-06-12 by kortschak