ev3dev/ev3dev #657: proposal: add support for motor waiting to the motor sysfs classes

kortschak, 2016-05-29:
Following from https://github.com/ev3dev/ev3dev/issues/447#issuecomment-222363710

It would be nice if there were a simple language-agnostic way to wait for specific motor states without busywaiting. In https://github.com/ev3dev/ev3dev/issues/447#issuecomment-162963155 @dlech suggested having a set of `*-wait` commands to match each of the existing motor commands. This has the problem that it makes motor command synchronous (yes, this is the intention). Complete synchrony prevents some interesting uses.

An alternative would be to add a new attribute, `wait` which is can have a set of flags written to it from the set of flags returned by `state`, possibly with a negation operator. The write would block until the flag set written is satisfied by the `state`.

So as an example mimicking the case in https://github.com/ev3dev/ev3dev/issues/447#issuecomment-163156678:

Thread 1:
```
echo 1000 >/sys/class/tacho-motor/motor0/time_sp
echo run-timed >/sys/class/tacho-motor/motor0/command
```

Thread 2:
```
echo "running !stalled" >/sys/class/tacho-motor/motor0/wait # Blocking.
# Do something when motor0 is finished running but is not stalled.
```

dlech, 2016-05-31:
I think it would be rather complex to implement a lexical parser in the kernel to handle arbitrary conditions as you are suggesting. I would be more in favor of using `poll()` to be notified when the `state` attribute changes and having the user mode program re-read the state attribute and decide what to do.

rhempel, 2016-05-31:
And back when I had a lot more status changes being reported while I was debugging the driver, that might have been a pain, but I think it makes more sense to report ANY change (as @dlech) suggests and then let the user program figure it out.

I am against adding a blocking command to the the already complex motor bindings - if your language supports `poll()` and a callback thread - that's the right way to do it, if not then a busy wait. It's the language - not the kernel - driver that has the shortcoming, I think

kortschak, 2016-05-31:
>I think it would be rather complex to implement a lexical parser in the kernel to handle arbitrary conditions as you are suggesting.

The syntax I'm suggesting is a `\x20` separated set of strings with possible `!` prefixes. This seems pretty straightforward to me.

`poll(2)` is supported by Go via C binding, but inclusion of C code breaks cross-compilation and so for an EV3 becomes a non-starter. I will file an issue to add Poll to the syscall package which avoids this, though I don't hold out much hope for that addition.

However, the justification for the proposal was to make a language agnostic interface for achieving things that are currently doable with EV3-G, in a straightforward and simple way. Even with the addition of Poll to syscall, other languages will not have the capacity to wait for completion of an operation; in one comment @dlech said that a user should be able to do things even in bash.

rhempel, 2016-05-31:
@dlech might have said that, but I like to think that I'm the driver behind that statement :-) I think the intent was to say that the robot should be programmable in `bash` as a baseline, but that does not mean it's a good idea - precisely for the reason that it's hard to wait for completion

kortschak, 2016-06-01:
Well, I've found that making syscalls does not require C interfacing, so I can do this in my packages without breaking cross-compilation. So, when poll support is added, I'll add API to mine that mimics this from the perspective of the user.

kortschak, 2016-06-11:
I have just added `poll(2)` support to the Go sys package. So this is now an option here.

kortschak, 2016-06-12:
Is there a preference for me to close this and open an issue for `poll(2)` support for motor state, or just rename this issue?

dlech, 2016-06-12:
I would say open a new issue

closed 2016-06-12 by kortschak