ev3dev/ev3dev #658: How to run the ev3dev on Raspberry Pi 3 + Brick Pi?

sportscar, 2016-06-01:
### System information:

* I am using: *RaspberryPi3 + Dexter Industries BrickPi (HW v2.7)*

* I installed ev3dev using this image file: *rpi2-ev3dev-jessie-2015-12-30.img.xz*

* My kernel version is: *I don't know it.*

* My host computer is running: *OS X 10.11.5*

* I am connecting using: *directly, with monitor, mouse and keyboard*

### About my issue:

Dear all, I'm a new guy on ev3dev. I want to run Lejos api on ev3dev.

I download the img file. and using Etcher app to write img to the SD card.

and then I edit config.txt like this:

 

 `####################################`
 `# Dexter Industries BrickPi config #`
 `####################################`

`# uncomment both if you are using BrickPi`
`dtoverlay=brickpi`
`init_uart_clock=32000000`

`# uncomment only if you are using BrickPi+`
`#dtparam=brickpi_battery=okay`

`# uncomment only if you are using Raspberry Pi 3`
`# by default the Bluetooth uses the same UART as BrickPi`
`dtoverlay=pi3-disable-bt`
 

others are keep default value.

after this, I put SD card into raspberryPi and boot it. 

but, after I turn it on, the monitor still dark. and the raspberry pi green light flashing slowly.

please tell me what's wrong in there. Thanks!

JorgePe, 2016-06-01:
The latest image available for download doesn't work with the RPi3 out of the box.
Do you have a Raspberry Pi 2? You can put you the SD card on it, boot, upgrade it to latest version (apt-get update , apt-get upgrade, apt-get dist-upgrade, reboot) and then put the SD card again in the Raspberry Pi 3.

JorgePe, 2016-06-01:
Also see #563

sportscar, 2016-06-01:
thanks JorgePe, I don't have Raspberry Pi 2. So, My question is : Do they have a plan to publish a ev3dev image for Raspberry Pi 3 ready? and when ? 

thanks!

JorgePe, 2016-06-02:
I'm pretty sure the next rpi2 image will also work with the rpi3. You'll have to ask the maintainers about their timings but since there is a milestone for ev3dev 1.0 (https://github.com/ev3dev/ev3dev/milestones) maybe it's just a matter of a couple of months.

sportscar, 2016-06-02:
OK, I see. Thanks JorgePe, I will waiting the next image package. This question can be closed. 
Thanks again.

closed 2016-06-02 by sportscar

JorgePe, 2016-07-27:
@sportscar please test the nightly builds - an image two days old worked OK with my Raspberry Pi 3 and BrickPi. If you want to use onboard bluetooth and onboard wi-fi please see #607

WasabiFan, 2016-07-27:
@sportscar the nightly build can be found here:

  [ [image] ](https://bintray.com/ev3dev/nightly/ev3dev-jessie/_latestVersion)

The one from the 26th is currently the most recent image, and it should be the one that works with the rpi3.