ev3dev/ev3dev #659: Updating packages from brickman

ensonic, 2016-06-01:
Below a proposal for updating a subset of packages from brickman. In ticket #487 having an update item was proposed for the menu already. In OpenRoberta/robertalab-ev3dev#19 I collected some info how we could do it.
As a first step clicking the update item could run the apt-get update for http://ev3dev.org/debian and check how if there are updates. If there are we could offer to install them and reboot. This would only install updates where we don't have to ask questions.

The apt-get update part for ev3dev.org took, when run with time reported:
```
real	0m12.827s
user	0m4.860s
sys	0m3.480s
```
and checking for updates under time reports:
```
real	0m3.842s
user	0m2.03
sys	0m1.060s
```
It is not too bad, I'd say.

We could:
1.) run the installation from brickman and then trigger a reboot, or 
2.) brickman could create a special file and then trigger the reboot and we use a systemd service (using ExecStop=...) to install the updates when shutting down (when the special file exists).

rhempel, 2016-06-01:
For long running processes like this, would it be worth thinking about some kind of progress update or at least an activity  indicator to help the user understand that the brick is not stuck ?

ensonic, 2016-06-01:
It is 15 seconds. We could collect the lines from the command and show the in a text-view like ```tail -f```.

dlech, 2016-06-01:
> and reboot

Most of the time, no reboot is necessary. Only if you install a new kernel package. Or if you have a service that does not automatically restart.

> his would only install updates where we don't have to ask questions.

There are environment variables you can set to suppress any questions.

> http://ev3dev.org/debian

The correct url is <http://archive.ev3dev.org/debian>. `http://ev3dev.org/debian` redirects for now, but it might not always be that way.

> when run with time reported:

I don't understand what is the difference between the two timed operations?

> run the installation from brickman and then trigger a reboot, or

I think we should run installation from brickman, but the reboot should be manual since it is not always needed.

ensonic, 2016-06-02:
Updated the repo url in the first comment. The two ops are 1) apt-get update, 2) a dry-run form of apt-get upgrade to check if there are things to upgrade. I already figured the flags to not ask the user.

dlech, 2016-06-02:
Doesn't `apt-get update` print out how many packages can be upgraded? If you could figure out how to read this, it would save having to call the --dry-run. Also, I wonder if `libapt` would be of any use here. I'm not sure how much of the apt features it exposes.

ensonic, 2016-06-03:
Unfortunately the output from apt-get update ends on
```
Fetched 123.4 kB in 5s (7,89 B/s)
Reading package lists... Done
```

WasabiFan, 2016-06-03:
How does the login logic work in Ubuntu? In Ubuntu, by default when you log in it prints the number of available updates for packages. It takes a few seconds but not too long. Can you get the logic from there? I can't remember where that code is, but I imagine that one of you know :wink:

WasabiFan, 2016-06-03:
The script paths are here, from what I can see:
```
/etc/update-motd.d/90-updates-available
/etc/update-motd.d/91-release-upgrade
```

I don't have access to any machines running Linux of any flavor at the moment, so I can't look deeper. Not sure if this is helpful.

bittner, 2018-01-02:
@WasabiFan The contents of `/etc/update-motd.d/90-updates-available` is:
```bash
#!/bin/sh

stamp="/var/lib/update-notifier/updates-available"

[ ! -r "$stamp" ] || cat "$stamp"
```
... and `/var/lib/update-notifier/updates-available` doesn't usually exist (`-r` tests for "readable"). Looks like that file is created by some service or cron job (probably `update-notifier` is involved).

From my point of view, running something like follows should do the trick with acceptable performance:
```bash
PKG_UPGRADE_COUNT=$(\
sudo apt-get update -qq && \
sudo apt-get upgrade --dry-run --yes | tail -n 1 | \
sed -e 's/.* to remove and //' -e 's/ not upgraded.//')
```
During the execution an hour glass may be shown, because it indeed _does_ take a few seconds. `$PKG_UPGRADE_COUNT` will contain the number of packages that can be upgraded in the end.