ev3dev/ev3dev #660: Use Etcher for flashing SD cards

dlech, 2016-06-01:
It looks like there's a new flashing tool: <http://www.etcher.io/>.

It's cross platform, so we can simplify getting started significantly.

dlech, 2016-06-01:
Well, I just tried this on Ubuntu 16.04 and it caused my system to be unresponsive as long as my SD card was attached, so maybe it's a bit too beta at this point.

WasabiFan, 2016-06-01:
Wow, it would be great to get that working! I'll try this on my Windows PC at some point and see how it goes. To be clear, it was your EV3 that froze, right? Not the PC?

dlech, 2016-06-01:
Nope, it was the PC. It eventually go unstuck without rebooting though.

WasabiFan, 2016-06-01:
Oh, wow... That's _interesting_. I'd expect that that won't happen on other platforms, then. You should probably open an issue on their repo and see if they have any thoughts on the matter. Although, more likely is it's an OS bug that the imager just triggered.

WasabiFan, 2016-06-01:
Ah, I see -- you already opened an issue 👍

WasabiFan, 2016-06-01:
I just tried it on Windows and my EV3 seems to be functional, so it worked end-to-end. _As a sidenote, it failed to image the card the first time, but when I hit the button to try again it worked. Not sure if that is repeatable._ Definitely a better experience overall than Win32DiskImager!

WasabiFan, 2016-06-07:
Now that the freeze issue has been (at least for now) sorted out, is there a reason that you think we shouldn't go ahead and add a cross-platform tutorial for Etcher? I'd volunteer to do it if we want it.

I'm thinking it should work to keep the existing tutorials, and then in the main guide recommend the Etcher guide with a link to the others if the user can't use it for some reason. An alert on the platform-specific pages pointing out the Etcher option might also be in order.

dlech, 2016-06-07:
Go for it. :smiley:

rhempel, 2016-06-07:
Yes, go for it!

WasabiFan, 2016-06-07:
This seems to be going well.

[image]

dlech, 2016-06-08:
You are trying to be too fancy. :wink:

closed 2016-06-15 by dlech

rhempel, 2016-06-15:
I got a fail (sd card failed to validate) the first time I tried using Etcher (on Debian Jessie with XFCE as Desktop) as well - but the card was fine. I removed and reinserted it, and the partition labels showed up in the file manager, so I removed the card and put it in the EV3 - booted fine.

I'm really liking Etcher - and the idea of havig an application that's really a self-contained VM/appliance is kind of awesome.