ev3dev/ev3dev #661: Motor runs past position_sp when using run-to-abs-pos with ramp_down_sp

dwalton76, 2016-06-03:
### System information:

* I am using: LEGO MINDSTORMS EV3

* I installed ev3dev using this image file: ev3-ev3dev-jessie-2015-12-30.img

* My kernel version is: 4.4.9-11-ev3dev-ev3

* My host computer is running: Linux

* I am connecting using: wifi

### About my issue:

The following causes the motor to run for forever
```
echo reset > /sys/class/tacho-motor/motor0/command
echo hold > /sys/class/tacho-motor/motor0/stop_action
echo 300 > /sys/class/tacho-motor/motor0/ramp_down_sp
echo -400 > /sys/class/tacho-motor/motor0/speed_sp
echo -288 > /sys/class/tacho-motor/motor0/position_sp
echo run-to-abs-pos > /sys/class/tacho-motor/motor0/command
```
I tried various values for ramp_down_sp but all with the same result.

ramp_up_sp is fine.

rhempel, 2016-06-03:
Grrr - what happens when you set the `speed_sp` to a positive value? The `speed_sp` should be interpreted as an absolute value so that it does not matter if the `position_sp` is in the positive or negative direction from the current position.
If this is a failing case, I need to add it to the test suite...

dwalton76, 2016-06-03:
Same behavior if speed_sp is positive

dwalton76, 2016-06-03:
I'll double check tomorrow but I am almost certain I saw the motor turning in the wrong direction which is what prompted me to use a negative speed_sp

rhempel, 2016-06-03:
That is very strange - I am pretty sure that my test suite has `run-to-abs-pos` in it - I'll have to look at it as well.

dwalton76, 2016-06-03:
ok the motor does turn in the wrong direction when ramp_down_sp is used, so that explains why it runs for forever because we never get to position_sp.

```

dwalton76, 2016-06-03:
ok there are two problems when using ramp_down_sp
- the motor always moves forward even when position_sp is less than position
- if position_sp is greater than position (so the motor is moving in the right direction) the motor still does not stop at position_sp and runs for forever. Example:

``` 
robot@ev3dev[ev3dev-lang-python]# echo reset > /sys/class/tacho-motor/motor2/command
robot@ev3dev[ev3dev-lang-python]# echo hold > /sys/class/tacho-motor/motor2/stop_action
robot@ev3dev[ev3dev-lang-python]# echo 300 > /sys/class/tacho-motor/motor2/ramp_down_sp
robot@ev3dev[ev3dev-lang-python]# echo 400 > /sys/class/tacho-motor/motor2/speed_sp
robot@ev3dev[ev3dev-lang-python]# echo 288 > /sys/class/tacho-motor/motor2/position_sp
robot@ev3dev[ev3dev-lang-python]# echo run-to-abs-pos > /sys/class/tacho-motor/motor2/command
robot@ev3dev[ev3dev-lang-python]# 
robot@ev3dev[ev3dev-lang-python]# ./utils/stop_all_motors.py 
robot@ev3dev[ev3dev-lang-python]# cat /sys/class/tacho-motor/motor2/position 
4540
robot@ev3dev[ev3dev-lang-python]# 
```

dlech, 2016-06-03:
> I am pretty sure that my test suite has run-to-abs-pos in it

It does, but not with ramping, which I think is what is making the difference.

rhempel, 2016-06-03:
Well, there's always something that's easy to miss - I'll work on this over the weekend.

dwalton76, 2016-06-03:
Where does the test code live?

dwalton76, 2016-06-11:
```
 660         /* Calculate where we would end up if we started a
 661          * linear ramp down right now, the 1 prevents the
 662          * value from being 0 and causing a division problem
 663          */
 664 
 665         rampdown_time = (abs(ev3_tm->speed) * ev3_tm->tm.active_params.ramp_down_sp)
 666                                 / (1 + ev3_tm->tm.info->max_speed);
 667 
 668         rampdown_endpoint = ev3_tm->position
 669                           + ((ev3_tm->speed * rampdown_time) / (2*MSEC_PER_SEC));
 670 
 671         new_speed_sp = (2*MSEC_PER_SEC * (ev3_tm->position_sp - ev3_tm->position)
 672                                 / (1 + rampdown_time));
```
I am trying to get my head wrapped around how the ramping works.  On lines 665-666
- ramp_down_sp is in ms so this should be divided by 1000 here no?
- can max_speed ever be 0?  Adding 1 is a good safety net but is also introducing a small amount of error.

Lines 668-669...why divide by 2000ms here?  I could see 1000ms to make up for the missing /1000 on line 665 but cannot figure out why 2s.  Same basic question for line 671.  Is seems to calculate how far the motor will go in 1/2 of the rampdown_time.

dlech, 2016-06-11:
> ramp_down_sp is in ms so this should be divided by 1000 here no?

Kernel doesn't do floating point, so if we divided it by 1000, it would be rounded to the nearest second.

> can max_speed ever be 0? Adding 1 is a good safety net but is also introducing a small amount of error.

Only if there is a bad driver. In other words, not practically. I think adding 1 is not necessary.

dwalton76, 2016-06-11:
Ack.  How about the 2s thing...what is the significance there?

dlech, 2016-06-11:
Not sure on that one. @rhempel?

rhempel, 2016-06-11:
Yes, there is a significance, or I would not have put it in there :-) We'll need a very small bit of geometry/calculus to figure this out.

Assume that we have a constant speed of 500 ticks per second - the motor will travel 500 ticks in that second, yes? You can verify this graphically by drawing a speed/time graph with a straight horizontal line at the 500 ticks per second - the area under the line (for 1 second) is 500 ticks.

Now, think about what happens if we ramp down (linearly) from the current speed, how far will the motor travel in 1 second? On the same speed/time graph draw a line where the speed ramps down from 500 to 0 in one second. See how it cuts that area in half?

That's where the 2 comes from - and here's how the driver figures out when to start ramping down...

If I'm currently running at 500 ticks per second, I can calculate how far I will travel if ramp down to 0 in the `ramp_down_sp` time - it's half as far as if I was not ramping down at all, right?

So when my calculated end point (given a current `speed`, `ramp_down_sp`, and `position`) is just past my designated `position_sp` I need to start ramping down.

I probably blew a comparison operator somewhere...

rhempel, 2016-06-11:
And I'm just building the current kernel source right now, so we'll probably end up at the same solution

dwalton76, 2016-06-11:
Ah makes sense now...we should put your post in the comments to explain the 2s. 

I am going to a party for the evening so odds are you will get to the solution before I do :)

rhempel, 2016-06-12:
Doing `printk()` this frequently can lead to Heisenbugs :-) I usually do it every 5, 10, or 20 times through the motor control handler.

It looks to me like the `rampdown_endpoint` calculation is incorrect because the position is going in the wrong direction.

rhempel, 2016-06-13:
It's the 'new_speed_sp' calculation, I think. Should be ratio of calculated ramp down time to ramp down setpoint?

dwalton76, 2016-06-14:
rampdown_time - shouldn't the denominator here be speed_sp? So before we have started ramping down our speed should be speed_sp so we have ramp_down_sp milliseconds to ramp down.  Once we have started ramping down though assume we are at 50% of speed_sp so we will know that we have 50% of ramp_down_sp ms left.  speed_sp just seems like a better fit here than max_speed.

rhempel, 2016-06-14:
No, the rampdown time calculation is OK, it's the ratio of the current speed to the max speed, because the rampdown_sp is how long it would take to go from max speed to 0. That keeps the rampdown slope constant regardless of your speed at the time you start ramping.

I think there's a more serious issue with the refactored drivers - still poking at it with a stick to figure out where to start looking.

dwalton76, 2016-06-14:
That seems really confusing for the end user....if I set the ramp_down_sp to 400ms then whether or not it actually takes 400ms depends on my speed_sp? The slope would still be constant with speed_sp as the denominator it would just a different slope depending on your speed_sp.  Either way I don't think this is a factor with the issue we are seeing just something I noticed while reading the code.

Some more debugs....with this:
```
                printk("position %d/%d, speed %d/%d, rampdown_time %d, rampdown_endpoint %d, setpoint %d",
                       ev3_tm->position, ev3_tm->position_sp, ev3_tm->speed, ev3_tm->tm.active_params.speed_sp,
                       rampdown_time, rampdown_endpoint, ev3_tm->speed_pid.setpoint);
```

I am doing

- reset the motor
- run_to_abs_pos() position_sp 1080 with speed_sp 200

I see the following but printk only seems to dump the "position" lines after I've stopped the motor...I've tried only calling printk when position a factor of 10 but there is still something odd happening here with output not being displayed.

One thing odd here is that I reset the motor so position is 0 and position_sp is 1080 why is the speed_pid.setpoint being set to -112?  The motor is turning in the correct direction but with a negative setpoint we are going to look for rampdown_endpoint <= ev3_tm->position_sp before we start ramping down and will look for ev3_tm->position <= ev3_tm->position_sp in order to stop but that will never happen so we run for forever.
```
legoev3_motor_run_regulated setpoint 200 
legoev3_motor_run_to_pos position_sp 1080, speed_pid.setpoint -112
position 1888/1080, speed 189/0, rampdown_time 90, rampdown_endpoint 1896, setpoint -112
position 1889/1080, speed 222/0, rampdown_time 105, rampdown_endpoint 1900, setpoint -112
position 1890/1080, speed 223/0, rampdown_time 106, rampdown_endpoint 1901, setpoint -112
position 1891/1080, speed 196/0, rampdown_time 93, rampdown_endpoint 1900, setpoint -112
position 1892/1080, speed 185/0, rampdown_time 88, rampdown_endpoint 1900, setpoint -112
position 1893/1080, speed 111/0, rampdown_time 52, rampdown_endpoint 1895, setpoint -112
position 1892/1080, speed 102/0, rampdown_time 48, rampdown_endpoint 1894, setpoint -112
```

rhempel, 2016-06-15:
OK, the `run-to-*-pos` capability is horribly broken right now :-) I've got conditional tracing in the motor driver callback and the `run_to_pos_active` flag is not even set while running.

The code has been refactored (good) but it's broken (bad) and I'm going to have to re-orient myself in the codebase to figure out what's wrong - it's probably at a higher level than the low level motor driver.

rhempel, 2016-06-17:
Well, I think I have a better handle on things now - `run_to_*_pos` works with no ramping, ignores the `ramp_up_sp` and gets completely flustered if there's a `ramp_down_sp` - so a fix will be in the works in the next few days.

rhempel, 2016-06-25:
I believe that this is now fixed with [https://github.com/ev3dev/lego-linux-drivers/pull/49] - turns out it was an order of operations thing ... integer math involving division in C can be the source of many mysterious bugs that in hindight leave obvious clues.

If you have the capability to build and test kernel modules, please give these a spin.

@dlech - for workflow on submodules, is what I'm doing in [https://github.com/ev3dev/lego-linux-drivers/pull/49] correct? I think we get the PR merged into `lego-linux-drivers` then you update the `ev3-kernel` repo with the hash for the driver submodule, yes?

dlech, 2016-06-26:
> for workflow on submodules, is what I'm doing in [ev3dev/lego-linux-drivers#49] correct?

yes

> I think we get the PR merged into lego-linux-drivers then you update the ev3-kernel repo with the hash for the driver submodule, yes?

The submodule gets updated eventually. I don't update it for every commit to `lego-linux-drivers`

dwalton76, 2016-07-04:
@rhempel I updated my repo and see your fix.  I built/installed a new kernel but still see the same issue.  Is there a special build step I need to take to make it pick up your fix in the lego drivers sub repo when I build the kernel? I am following the instructions here:
https://github.com/ev3dev/ev3dev-buildscripts/

dlech, 2016-07-04:
cd drivers/lego
    git pull

dwalton76, 2016-07-04:
Had already done that

rhempel, 2016-07-04:
And you've installed the kernel? ie copied the .deb file after building a "shareable" kernel to the EV3  and then `dpkg -i some-kernel-release-file.deb`

rhempel, 2016-07-04:
I'm just doing a clean rebuild and reinstall here before I get to finishing up the port of the Python binding to Python3

dwalton76, 2016-07-05:
I built via

dwalton76, 2016-07-05:
I also rebooted the ev3 after upgrading the kernel but still see the same behavior.

rhempel, 2016-07-05:
Hmmmm - I am doing a completely clean rebuild now - and reinstalling. One trick I use to be absolutely sure I am getting the right driver is to put a one time `printk()` into the module so that I can ensure I have  a clean build with the right modules via `dmesg`

dlech, 2016-07-05:
@dwalton76, what is `uname -r` after reboot?

dwalton76, 2016-07-05:
```
robot@ev3dev[~]# uname -r
4.4.13-12-ev3dev-ev3-dwalton-ev3+
robot@ev3dev[~]# 
```

rhempel, 2016-07-06:
What is in your `/lib/modules` directory - I am not sure about this, but I had a situation where I built a clean version of the kernel and tried to install over the same `4.4.13-12-ev3dev-ev3-rhempel-ev3+` kernel and it ended up using the OLD kernel and modules.

Maybe try downgrading the kernel to an older version, then upgrading again?

rhempel, 2016-07-06:
Well, this is really wierd. I rebuilt the 4.4.9 kernel from scratch, downgraded my kernel and verified that I was running that kernel with `uname`:
```
Linux ev3dev 4.4.9-11-ev3dev-ev3-rhempel-ev3+ #15 PREEMPT Tue Jul 5 18:22:09 EDT 2016 armv5tejl GNU/Linux```
And then I ran my little test script:
```
port="motor0"

echo brake > /sys/class/tacho-motor/${port}/stop_action
echo 500 > /sys/class/tacho-motor/${port}/ramp_up_sp
echo 500 > /sys/class/tacho-motor/${port}/ramp_down_sp
echo $1 > /sys/class/tacho-motor/${port}/speed_sp
echo $2 > /sys/class/tacho-motor/${port}/position_sp

echo run-to-rel-pos > /sys/class/tacho-motor/${port}/command
```
And it works just fine - I also verified that the has of `drivers/lego` was OK for this version of kernel:
```
commit a99ef965585d7171b78ab353ca25244815d59d25
Author: David Lechner <david@lechnology.com>
Date:   Sat May 7 23:02:06 2016 -0500

    legoev3_bluetooth: don't kfree after devm_kzmalloc
    
    It will crash.
```
So I'm having a lot of trouble figuring out exactly why things were broken at all ...

dlech, 2016-07-06:
> echo brake > /sys/class/tacho-motor/${port}/stop_action

The problem was reported with `stop_action` of `hold`, not `brake`.

rhempel, 2016-07-06:
I know - so I tested with `coast` `brake` and `hold` and they are all working with both scratch built kernels (4.4.9-* and 4.4.13-*)

dwalton76, 2016-07-09:
@rhempel sorry for being slow to respond, /lib/modules/ for me is
```
robot@ev3dev[~]# ls -ltr /lib/modules/
total 24
drwxr-xr-x 4 root root 4096 Dec 30  2015 3.16.7-ckt21-9-ev3dev-ev3
drwxr-xr-x 4 root root 4096 May  7 15:24 3.16.7-ckt26-10-ev3dev-ev3
drwxr-xr-x 4 root root 4096 May 16 00:44 4.4.9-11-ev3dev-ev3
drwxr-xr-x 3 root root 4096 Jun 14 12:40 4.4.9-11-ev3dev-ev3-dwalton-ev3+
drwxr-xr-x 3 root root 4096 Jul  4 23:11 4.4.13-12-ev3dev-ev3+
drwxr-xr-x 3 root root 4096 Jul  5 00:07 4.4.13-12-ev3dev-ev3-dwalton-ev3+
robot@ev3dev[~]# 
```

dwalton76, 2016-07-09:
I am going to purge some of these old kernels and see if that gets me anywhere
```
robot@ev3dev[~]# dpkg --list | grep linux-image
ii  linux-image-3.16.7-ckt21-9-ev3dev-ev3          1                          armel        Linux kernel binary image for version 3.16.7-ckt21-9-ev3dev-ev3
ii  linux-image-3.16.7-ckt26-10-ev3dev-ev3         1                          armel        Linux kernel binary image for version 3.16.7-ckt26-10-ev3dev-ev3
ii  linux-image-4.4.13-12-ev3dev-ev3+              1                          armel        Linux kernel, version 4.4.13-12-ev3dev-ev3+
ii  linux-image-4.4.13-12-ev3dev-ev3-dwalton-ev3+  1                          armel        Linux kernel, version 4.4.13-12-ev3dev-ev3-dwalton-ev3+
ii  linux-image-4.4.9-11-ev3dev-ev3                1                          armel        Linux kernel binary image for version 4.4.9-11-ev3dev-ev3
ii  linux-image-4.4.9-11-ev3dev-ev3-dwalton-ev3+   1                          armel        Linux kernel, version 4.4.9-11-ev3dev-ev3-dwalton-ev3+
ii  linux-image-ev3dev-ev3                         26                         armel        Linux kernel for ev3dev (meta-package)
robot@ev3dev[~]# 
```

dwalton76, 2016-07-09:
While trying to remove older kernels I managed to get my ev3 in a state where neither bluetoot or wifi would work so I re-imaged from ev3-ev3dev-jessie-2015-12-30.img and then upgraded to
```
robot@ev3dev[~]# uname -a
Linux ev3dev 4.4.13-12-ev3dev-ev3 #1 PREEMPT Tue Jun 14 20:26:05 CDT 2016 armv5tejl GNU/Linux
robot@ev3dev[~]# 
```

I am able to reproduce the issue:
```
robot@ev3dev[~]#  echo reset > /sys/class/tacho-motor/motor1/command
robot@ev3dev[~]# echo hold > /sys/class/tacho-motor/motor1/stop_action
robot@ev3dev[~]# echo 300 > /sys/class/tacho-motor/motor1/ramp_down_sp
robot@ev3dev[~]# echo 400 > /sys/class/tacho-motor/motor1/speed_sp
robot@ev3dev[~]# echo -288 > /sys/class/tacho-motor/motor1/position_sp
robot@ev3dev[~]# echo run-to-abs-pos > /sys/class/tacho-motor/motor1/command
robot@ev3dev[~]# 
robot@ev3dev[~]# 
robot@ev3dev[~]# cat /sys/class/tacho-motor/motor1/position
57
robot@ev3dev[~]# cat /sys/class/tacho-motor/motor1/position
809
robot@ev3dev[~]# cat /sys/class/tacho-motor/motor1/position
1069
robot@ev3dev[~]# cat /sys/class/tacho-motor/motor1/position
1308
robot@ev3dev[~]# 
```

@rhempel can you try the exact steps I am trying on your 4.4.13 and see if you see the issue?

dwalton76, 2016-07-10:
I tried to reproduce this on PiStorms but cannot there due to https://github.com/ev3dev/ev3dev/issues/697 we are unable to set the ramp_down_sp on PiStorms

rhempel, 2016-07-13:
Just got back tonight - well, it's a good thing you are persistent.

Any value of `ramp_up_sp` greater than about 160 causes the code you posted above to work "correctly"

At values of `ramp_up_sp` smaller than 150, the motor starts turning in the opposite direction and never recovers.

At values of `ramp_up_sp` greater than 150 but less than 700 or so, the motor starts running backwards very briefly.

At values of `ramp_up_sp` around 1000 the ramping operation is just fine.

So I suspect a defect in the code where I set up the initial ramp where things get messed up for small values of `ramp_up_sp`.

As a workaround, I suggest setting `ramp_up_sp` to a value greater than 500 for now.

Thanks again for being so persistent and providing a great test case to follow.

rhempel, 2016-07-17:
I got a bit further debugging this today and i think I have the solution. There is a race condition in setting `tm->command`. It should be done before ramping starts so that the motor gets the correct command if the ramp up time is very short.
Short ramp up times happen if the `ramp_up_sp` is very small or if the `speed_sp` is very small.
Fix has been committed here: https://github.com/ev3dev/lego-linux-drivers/pull/53

dlech, 2016-07-24:
Fix has been released in the `13-ev3dev` kernels.

closed 2016-07-24 by dlech

dwalton76, 2016-07-25:
Everything passes for me now, thanks!