ev3dev/ev3dev #663: EVB FATCATLAB light sensor and IR receiver not detected

kortschak, 2016-06-03:
### System information:

* I am using: EVB

* I installed ev3dev using this image file: evb-ev3dev-jessie-2016-01-25.img

* My kernel version is: 4.4.9-ti-rt-r25-11-ev3dev-bb.org

* My host computer is running: linux

* I am connecting using: Wi-Fi

### About my issue:

Plug in light sensor or IR receiver. No sensor is detected by Brickman or is visible in `/sys/class/lego-sensor/`

Output of `dmesg` with two cycles of plug-in/remove of IR receiver (same result with light sensor):
```
[   39.955818] lego-port port0: Added new device 'evb-ports:in1:ev3-uart-host'
[   40.221372] Registered EV3 UART sensor line discipline. (29)
[   41.411588] lego-port port0: Removed device 'evb-ports:in1:ev3-uart-host'
[   41.461553] pin1_mv: 4997
[   41.464216] pin6_mv: 58
[   41.466680] pin5_gpio: 1
[   41.469228] pin6_gpio: 0
[   42.647608] lego-port port0: Added new device 'evb-ports:in1:ev3-uart-host'
[   44.981587] lego-port port0: Removed device 'evb-ports:in1:ev3-uart-host'
[   45.031574] pin1_mv: 4997
[   45.034251] pin6_mv: 58
[   45.036714] pin5_gpio: 1
[   45.039264] pin6_gpio: 0
```

dlech, 2016-06-03:
This tells me that automatic detection of the sensors is working. It detects these as EV3/UART sensors.

Relevant comments from <https://github.com/ev3dev/ev3dev/issues/654#issuecomment-222318016>:

> This will be easiest if you have a logic analyzer. By attaching it to pins 5 and 6, we can monitor what the sensors is sending and see if it is something the driver knows how to handle or not.
>
> If you don't have a logic analyzers, you can at least check dmesg for related messages. You can also do some printk debugging, but I have found this troublesome with UART sensors because they receive data much faster than it can be printed out. The driver in question is https://github.com/ev3dev/lego-linux-drivers/blob/master/sensors/ev3_uart_sensor_ld.c.

There are already some debug messages in `ev3_uart_sensor_ld.c` that you can enable by adding `#define DEBUG` to the beginning of the file. Try this and see what it says.

kortschak, 2016-06-19:
Using the same changes you mentioned in https://github.com/ev3dev/ev3dev/issues/680#issuecomment-226521423, I get this.

```
[ 1116.933588] lego-port port0: Added new device 'in1:ev3-uart-host'
[ 1123.355000] lego-port port0: Removed device 'in1:ev3-uart-host'
[ 1139.091460] lego-port port0: Added new device 'in1:ev3-uart-host'
[ 1143.131985] lego-port port0: Removed device 'in1:ev3-uart-host'
```

kortschak, 2016-06-19:
And more informatively, what you have above:

```
[ 1612.275604] Registered EV3 UART sensor line discipline. (29)
[ 1630.527106] lego-port port1: Added new device 'in2:ev3-uart-host'
[ 1630.931828] received: 0x00 (1)
[ 1630.967225] received: 0x00 0x00 0x00 w 0x90 0x80 0x01 0x02 0x05 (9)
[ 1631.014119] received: 0x02 0x05 0x00 0xe9 0x04 (5)
[ 1631.078035] received: 0xe9 0x04 0x00 (3)
[ 1631.617376] received: 0x04 0x00 @ I 0xf6 I 0x00 0x00 0xb6 R (10)
[ 1631.617638] msg_size: 4processing: 0x49 0x00 0x00 0xb6  (4)
[ 1631.617730] chksum:182, actual:182
[ 1631.617757] CMD:1
[ 1631.617779] invalid state: Number of modes is out of range.received: R 0x00 0xe1 0x00 0x00 L 0xa0 0x00 I (9)
[ 1631.684355] received: 0x00 I L L U M I N A N (10)
[ 1631.717937] received: A N C E 0x00 0x00 0x00 0x00 0x00 0x00 (10)
[ 1631.751482] received: 0x00 0x00 0x98 0x01 0x00 0x00 0x00 0x00 0x00 P (10)
[ 1631.784922] received: 0x00 P C G 2 0x98 0x03 0x00 0x00 0x00 (10)
[ 1631.818398] received: 0x00 0x00 0x00 0x00 P C G 0 0x98 0x04 (10)
[ 1631.851962] received: 0x98 0x04 l x 0x00 0x00 0x00 0x00 0x00 0x00 (10)
[ 1631.885675] received: 0x00 0x00 w 0x90 0x80 0x01 0x02 0x05 0x00 0xe9 (10)
[ 1631.924018] received: 0x00 0xe9 0x04 (3)
[ 1631.987729] received: 0xe9 0x04 0x00 (3)
[ 1632.527178] received: 0x04 0x00 @ I 0xf6 I 0x00 0x00 0xb6 R (10)
[ 1632.527442] msg_size: 4processing: 0x49 0x00 0x00 0xb6  (4)
[ 1632.527537] chksum:182, actual:182
[ 1632.527563] CMD:1
[ 1632.527587] invalid state: Number of modes is out of range.received: R 0x00 0xe1 0x00 0x00 L 0xa0 0x00 I (9)
[ 1632.594152] received: 0x00 I L L U M I N A N (10)
[ 1632.627732] received: A N C E 0x00 0x00 0x00 0x00 0x00 0x00 (10)
[ 1632.661263] received: 0x00 0x00 0x98 0x01 0x00 0x00 0x00 0x00 0x00 P (10)
[ 1632.694721] received: 0x00 P C G 2 0x98 0x03 0x00 0x00 0x00 (10)
[ 1632.728190] received: 0x00 0x00 0x00 0x00 P C G 0 0x98 0x04 (10)
[ 1632.761744] received: 0x98 0x04 l x 0x00 0x00 0x00 0x00 0x00 0x00 (10)
[ 1632.795323] received: 0x00 0x00 w 0x90 0x80 0x01 0x02 0x05 0x00 0xe9 (10)
[ 1632.837800] received: 0x00 0xe9 0x04 (3)
[ 1632.897614] received: 0xe9 0x04 0x00 (3)
[ 1633.437094] received: 0x04 0x00 @ I 0xf6 I 0x00 0x00 0xb6 R (10)
[ 1633.437287] msg_size: 4processing: 0x49 0x00 0x00 0xb6  (4)
[ 1633.437381] chksum:182, actual:182
[ 1633.437408] CMD:1
[ 1633.437512] invalid state: Number of modes is out of range.received: R 0x00 0xe1 0x00 0x00 L 0xa0 0x00 I (9)
[ 1633.503998] received: 0x00 I L L U M I N A N (10)
[ 1633.537575] received: A N C E 0x00 0x00 0x00 0x00 0x00 0x00 (10)
[ 1633.571139] received: 0x00 0x00 0x98 0x01 0x00 0x00 0x00 0x00 0x00 P (10)
[ 1633.604573] received: 0x00 P C G 2 0x98 0x03 0x00 0x00 0x00 (10)
[ 1633.638057] received: 0x00 0x00 0x00 0x00 P C G 0 0x98 0x04 (10)
[ 1633.671578] received: 0x98 0x04 l x 0x00 0x00 0x00 0x00 0x00 0x00 (10)
[ 1633.705257] received: 0x00 0x00 w 0x90 0x80 0x01 0x02 0x05 0x00 0xe9 (10)
[ 1633.743714] received: 0x00 0xe9 0x04 (3)
[ 1633.807499] received: 0xe9 0x04 0x00 (3)
[ 1634.060570] lego-port port1: Removed device 'in2:ev3-uart-host'
[ 1646.609653] lego-port port2: Added new device 'in3:ev3-uart-host'
[ 1647.022580] received: 0x08 B 0x81 0x92 0xdd 0xe2 0x12 0xd9 0xca H z 0xf1 0x84 ' 2 0xc2 (16)
[ 1647.167138] received: 2 0xc2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 (18)
[ 1647.283188] received: 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 (18)
[ 1647.398981] received: 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 (18)
[ 1647.514879] received: 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 (18)
[ 1647.630811] received: 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 (18)
[ 1647.746801] received: 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 (18)
[ 1647.863266] received: 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 (18)
[ 1647.978560] received: 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 (18)
[ 1648.584694] received: 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0x00 @ K 0xf4 I (18)
[ 1648.584973] msg_size: 4received: I 0x00 0x00 0xb6 R 0x80 % 0x00 0x00 0x08 0x98 0x00 I R   D A (17)
[ 1648.651885] msg_size: 4processing: 0x49 0x00 0x00 0xb6  (4)
[ 1648.651990] chksum:182, actual:182
[ 1648.652141] CMD:1
[ 1648.652168] invalid state: Number of modes is out of range.received: R 0x80 % 0x00 0x00 0x08 0x98 0x00 I R   D A T A 0x00 L 0x98 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x7f C Z 0x98 (29)
[ 1648.784927] received: Z 0x98 0x03 0x00 0x00 0x00 0x00 0x00 0x00 0x7f C X 0x90 0x80 0x01 0x00 0x03 0x00 (18)
[ 1648.799061] received: 0x03 0x00 0xed (3)
[ 1648.819568] received: 0x00 0xed 0x04 (3)
[ 1648.979379] received: 0xed 0x04 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 (18)
[ 1649.095330] received: 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 (18)
[ 1649.211248] received: 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 (18)
[ 1649.327153] received: 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 (18)
[... cut ...]
[ 1649.559053] received: 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 (18)
[ 1649.674860] received: 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 (18)
[ 1649.790869] received: 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 (18)
[ 1650.277969] lego-port port2: Removed device 'in3:ev3-uart-host'
[ 1650.299854] received: 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0xf2 0x00 (14)
```

dlech, 2016-06-19:
Should be fixed now. Can you confirm?

dlech, 2016-07-24:
Fix has been released in the `13-ev3dev` kernels.

closed 2016-07-24 by dlech