ev3dev/ev3dev #664: Stopped when booting up

MingqiHuang, 2016-06-04:
### System information:

* I am using: LEGO MINDSTORMS EV3

* I installed ev3dev using this image file:ev3-ev3dev-jessie-2015-12-30.img

* My kernel version is: DON'T KNOW.

* My host computer is running: Windows

* I am connecting using: USB

### About my issue:

I am just getting start with ev3dev. But when I am booting the ev3 up, it stopped working. I can see the LEDs flashing, the ultrasonic sesor's LEDs flash from time to time, and it just stays there for more than half an hour.
I connected the lego ultrasonic sensor, lego color sensor, and three motors before I boot it up. I know it will stopped, but when I boot the brick up, connect with the motors and sensors, it stops working again.
I want to ask how to solve this problem and what can I do. Many thanks!

dlech, 2016-06-04:
Duplicate of #623?

MingqiHuang, 2016-06-04:
No, the main problem is, it even stuck when I plug them in after I boot it up. I use the same kernel, I am trying to upgrade the kernel to 4.4.9-11-ev3dev-ev3, if it helps, I will close it.

MingqiHuang, 2016-06-04:
I have tried it. It doesn't work. I pluged the ultrasound sensor, and it worked well, after pluggling the color sensor, it stuck. The PuTTY said Network error: Software caused connection abort. And then I relog in it, it said Connection timeout.

MingqiHuang, 2016-06-05:
The problem seems caused by the ultrasound sensor driver. Now I plug the color, gyro, touch sensor, it works. I also tried to plug the ultrasound sensor only, it also works, but when I plug anouther sensor, it stuck.

dlech, 2016-06-05:
OK, to summarize:

* Boots OK with no sensors plugged in
* No problem with LEGO EV3 Ultrasonic sensor only plugged in
* No problem with other sensors plugged in as long as LEGO EV3 Ultrasonic sensor is not plugged in
* Problem with LEGO EV3 Ultrasonic sensor and one of the other EV3/UART sensors (LEGO EV3 Color, LEGO EV3 Gyro or LEGO EV3 Infrared)

dlech, 2016-06-05:
Can anyone else reproduce this problem?

dlech, 2016-06-05:
If you are able to connect a USB serial adapter to Input Port 1, you may be able to get a stack trace for the kernel crash. 

http://botbench.com/blog/2013/08/15/ev3-creating-console-cable/

MingqiHuang, 2016-06-05:
After updating the kernel, this problem had been solved. But still not able to boot up with sensors plugged in.

dlech, 2016-06-05:
OK, not booting with sensors plugged in sounds like #623. We still haven't figured that one out.

MingqiHuang, 2016-06-05:
Also, I can't connect the EV3 using host name,but IP address works.

dlech, 2016-06-05:
> Also, I can't connect the EV3 using host name,but IP address works.

That is not surprising since you are using windows. If you reboot Windows, it might fix it. Also disabling and reconfiguring Internet Connection Sharing in Windows usually fixes connectivity issues.

MingqiHuang, 2016-06-05:
OK, thanks.

closed 2016-06-05 by MingqiHuang