ev3dev/ev3dev #667: Potentially incorrectly printed value for U32 types in value_show

bmegli, 2016-06-07:
Again, I am not sure if this is the right place to discuss code. But better to have the issue around.

This is of very low prioroty. I know of no sensor that uses datatype LEGO_SENSOR_DATA_U32.

value_show in https://github.com/ev3dev/lego-linux-drivers/blob/master/sensors/lego_sensor_class.c

last line:

```C
return sprintf(buf, "%ld\n", value);
```

will work incorrectly for U32 types > INT_MAX (will print negative values)

Possible fix: e.g. just if condition and `%lu% specifier, this may also depend on scaling, all is not pressing
```C
if(...)
   return sprintf(buf, "%lu\n", value);
else
   return sprintf(buf, "%ld\n", value);
```

dlech, 2016-06-07:
For now, u32 is not allowed for sensor data.

bmegli, 2016-06-08:
Ok, I am closing it then.

dlech, 2016-06-08:
Actually, I think we should remove `LEGO_SENSOR_DATA_TYPE_U32` from `lego_sensor_class.{h,c}` since it is not allowed.

dlech, 2016-11-16:
Fix released in `17-ev3dev` kernel.

closed 2016-11-16 by dlech