ev3dev/ev3dev #668: Help wanted: 1:1 transfer of Mindstorms LabView block program

Zordid, 2016-06-08:
Hi!
I have extreme difficulties to "simply" transfer a simple LabView program to use EV3DEV in Python.
This is the project I need to port: http://www.philohome.com/sort3r/sort3r.htm

The problems mainly come from the different ways to run the motors and the precise positioning.
Within LabView it seems 'easy' and obvious. All blocks use a Power value (I translate this to duty_cycle_sp) - whereas in EV3DEV I must supply a speed which is unknown in the blocks.
But sometimes they use unregulated motor blocks to detect the initial positions.

The result that I get often overshoots - the positions are very unprecise. I use the exact same "coast" or "brake" modes, I use ramp up and down (which I cannot find in LabView at all).

So, are there any valuable hints on how to exactly mimic the behavior of a LabView block program?
Thank you so much! This drives me crazy! :-(

WasabiFan, 2016-06-08:
Although I don’t use LabVIEW, I think I can offer some tips:
- My guess is that “power” in LabVIEW corresponds to “speed” in ev3dev. While the duty cycle is just raw power being applied to the motor, speed is the physical speed that you want the motor to move at. If the motor controlled by the LabVIEW code is attempting to be precise, you can be 90% sure that speed is what you want to translate that to. 
- You are using an old kernel. The newest one hasn’t been released in an image yet, and it has some bugs (there is one nasty one that affects ramping down). Also, newer kernels have partially rewritten motor drivers that change the interface to control them. Be aware that code written for previous kernels will probably not work for the latest one. Some of the big changes include removal of most duty cycle control and removal of the `speed_regulation` properties.  This kernel info is relevant once reading the points below.
- In the newest kernel, there are `max_speed` properties that you can use to get the maximum speed of the motor. This should help you with your issue about value scaling; just use a speed value between 0 and the number returned from that property.
- The “unregulated” blocks that you refer to are probably the ones that use plain duty cycle. In the newer kernels, the primary commands (that move for an amount of time, to a position, etc.) don’t support the duty cycle values because they caused more problems than good. Instead, when you want to send a duty cycle directly, you use the `run_direct` command, which gives you direct control of the duty cycle that updates whenever you change the setpoint.

dlech, 2016-06-08:
Please answer the questions in the issue template rather than deleting them. You have omitted important information, like what kernel are you using?

> In the newest kernel, there are max_speed properties that you can use to get the maximum speed of the motor.

This is not quite the same as the max speed using in the LEGO software though. LEGO uses 900 for the max speed of the large motor and 1200 for the max speed of the medium motor IIRC.

laurensvalk, 2016-06-08:
run-direct mode with duty_cycle_sp in ev3dev is equivalent to the dark blue Unregulated Motor block in EV3-G (The LEGO software). This value corresponds to the percentage of time that the motor gets the maximum voltage. This sets the effective voltage for the motor (between 0 and 100% of the available voltage, which is slightly less than the battery voltage).

All the green blocks in EV3-G do speed control. For the large motor, 1% value means 1.7 rpm. For the medium motor, 1% means 2.67 rpm. 

So if you set a Large motor to, say, 50%, the EV3 is going to do all it takes to get it to move at 50*1.7 = 85 rpm. If something slows it down, it will increase the voltage to attempt to maintain desired speed.

It looks like you also want this in ev3dev. In that case, you can use any of the other modes, such as run-forever or run-to-abs-pos, depending on which mode of the green blocks you want to replicate.

Even though EV3-G is an educational tool, almost all of the terminology associated with speed, torque, force, power, current, or even "current power" is wrong, even in the documentation, so take that with a grain of salt.

Zordid, 2016-06-09:
Thanks for all the answers so far! I wanted this to be tagged as Help-Wanted... Sorry!
(I do use the newest kernel 11)

Laurensvalk, your help with the Power-to-Speed formula might be exactly what will help me!! Thanks! Do you have some information as to how the Lego LabView blocks use ramp up/ramp down to achieve an exact positioning?

The issue I am facing right now is that the given LabView program is 100% exact in the positioning, whereas my replicated Python program fails very often. Overshoots when I try to do a run-to-abs/rel-pos. Or it does not run smooth.

For example in the original LabView program they do the following with three blocks in a loop until this happened 5 times:

LOOP [
1) Set motor to run with 30% power (green block) [-> need to translate that to speed for me]
2) Timer wait for IR-proximity to drop below the value of 4
3) Let the motor run again for exactly 80 degrees, same power. (green block)
] UNTIL 5 cycles

My Python script looks like this:
    move.reset()
    while detected_bins<5:
        move.run_forever(speed_sp=-200)
        while ir.value() >= 4:
            pass
        detected_bins+=1
        move.run_to_rel_pos(speed_sp=-200, position_sp=-80, stop_action='coast')
        while 'running' in move.state: pass
        log.info('Detected %d bin(s)...' % detected_bins)

The original program runs totally smooth, you don't see that anything "happens" until it reaches its final destination.

In my program, you see the coasting phases very easily. The robot "hops" more or less, does not run smooth. I deliberately used coast because I wanted the next loop run's move.run_forever to keep it going smoothly...

Zordid, 2016-06-09:
Laurensvalk, one other question: when applying a power (duty_cycle_sp) to a motor to let it run into a mechanical stop, is it okay to test wether speed is 0 to detect to end? In the LabView programs they actually test "current power" to be 0, which is completely wrong as especially when using regulated blocks, the power will be max when motion is stopped as the motor tries to overcome the obstacle...

laurensvalk, 2016-06-09:
Indeed. In the EV3-G software, "Current power" isn't current, it isn't power, and it is also not the power currently being supplied to the motor.

It's a unit that scales linearly with speed. You could verify this experimentally, or look at the source code. The value is a certain number of encoder ticks per unit of time, which is really a form speed.

You can use the conversion factors I gave above. (For the large motor, 50% "current power" means 50*1.7 = 85 rpm.) 

Checking for 0 is fine. You may also want to consider checking whether it is, say -5 < speed < 5, so you even pass the check if you are nearly zero, which works well in some applications.

Honestly, others might have to fill in which speed controlled ev3dev functions work best for your application. I've only used unregulated speed with ev3dev so far.

dlech, 2016-06-09:
> Do you have some information as to how the Lego LabView blocks use ramp up/ramp down to achieve an exact positioning?

The official LEGO EV3 software does not use ramping.

> Let the motor run again for exactly 80 degrees
> ...
>
>     move.run_to_rel_pos(speed_sp=-200, position_sp=-80, stop_action='coast')

If you want the motor to stop after exactly 80 degrees, you need to use `stop_action='hold'`. If it is set to `coast`, then it will coast right past 80 degrees.

If `hold` is not working like the official LEGO software, then you should open a new issue to fix our drivers. There is already an issue for ramping #661

kortschak, 2016-06-12:
>This is not quite the same as the max speed using in the LEGO software though. LEGO uses 900 for the max speed of the large motor and 1200 for the max speed of the medium motor IIRC.

@dlech Do you have a source for this?

dlech, 2016-06-12:
I don't remember how I calculated those numbers originally, but I imagine that I based them on [this](https://github.com/mindboards/ev3sources/blob/master/lms2012/d_pwm/Linuxmod_AM1808/d_pwm.c#L56).

If we take the numbers from @laurensvalk [here](https://github.com/ev3dev/ev3dev/issues/668#issuecomment-224670770) and convert the units from RPM to pulses per second, I come up with 1020 for the large motor and 1602 for the medium motor.

    [1.7 rev/min] * [360 pulse/rev] / [60 sec/min] * [100 %] = 1020 pulse/sec

dlech, 2016-06-12:
There is also some issue around here that has some graphs of actual motor speed vs. "power" using official LEGO software, but I can't find it right now.

dlech, 2016-06-12:
Ah, [here it is](https://github.com/ev3dev/ev3dev/issues/356#issuecomment-124866705).

kortschak, 2016-06-12:
>I don't remember how I calculated those numbers originally, but I imagine that I based them on this.

Cool. Reading through trying to figure out what their time unit is. The commentary there is woeful

laurensvalk, 2016-06-12:
> uses 900 for the max speed of the large motor and 1200 for the max speed of the medium motor IIRC.

In my opinion, a maximum speed value isn't really meaningful. For example, in the LEGO software you can get, say, a 115% speed if you turn it with your hands. What does that mean?

If you want to use a value like this, I think "rated speed" is the more common engineering term that is found in data sheets. (Usually a speed for prescribed nominal voltage and nominal or no load.)

> Ah, here it is.

Those are very similar to the values I found. For example, taking his 40% speed for large motor: 415 deg/s = 69.16 rpm --> 69.16 / 40  = 1.73. For his 30% you get: 1.67.  (I found 1.7 as mentioned above).

> If we take the numbers from @laurensvalk here 

I've taken a similar approach as above at the time when I investigated this for my book. I let the motors run for several minutes and started the timer after achieving steady speed to average out end effects. I have done so at different speeds and with different (light) loads. I believe I also used an external timer but I'm not entirely sure.

To test the meaning of "Current Power", you can drive the motor at a known speed and display the value, which turns out to correspond exactly to the formula above. If you block it with your hands, you'll see the value go to "0" even though you feel a lot of torque (clearly a nonzero current or electrical power). 

Likewise, if you don't power the motor but you drive it with your hands or another motor, you get a (speed) value too.

kortschak, 2016-06-12:
>In my opinion, a maximum speed value isn't really meaningful. For example, in the LEGO software you can get, say, a 115% speed if you turn it with your hands. What does that mean?

It's meaningful as a reference point. As an example, I'm in the process of translating EV-G programs into programs that run under ev3dev as educational examples. The motors are driven in EV-G as a fraction of their "maximum speed", so to be able to directly compare them I need to know what EV-G thinks the maximum speed is. I believe this is the same reason behind the OP's issue. Essentially its a known reference point for *driven* speed control.

laurensvalk, 2016-06-12:
I understand that it's useful as a reference point, but it is the terminology in the LEGO software that I find problematic. It's not a maximum speed. Hence the suggestion for "rated speed" if it's something we use in the ev3dev motor parameters and documentation.

dlech, 2016-06-12:
> Hence the suggestion for "rated speed"

where were you in <https://github.com/ev3dev/ev3dev/issues/356#issuecomment-118627868>. 😆 

`max_speed` is already in the kernel and we will have to make another breaking change to fix it. (so open a new issue if you think we should fix it).

dlech, 2016-06-12:
> where were you in #356 (comment). 😆

probably studying or something

laurensvalk, 2016-06-12:
>>    where were you in #356 (comment). :laughing:

> probably studying or something

Very likely about something involving torque, speed and power, but likely not current. :)

> (so open a new issue if you think we should fix it).

First I'd like to go over what you have now, as I haven't tried any of the controlled modes recently. That won't be until August though.

WasabiFan, 2016-06-12:
It is also worth pointing out that, as far as I could tell, the `max_speed` is a hard limit; you will get an error if you send a value larger than the max speed. This is a limitation imposed in software though, not hardware.

closed 2019-03-25 by dlech