ev3dev/ev3dev #669: Losing Wifi connection without reconnect - only toggle "powered" helps

Zordid, 2016-06-09:
### System information:
LEGO MINDSTORMS EV3

* I installed ev3dev using this image file: ev3-ev3dev-jessie-2015-12-30.img
* My kernel version is: 4.4.9-11-ev3dev-ev3
* I am connecting using: WiFi using Edimax Dongle EW-7811UN

### About my issue:
When not at home with my setup, I use an Android WiFi Hotspot for my Mindstorms EV3 to connect to the Internet and to communicate with it.
At home, I never had this issue, but using the WiFi hotspot, this is quite common:

- after a while (minutes) I notice that the displayed IP address is gone - the connectivity is lost.
- going into Wireless reveals that there are no Networks displayed whatsoever. 
- the only remidy is to power off the WiFi dongle and repower it, after that the connection is immediately back, as are all available networks.

JorgePe, 2016-06-09:
My first guess is that your Android is entering some kind of power-reducing mode that affects Wi-Fi hotspot connection. Not quite sleep mode since repowering the EV3 Wi-Fi dongle repairs it.
Have you seen your Android power saving settings?

dlech, 2016-06-09:
When we updated to kernel version 4.4.x, we dropped the 3rd party driver for this dongle. Raspberry Pi did the same by had to revert to the 3rd party driver because of complaints. See <https://github.com/raspberrypi/linux/pull/1468>. We probably need to do the same.

JorgePe, 2016-06-09:
But if it is a driver problem wouldn't happen also at home?

bmegli, 2016-06-10:
@Zordid 

Yes, I confirm that one. I also use EW-7811-UN and have similiar issues sometimes.

From memory so I may mess something!

This device was supported by rt8192cu kernel module and patched 8192cu
Usually one had to build 8192cu from sources and blacklist rt8192cu. Additionally some parameters had to be passed to module (disable power saving, etc, but this device only takes 10 mA at peak)

In the past you could tell the difference between packages by diode on EW7811un - patched was blinking, flawed was lit with solid blue.

The package in ev3dev had been doing all of that automatically (blacklisting, passing parameters, and had the patched kernel module)

https://github.com/ev3dev/ev3dev/releases/tag/ev3dev-jessie-2015-12-30 image was also last version  with patched package for 8192cu. 

I don't know what kernel module it is now, but solid blue and with problems and doesn't like parameters like 8192cu.

If you are in hurry you can build patched version from sources. If you are not, you can wait until some nice guy will package it for you...

The good news is that with the right driver (8192cu) connection was rock solid (I kept EV3 on power supply connected for days).

Zordid, 2016-06-13:
Why was that removed / changed?? I don't get it. I am not a kernel specialist able to compile that by myself. I desparately need this to work though. :(

bmegli, 2016-06-13:
It's all in the link that @dlech gave:

> When we updated to kernel version 4.4.x, we dropped the 3rd party driver for this dongle. Raspberry Pi did the same by had to revert to the 3rd party driver because of complaints. See raspberrypi/linux#1468. We probably need to do the same.

For short:
- there is upstream driver that the linux community supports and works on that's supposed to work with most devices
- but it turns out that some devices work better with a different one

And RaspberryPi community (they also use EW-7811UN a lot and other supported by this kernel module) is planning to include both and allow users to choose between them.

Zordid, 2016-06-13:
Unfortunately I don't know exactly what I must do to get this upstream driver to work..
 
Where can I get step by step instructions for doing this?

Up to now, all I did was write the image to sd card and using apt-get to go to the newest kernel...

dlech, 2016-06-15:
I've restored the `rtl8192cu` driver in the [12-ev3dev](http://www.ev3dev.org/news/2016/06/14/Kernel-Release-Cycle-12/) kernel. Please upgrade your kernel and give it a try.

bmegli, 2016-06-16:
I think that for some reason we have ended without kernel module in EV3 tree.

We have it in Raspberry Pi tree.

Also in RPI - working, EV3 - not working

I haven't figured out why it's missing but I mean the result (ko file). I have seen  it selected in kernel config.

dlech, 2016-06-16:
OK, I should have this building now. We will have to wait for the next kernel release though.

bmegli, 2016-06-23:
Kernel built from https://github.com/ev3dev/ev3-kernel/
With latest commit Thu Jun 16 10:34:05 (fixup! ...)

I confirm that:
- module 8192cu is building
- with default module parameters and EW-7811UN I get occasional connection loss but no power on/off is needed
- with added module parameters `options 8192cu rtw_power_mgnt=0 rtw_enusbss=0` the connection is stable (2 hours so far without any problems)

So EW-7811UN with 8192cu needs as usual:

Create file:

``` bash
sudo nano /etc/modprobe.d/8192cu.conf
```
With those lines:
``` bash
# Disable power management
options 8192cu rtw_power_mgnt=0 rtw_enusbss=0
```

dlech, 2016-06-23:
So maybe we should make those options the default in the kernel module so we don't have to have the `.conf` file.

bmegli, 2016-06-24:
@dlech
I see this is also what is in:
https://github.com/pvaret/rtl8192cu-fixes/blob/master/8192cu-disable-power-management.conf
repository you have also contributed to in the past.

And from what I remember this was in `8192cu-common` (or something like that) package

I would say yes but I don't know if other devices need it (apart from EW-7811un)

fryfrog, 2016-07-24:
It has been over a month since the last kernel release and this fix, is there going to be one soon?

@bmegli, any chance you'd make available the kernel and modules you compiled? :)

dlech, 2016-07-24:
Fix released in the `4.4.15-13-ev3dev-ev3` kernel.

closed 2016-07-24 by dlech

bmegli, 2016-07-25:
@fryfrog 

> It has been over a month since the last kernel release and this fix, is there going to be one soon?

It's fixed in new release.

In the future also have a look at *Nightly Image Builds* that @dlech made:

See the annoucement:
http://www.ev3dev.org/news/2016/07/11/nightly-image-builds/

That's where you can test (and could have tested!)  unreleased yet features.

Cheers!

dlech, 2016-07-25:
> In the future also have a look at Nightly Image Builds

We don't have nightly builds for the kernel package.

WasabiFan, 2016-07-25:
@dlech am I correct in that not having kernel packages just means that one needs to re-flash the SD card? From the way you say that it sounds like that is preventative.

dlech, 2016-07-25:
No, there is a kernel package. Just not nightly ones. You can still upgrade to the latest kernel using `sudo apt-get dist-upgrade` without having to re-flash your SD card.

I'm just saying that the nightly image builds before 2016-07-24 did not include any unreleased kernel fixes as @bmegli implied.

tprk77, 2016-07-25:
Yay! Thank you! The Edimax EW-7811Un is working flawlessly now.

bmegli, 2016-07-25:
> I'm just saying that the nightly image builds before 2016-07-24 did not include any unreleased kernel fixes as @bmegli implied.

Oh, ok. I am sorry for the confusion then. I haven't had time to try those yet too.