ev3dev/ev3dev #670: Ev3 not booting

FcBayernM, 2016-06-09:
### System information:

* I am using: LEGO MINDSTORMS EV3

* I installed ev3dev using this image file: ev3-ev3dev-jessie-2015-12-30.img

* My host computer is running: Windows

* I am connecting using: USB

### About my issue:

When I boot my EV3 with ev3dev, It either shows a blank screen with green led or just gets stuck on the boot splash. I've tried reflashing my sd card (32GB SDHC) and I'm using the latest img file yet i havn't had any success.

dlech, 2016-06-09:
Try a different SD card. Also make sure you don't have anything plugged into the EV3 (sensors, motors, USB).

Brian0526, 2016-06-11:
I'm pretty sure it said ev3 does not support any sd card more than 32GB.

dlech, 2016-10-11:
Please reopen if this is still an issue using the latest image from [here](https://oss.jfrog.org/list/oss-snapshot-local/org/ev3dev/brickstrap/)

closed 2016-10-11 by dlech