ev3dev/ev3dev #671: Possible Cause for "Reconnected due to: Bad message size"

bmegli, 2016-06-10:
Kernel compiled from https://github.com/ev3dev/ev3-kernel (Latest commit 3aaf365 on 8 May)

I think might have found it, not sure though.

In https://www.kernel.org/doc/Documentation/serial/tty.txt

> receive_buf()	-	(optional) Called by the low-level driver to hand
			a buffer of received bytes to the ldisc for
			processing. The number of bytes is guaranteed not
			to exceed the current value of tty->receive_room.
			All bytes must be processed.

The last line - *All bytes must be processed*

From what I have seen in kernel code the calling layer has no way to know that some data was not processed.

From https://github.com/ev3dev/ev3-kernel/blob/b23fdaca737d55757360027be39c9ae1fadb0b72/drivers/tty/tty_buffer.c

``` C
static int
receive_buf(struct tty_struct *tty, struct tty_buffer *head, int count)
{
	struct tty_ldisc *disc = tty->ldisc;
	unsigned char *p = char_buf_ptr(head, head->read);
	char	      *f = NULL;

	if (~head->flags & TTYB_NORMAL)
		f = flag_buf_ptr(head, head->read);

	if (disc->ops->receive_buf2)
		count = disc->ops->receive_buf2(tty, p, f, count);
	else {
		count = min_t(int, count, tty->receive_room);
		if (count && disc->ops->receive_buf)
			disc->ops->receive_buf(tty, p, f, count);
               //OUR receive_room is 65536, we almost always claim to have processed everything
	}
	return count;
}
```

But in line discipline in https://github.com/ev3dev/lego-linux-drivers/blob/master/sensors/ev3_uart_sensor_ld.c we have

``` C
static void ev3_uart_receive_buf(struct tty_struct *tty,
				 const unsigned char *cp, char *fp, int count)
{
	struct ev3_uart_port_data *port = tty->disc_data;
	struct circ_buf *cb = &port->circ_buf;
	int size;

	if (port->closing)
		return;

	if (count > CIRC_SPACE(cb->head, cb->tail, EV3_UART_BUFFER_SIZE))
		return; //here we have not processed everything, some data may be lost

	size = CIRC_SPACE_TO_END(cb->head, cb->tail, EV3_UART_BUFFER_SIZE);
	if (count > size) {
		memcpy(cb->buf + cb->head, cp, size);
		memcpy(cb->buf, cp + size, count - size);
		cb->head = count - size;
	} else {
		memcpy(cb->buf + cb->head, cp, count);
		cb->head += count;
	}

	schedule_work(&port->rx_data_work);
}
```

So we may occasionaly claim to process more than we did and miss some bytes from uart which would lead to incorrect message and resync with UART sensor.

But I only read parts of your code so far, so it may be a false alarm!

bmegli, 2016-06-10:
Oh,  I forgot.

The fix could probably be to switch to `tty_ldisc_ops.receive_buf2` instead which returns the number of bytes processed.

dlech, 2016-06-10:
I think using `tty_ldisc_ops.receive_buf2` would be a good move. We could also make our local buffer bigger. It is currently 256 bytes.

bmegli, 2016-06-10:
This may be the original cause for:
https://github.com/ev3dev/ev3dev/issues/662 
that along with buffer overflow fixed in:
https://github.com/ev3dev/lego-linux-drivers/commit/2fefd7d3a609176ba80cf3a0b948b3fe58b38b82
lead to crash

dlech, 2016-06-15:
This fix (just extra kernel message for buffer overrun) has been released in the [12-ev3dev](http://www.ev3dev.org/news/2016/06/14/Kernel-Release-Cycle-12/) kernel.

dlech, 2018-07-22:
Reopening because of https://github.com/ev3dev/ev3dev/issues/679#issuecomment-406778567

dlech, 2018-09-08:
Just had a discussion about this with @iassemble. It appears that the official LEGO firmware also suffers from such hardware buffer overruns.