ev3dev/ev3dev #673: Boot Parameter to Rotate Color LCD

rhempel, 2016-06-11:
### Feature Request

I built the color LCD hack and realized (too late) that the display was actually upside down. I also posted this on the web and a few people have followed my footsteps and also created upside-down displays.

Fixing the orientation is easy if you know how to rebuild a kernel by editing this code in `board-legoev3.c` to change `rotate` to 270. 

Or we could use a boot parameter to rotate the display - comments?

```
/* Adafruit 1.8" TFT */

static const struct fbtft_platform_data legoev3_st7735r_data = {
	.display = {
		.buswidth = 8,
		.backlight = 1,
	},
	.rotate = 90,
	.gpios = (const struct fbtft_gpio []) {
		{ "reset", EV3_LCD_RESET_PIN },
		{ "dc", EV3_LCD_A0_PIN },
		{ "led", EV3_LCD_CS_PIN },
		{},
	},
};
```

dlech, 2016-06-11:
It sounds like my display is now the one that is upside-down. My mounting is not permanent, so I say change the rotation to 270 and I will flip mine around to match the others.

rhempel, 2016-06-26:
Closed by https://github.com/ev3dev/ev3-kernel/pull/10

closed 2016-06-26 by rhempel

dlech, 2016-06-26:
Tip, if you write `Issue: ev3dev/ev3dev#673` in your commit message, it will automatically link to this issue.

dlech, 2016-10-17:
Fix was released several kernel versions ago.