ev3dev/ev3dev #674: proposal: add poll(2) support for changes to motor state attribute

kortschak, 2016-06-12:
See #657 for background.

It would be useful for synchronous operation of motors to be able to block on motor states. This can be implemented in user space if the user has access to `poll(2)` support in their language and the motor `state` attribute supports `poll(2)`. Many languages either have `poll(2)` support or can have it added reasonably easily, but the `state` attribute [no longer supports `poll(2)`](https://github.com/ev3dev/ev3dev/issues/447#issuecomment-222363710). So re-add it.

kortschak, 2016-10-05:
Link to [PR](https://github.com/ev3dev/lego-linux-drivers/pull/54) for the commit above (github does not link all commits in the PR apparently).

dlech, 2016-10-17:
Fix released in `16-ev3dev` kernel.

closed 2016-10-17 by dlech