ev3dev/ev3dev #677: Color LCD tft needs 18-bits but fb uses 16 bits. 2 bits are dropped?

mbitt, 2016-06-14:
### System information:

* I am using: *LEGO MINDSTORMS EV3*

* I installed ev3dev using this image file: *ev3-ev3dev-jessie-2015-12-30.img*

* My kernel version is: *3.16.7-ckt21-9-ev3dev-ev3*

* My host computer is running: *Linux*

* I am connecting using: *USB*

### About my issue:
After playing around with the color-lcd I've come to some obstacles.
- The documentation (https://www.adafruit.com/product/358) says it is a 18 bit-color display.
- fbset -i says that there are 16-bit per pixel.

I can get a bright blue with $3f that is %00000000 .00111111 and a wonderful green with $FC0  that is %00001111 .11000000 but I guess there is no way to put %0011.11110000 .00000000 that is $3F000 into 2 Bytes. Dropping the two most significant bits to %11110000 .00000000 gives a dumb red.

Do I see something wrong?

If so I would be glad to be pointed to the right way.

If not I suggest to change the framebuffer in size to 61440 (160 *128 *3) and the bytes per pixel to three. For convenience reasons it would be nice to have one byte for R, one byte for G and one for B values. No packed-pixel-format.

Kindly regards

Martin

PS I'm not a native speaker, so apologizes for crude English or misleading typos.

rhempel, 2016-06-14:
Martin,

The AdaFruit Color Screen is a 565 RGB screen, and we have some Python code to translate 24bit RGB to 565 here:

```
    def _color565(self, r, g, b):
        """Convert red, green, blue components to a 16-bit 565 RGB value. Components
        should be values 0 to 255.
        """
        return (((r & 0xF8) << 8) | ((g & 0xFC) << 3) | (b >> 3))

    def _img_to_rgb565_bytes(self):
        pixels = [self._color565(r, g, b) for (r, g, b) in self._img.getdata()]
        return pack('H' * len(pixels), *pixels)
```
Hope this helps, we are not going to change the underlying framebuffer format.

dlech, 2016-06-14:
Even though most of that AdaFruit TFTs are capable of 18 bits per pixel, the kernel drivers written for them only use the 16 bit per pixel mode, which as @rhempel said is RGB 565. You can see this the `rgba` you see in the output of `fbset`.

    rgba 5/11,6/5,5/0,0/16

It means the format is 5 bits for red shifted to the left 11, 6 bits for green shifted to the left 5 and 5 bits for blue shifted to the left 0 (and no alpha).

The fbdev driver matches the memory layout of the actual hardware, so that is not going to change. You can use an imaging library to convert from 24bpp to 16bpp if you would like to use 1 byte for each color.

mbitt, 2016-06-16:
David,

explaining the correct meaning of 
`rgba 5/11,6/5,5/0,0/16`
did the job for me. 

Ralph,

I tried your way i.e. dropping the least significant bits and my way i.e scaling the byte down. I couldn't see any differences in photographs showing in the display. I made a sort of split-screen for this.

Tanks to you both!

For interested people I add some commented forth-code that shows how I did it.

Regards Martin

```
: scale5 ( n -- n ) 		\ scale 8 bits down to 5 bits
  %11111 $FF */ ; 
  
: scale6 ( n -- n ) 		\ scale 8 bits down to 6 bits
  %111111 $FF */ ; 
  
: rgb: ( r g b -- ; -- ) 	\ r, g and b are in one byte
  create 			\ (we're in compiling mode) make new word
  scale5 			\ change blue byte to 5 bits
  swap scale6 5 lshift +	\ change green byte to 6 bits, combine with red
  swap scale5 11 lshift + 	\ change red byte to 5 bits, combine
  ,				\ store rgb value into created word
  does>  			\ (in interpreting mode) 
  @ color ! ;			\ fetch stored value and put it into variable color

\ Now we go to use the defining word rgb: to define some colors

    0   0   0 rgb: black
  255 255 255 rgb: white
  255   0   0 rgb: red
    0 255   0 rgb: green
    0   0 255 rgb: blue 
  255 255   0 rgb: yellow
  255   0 255 rgb: magenta
    0 255 255 rgb: cyan  
  
```

rhempel, 2016-06-17:
Glad to see you are still using Forth! Any particular reason why you re-opened the issue?

mbitt, 2016-06-17:
> Glad to see you are still using Forth! Any particular reason why you
> re-opened the issue?

No - there is no particular reason. Just an inexperienced user. :-(

Question: If I close this issue and anyone wants to leave a comment later: Is 
this than blocked?
Should I close it now?

BTW As you can presume I'm writing an (g)forth API based on ev3dev. (in my 
spare time.) Sensor class, Motor class, lcd, keys, sound etc. are working. But 
there are some 'philosophical' issues left (most about the user interface and 
naming conventions, some about basic tricks in open, enumerate and close 
devices).

Maybe I'll reach a state that could be called beta at the end of this year. 

Hope it would be good enough to became 'official' mentioned on ev3dev.org

Regards Martin

> 
> 
> ---
> https://github.com/ev3dev/ev3dev/issues/677#issuecomment-226761941

rhempel, 2016-06-17:
Feel free to close, we can always leave comments even on a closed thread.

Please consider using the autogen methods for building a Forth binding so that when the API changes, your binding is automatically updated.

I can help you with that

mbitt, 2016-06-17:
> Feel free to close, we can always leave comments even on a closed thread.
> 
> Please consider using the autogen methods for building a Forth binding so
> that when the API changes, your binding is automatically updated.
> 
> I can help you with that

I appreciate your help.

I'll let you know when.

Regards

Martin

> 
> ---
> https://github.com/ev3dev/ev3dev/issues/677#issuecomment-226773181

closed 2016-06-17 by mbitt