ev3dev/ev3dev #678: Keep all documentation versions available

VincentLorenzo, 2016-06-15:
Hello, 
   we are using ev3dev with c++ in our labs to illustrate the capabilities of our software [http://eclipse.org/papyrus/](url).  We started it 8 months ago with the available version at this time, with Ubuntu and Windows 7. Since the start of our project, we noticed that the ev3dev version changed and the documentation is updated each times (a good thing!), but the documentation of the previous versions is each time lost. Shall I suggest you to try to keep available each version of the documentation ? 

It is very annoying for us to not find the same informations each time we need to install a new brick. 
Maybe a wiki with the history of the changes or a tab by version could be helpful. 

Mainly, we had troubles with the page[ http://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-usb/]( http://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-usb/) . Previous version of Ubuntu disappeared, Window 7 disappears, ... We can't be up-to-date on the ev3dev version neither on the lastest Ubuntu/Windows release. 

Currently we used this website to try to found previous version: [https://web.archive.org/web/*/ev3dev.org](url), but it is not the best way. 

Nevertheless, we are very happy with ev3dev, and I will create a project page on our website asap (when we will get interesting stuff to share with lego user with clean code and documentation!)

Best Regards, 
--
 Vincent Lorenzo

WasabiFan, 2016-06-15:
We briefly discussed this in ev3dev/ev3dev.github.io#133. In summary, the current decision is that we can't support older versions at this time and as a result the documentation is not kept. I will take another look to see if I can find a good way of maintaining older documentation versions, but with the way our site is built it isn't particularly easy to do it in a reliable way. The primary issue is that the work required to keep old docs functional as the site architecture changes might be more than it is worth in the short term. Once we have stable ev3dev releases (and those releases come less frequently), this will definitely need to be addressed.

WasabiFan, 2016-06-15:
As a short-term solution @dlech I think we should start tagging the website with each kernel release; that way at a minimum one can checkout the docs for a particular version for personal reference.

dlech, 2016-06-15:
> Mainly, we had troubles with the page http://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-usb/ . Previous version of Ubuntu disappeared, Window 7 disappears, ... We can't be up-to-date on the ev3dev version neither on the lastest Ubuntu/Windows release.

Although the screenshots have been updated, these should still work with Windows 7 and Ubuntu 14.04. If there is something different about Windows 7, we would be glad to take a pull request to fix it since there are many people still using Windows 7.

dlech, 2016-06-15:
And the website is all in git. You can checkout any commit you like and run your own local copy of the website using jekyll. See the readme at <https://github.com/ev3dev/ev3dev.github.io>.

VincentLorenzo, 2016-06-16:
Thank you for your answer. I didn't see that the website is stored in the git. It will help us the next time we get some doubt with the doc!

dlech, 2017-03-10:
We are in the process of moving documentation to readthedocs.org, so we will have the possibility of versioned documentation from now on.

closed 2017-03-10 by dlech