ev3dev/ev3dev #679: FATCATLAB 9DOF causes buffer overrun on Ev3

kortschak, 2016-06-16:
### System information:

* I am using: LEGO MINDSTORMS EV3 and FATCATLAB EVB

* I installed ev3dev using this image file: ev3-ev3dev-jessie-2015-12-30.img

* My kernel version is: 4.4.13-12-ev3dev-ev3 also the release 12 on the EVB.

* My host computer is running: Linux

* I am connecting using: Wi-Fi

### About my issue:

#### What led up to the situation?
Recent fix of 9DOF kernel crash allows a hidden bug to be identified.

#### What exactly did you do (or not do) that was effective (or ineffective)?
Plug in FATCATLAB 9DOF to EV3 (or EVB - though slightly different outcome there due presumably to difference in cpu performance between the two).
Then navigate to device browser and click watch values.

#### What was the outcome of this action?
The following notification by brickman for the EV3 "Error reading from file: No such device or address" and the following `dmesg` output:

```
[   68.492177] lego-port port0: Added new device 'in1:ev3-uart-host'
[   69.777069] Registered EV3 UART sensor line discipline. (29)
[   70.739787] lego-sensor sensor0: Registered 'ev3-uart-76' on 'in1'.
[   70.792347] tty ttyS1: Reconnected due to: Bad message size
[   73.023261] tty ttyS1: Reconnected due to: Bad message size
[   73.444670] ttyS1: buffer overrun
[   73.445291] ttyS1: buffer overrun
[   73.447617] ttyS1: buffer overrun
[   77.135573] ttyS1: buffer overrun
[   77.136330] ttyS1: buffer overrun
[   77.138547] ttyS1: buffer overrun
[   77.139235] ttyS1: buffer overrun
[   77.141624] ttyS1: buffer overrun
[   77.141975] ttyS1: buffer overrun
[   77.142622] ttyS1: buffer overrun
[   79.404329] tty ttyS1: Reconnected due to: Bad message size
[   87.576531] ev3_uart_receive_buf: 7 callbacks suppressed
[   87.576584] ttyS1: buffer overrun
[   87.577162] ttyS1: buffer overrun
[   89.786367] tty ttyS1: Reconnected due to: Bad message size
[   90.956587] ttyS1: buffer overrun
[   90.957222] ttyS1: buffer overrun
[   90.959295] ttyS1: buffer overrun
[   90.960065] ttyS1: buffer overrun
[   90.962575] ttyS1: buffer overrun
[   90.962862] ttyS1: buffer overrun
[   90.963937] ttyS1: buffer overrun
[   90.965152] ttyS1: buffer overrun
[   93.167001] tty ttyS1: Reconnected due to: Bad message size
[   94.177694] ev3_uart_receive_buf: 10 callbacks suppressed
[   94.177833] ttyS1: buffer overrun
[   94.178349] ttyS1: buffer overrun
[   96.447631] tty ttyS1: Reconnected due to: Bad message size
[  123.715056] ttyS1: buffer overrun
[  123.715812] ttyS1: buffer overrun
[  133.015177] lego-port port0: Removed device 'in1:ev3-uart-host'
[  133.138656] lego-sensor sensor0: Unregistered 'ev3-uart-76' on 'in1'.
[  135.596631] lego-port port0: Added new device 'in1:ev3-uart-host'
[  137.836522] lego-sensor sensor1: Registered 'ev3-uart-76' on 'in1'.
[  155.711342] ttyS1: buffer overrun
[  155.712021] ttyS1: buffer overrun
[  155.714145] ttyS1: buffer overrun
[  155.714862] ttyS1: buffer overrun
[  157.920986] tty ttyS1: Reconnected due to: Bad message size
[  188.358156] lego-port port0: Removed device 'in1:ev3-uart-host'
[  188.444315] lego-sensor sensor1: Unregistered 'ev3-uart-76' on 'in1'.
[  190.940146] lego-port port0: Added new device 'in1:ev3-uart-host'
[  193.189607] lego-sensor sensor2: Registered 'ev3-uart-76' on 'in1'.
[  195.470074] tty ttyS1: Reconnected due to: Bad message size
```

Pressing back and re-clicking watch values restores the expected brickman behaviour for the EV3, presumably because the device has stabilised after the overruns.

Similar behaviours are seen on EVB (see https://github.com/ev3dev/ev3dev/issues/662#issuecomment-226161965).

#### What outcome did you expect instead?
Normal sensor operation and no kernel noise.

dlech, 2016-06-16:
Does this also happen with the LEGO EV3 Gyro sensor?

Can you enable the debug prints [here](https://github.com/ev3dev/lego-linux-drivers/blob/a5219d6036851ea8654bcb007f682bf70b375806/sensors/ev3_uart_sensor_ld.c#L561) and [here](https://github.com/ev3dev/lego-linux-drivers/blob/a5219d6036851ea8654bcb007f682bf70b375806/sensors/ev3_uart_sensor_ld.c#L561) (replace `debug_pr` with `printk`) so we can see that the bad data looks like.

bmegli, 2016-06-16:
@dlech - you pasted the same link twice (here and here is the same there!) 

It's interesting that `Bad message size` comes out first.
Also it is possible that only first buffer overrun matters, the later could be side effects of printing from interrupt.

So it seems that it may be harder to debug: bad message could be the cause for reconnect, this in turn for overrun. So still cause for bad message may be unknown.

dlech, 2016-06-16:
> It's interesting that Bad message size comes out first.

It is because I used `printk_ratelimited`, I think. This is also where the `ev3_uart_receive_buf: 7 callbacks suppressed` comes from. It suppressed 7 more of the same message. The printing may be deferred.

dlech, 2016-06-16:
> @dlech - you pasted the same link twice (here and here is the same there!)

I have corrected the link

bmegli, 2016-06-16:
@dlech May be deferred but from what I understand from @kortschak there is `bad message size` in EVB and no overrun.

@kortschak Could you confirm that (if you remember)? Was there only `bad message size` and no `overrun` message in EVB? Overrun message only comes out in recently released kernel so if you tried previously you wouldn't get it.

I think that the order may be important here.

kortschak, 2016-06-16:
Explicity kernel version

```
$ uname -r
4.4.12-ti-rt-r30-12-ev3dev-bb.org
```

@bmegli Yes, that's correct, only `bad message size` on EVB.

Here is three cycles on attach/remove without brickman watching, followed by two cycles of attach remove with brickman watching.

```
[  107.968045] lego-port port0: Added new device 'evb-ports:in1:ev3-uart-host'
[  108.170915] Registered EV3 UART sensor line discipline. (29)
[  110.213990] lego-sensor sensor0: Registered 'ev3-uart-76' on 'evb-ports:in1'.
[  110.604061] lego-port port0: Removed device 'evb-ports:in1:ev3-uart-host'
[  110.625485] lego-sensor sensor0: Unregistered 'ev3-uart-76' on 'evb-ports:in1'.
[  110.653904] pin1_mv: 4992
[  110.656563] pin6_mv: 58
[  110.659024] pin5_gpio: 1
[  110.661569] pin6_gpio: 0
[  111.937544] lego-port port0: Added new device 'evb-ports:in1:ev3-uart-host'
[  112.853799] lego-sensor sensor1: Registered 'ev3-uart-76' on 'evb-ports:in1'.
[  114.013820] lego-port port0: Removed device 'evb-ports:in1:ev3-uart-host'
[  114.033166] lego-sensor sensor1: Unregistered 'ev3-uart-76' on 'evb-ports:in1'.
[  114.063634] pin1_mv: 4992
[  114.066290] pin6_mv: 63
[  114.068752] pin5_gpio: 1
[  114.071299] pin6_gpio: 0
[  115.284066] lego-port port0: Added new device 'evb-ports:in1:ev3-uart-host'
[  116.203526] lego-sensor sensor2: Registered 'ev3-uart-76' on 'evb-ports:in1'.
[  116.903614] lego-port port0: Removed device 'evb-ports:in1:ev3-uart-host'
[  116.922434] lego-sensor sensor2: Unregistered 'ev3-uart-76' on 'evb-ports:in1'.
[  116.953431] pin1_mv: 4992
[  116.956088] pin6_mv: 58
[  116.958556] pin5_gpio: 1
[  116.961100] pin6_gpio: 0
[  139.775907] lego-port port0: Added new device 'evb-ports:in1:ev3-uart-host'
[  140.691919] lego-sensor sensor3: Registered 'ev3-uart-76' on 'evb-ports:in1'.
[  142.981165] tty ttyS4: Reconnected due to: Bad message size
[  155.051131] lego-port port0: Removed device 'evb-ports:in1:ev3-uart-host'
[  155.083038] lego-sensor sensor3: Unregistered 'ev3-uart-76' on 'evb-ports:in1'.
[  155.090945] pin1_mv: 4992
[  155.090950] pin6_mv: 63
[  155.090956] pin5_gpio: 1
[  155.090960] pin6_gpio: 0
[  158.251284] lego-port port0: Added new device 'evb-ports:in1:ev3-uart-host'
[  159.172018] lego-sensor sensor4: Registered 'ev3-uart-76' on 'evb-ports:in1'.
[  161.464033] tty ttyS4: Reconnected due to: Bad message size
[  298.604487] lego-port port0: Removed device 'evb-ports:in1:ev3-uart-host'
[  298.635923] lego-sensor sensor4: Unregistered 'ev3-uart-76' on 'evb-ports:in1'.
[  298.644312] pin1_mv: 4992
[  298.644317] pin6_mv: 63
[  298.644323] pin5_gpio: 1
[  298.644326] pin6_gpio: 0
```

bmegli, 2016-06-17:
@dlech asked
> Does this also happen with the LEGO EV3 Gyro sensor?

No, I have just tried more than 10 times with EV3 gyro and no overruns or bad messages.

The only interesting thing is that depending on timing (of plug-unplug) the gyro is registered and unregistered (if time allows) or it's only ev3-uart-host  that is added and removed. But this seems expected.

Considering EVB and EV3 behaviour the `bad message size` really seems to be coming first with reasonable @kortschak explanation that the difference (EV3 vs EVB) is due to CPU performance.

bmegli, 2016-06-17:
Interesting.

I have just tried plugging and unplugging 3 UART sensors at the same time (gyro, IR, US) and got both:
- `serial8250: too much work for irq53`
- `ttyS1: buffer overrun`

No `bad message size`.

The interesting fragment from dmesg:

``` bash
[ 4213.215287] lego-port port2: Removed device 'in3:ev3-uart-host'
[ 4213.313582] lego-port port1: Removed device 'in2:ev3-uart-host'
[ 4213.702634] lego-port port0: Removed device 'in1:ev3-uart-host'
[ 4213.829897] lego-sensor sensor7: Unregistered 'lego-ev3-gyro' on 'in1'.
[ 4214.543520] lego-port port2: Added new device 'in3:ev3-uart-host'
[ 4214.664300] lego-port port1: Added new device 'in2:ev3-uart-host'
[ 4214.875353] lego-port port0: Added new device 'in1:ev3-uart-host'
[ 4215.667582] serial8250: too much work for irq53
[ 4216.024008] serial8250: too much work for irq53
[ 4216.026508] ttyS1: buffer overrun
[ 4216.026560] ttyS1: buffer overrun
[ 4216.026586] ttyS1: buffer overrun
[ 4216.026605] ttyS1: buffer overrun
[ 4216.026625] ttyS1: buffer overrun
[ 4216.026643] ttyS1: buffer overrun
[ 4216.026660] ttyS1: buffer overrun
[ 4216.026677] ttyS1: buffer overrun
[ 4216.026695] ttyS1: buffer overrun
[ 4216.026712] ttyS1: buffer overrun
[ 4216.391428] serial8250: too much work for irq53
[ 4216.606738] serial8250: too much work for irq53
[ 4216.826677] serial8250: too much work for irq53
[ 4217.046658] serial8250: too much work for irq53
[ 4217.204882] lego-port port2: Removed device 'in3:ev3-uart-host'
[ 4217.266713] serial8250: too much work for irq53
[ 4217.486665] serial8250: too much work for irq53
[ 4217.706605] serial8250: too much work for irq53
[ 4217.926591] serial8250: too much work for irq53
[ 4218.252220] lego-port port0: Removed device 'in1:ev3-uart-host'
[ 4218.470897] lego-port port2: Added new device 'in3:ev3-uart-host'
[ 4218.866237] lego-port port0: Added new device 'in1:ev3-uart-host'
[ 4221.575695] lego-sensor sensor9: Registered 'lego-ev3-ir' on 'in2'.
[ 4222.412775] lego-sensor sensor10: Registered 'lego-ev3-us' on 'in3'.
[ 4222.599352] lego-sensor sensor11: Registered 'lego-ev3-gyro' on 'in1'.
```

dlech, 2016-06-17:
> Considering EVB and EV3 behaviour the bad message size really seems to be coming first

Yes, and when we get bad message size, we stop emptying the circular buffer, which leads to overrun.

bmegli, 2016-06-20:
As I understand from @kortschak input, in the faster EVB we get `Bad Message Size` only when Brickman is watching, in slower EV3 we get it always.

I only have wild guess idea that we get `Bad Message Size` when the uart baudrate is changed too late on our side and the sensor has already raised it's own.

From: https://github.com/mindboards/ev3sources/blob/master/lms2012/d_uart/Linuxmod_AM1808/d_uart_mod.c#83

``` 
 - Exchange informations                                                             - Exchange informations                              |
      - Receive command data                                         <-- CMD              - Send command data (type,modes,speed, etc)        |
      - Send command data (type,modes,speed, etc)           CMD  -->                      - Receive command data                             |
      - Receive info data                                            <-- INFO             - Send info data (name,scaling,data format, etc)   |
      - Send info data (name,scaling,data format, etc)      INFO -->                      - Receive info data                                |
      - Receive acknowledge                                          <-- ACK              - When finished info send acknowledge              |
                                                                                          - Timeout (80mS)      -----------------------------'
      - When finished info send acknowledge                 ACK  -->                      - Receive acknowledge
      - Switch to valid communication speed                                               - Switch to valid communication speed
```

It doesn't seem to be documented in the protocol but original EV3 waits some time after sending ACK and changes the baudrate:

https://github.com/mindboards/ev3sources/blob/master/lms2012/d_uart/Linuxmod_AM1808/d_uart_mod.c#L3430

``` C

          case UART_ACK_WAIT :
          {
            if (++(UartPort[Port].Timer) >= (UART_ACK_DELAY / UART_TIMER_RESOLUTION))
            {
              (*pUart).Status[Port]  |=  UART_PORT_CHANGED;
              UartPortSend(Port,BYTE_ACK); //SEND ACK HERE
              UartPort[Port].Timer    =  0;
              UartPort[Port].State    =  UART_ACK_INFO;
            }
          }
          break;

          case UART_ACK_INFO :
          {
            if (++(UartPort[Port].Timer) >= (UART_CHANGE_BITRATE_DELAY / UART_TIMER_RESOLUTION)) //WAIT HERE, I believe it's 1 ms here only
            {
              UartPort[Port].DataOk       =  1;
              UartPort[Port].DataErrors   =  0;
              UartPort[Port].Mode         =  0;
              UartPort[Port].BitRate      =  UartPort[Port].BitRateMax;
              UartPortSetup(Port,UartPort[Port].BitRate);
              UartPort[Port].WatchDog     =  (UART_WATCHDOG_TIME / UART_TIMER_RESOLUTION);
              UartPort[Port].State        =  UART_DATA;
            }
          }
break;

```

I think it waits only 1 ms after sending ACK to switch baudrate

Now, on our side:
https://github.com/ev3dev/lego-linux-drivers/blob/master/sensors/ev3_uart_sensor_ld.c#L387

``` C

static void ev3_uart_send_ack(struct work_struct *work)
{
	struct ev3_uart_port_data *port = container_of(to_delayed_work(work),
				struct ev3_uart_port_data, send_ack_work);
	int err;

	if (port->closing)
		return;

	ev3_uart_write_byte(port->tty, EV3_UART_SYS_ACK);

	if (!port->sensor.context && port->type_id <= EV3_UART_TYPE_MAX) {
		port->sensor.context = port->tty;

              //Is this time consuming?
		err = register_lego_sensor(&port->sensor, port->tty->dev);
		if (err < 0) {
			port->sensor.context = NULL;
			if (port->in_port) {
				put_device(&port->in_port->dev);
				port->in_port = NULL;
			}
			dev_err(port->tty->dev,
				"Could not register UART sensor on tty %s",
				port->tty->name);
			return;
		}
	} else
		dev_err(port->tty->dev, "Reconnected due to: %s\n",
			port->last_err);

        //Why 4?
	mdelay(4);
	schedule_work(&port->change_bitrate_work);
}

```

Is registering device time consuming? Why do we wait exactly 4 before scheduling bitrate change?

So I am only guessing but we could add printk here:

https://github.com/ev3dev/lego-linux-drivers/blob/master/sensors/ev3_uart_sensor_ld.c#L552

``` C
	/*
		 * Sometimes we get 0xFF after switching baud rates, so just
		 * ignore it.
		 */
		if ((u8)cb->buf[cb->tail] == 0xFF) {

			cb->tail++;
			if (cb->tail >= EV3_UART_BUFFER_SIZE)
				cb->tail = 0;
			count--;
			continue;
}
```

So that we know if `bad message size` follows corrupted data from UART after changing the baudrate.

I expect this can be not only corrupted byte after changing the baudrate but it could be a start.

dlech, 2016-06-20:
> Is registering device time consuming?

Yes

> Why do we wait exactly 4 before scheduling bitrate change?

I imagine that it was empirically selected. If we don't wait long enough, the sensor is busy when we try to send the first keep alive (NACK) and if we wait too long the sensor resets.

It could be though that we are changing too late and we are missing the first part of the message after the sensor changes baud rate.

bmegli, 2016-06-20:
For short:

We could add printk here for the start
https://github.com/ev3dev/lego-linux-drivers/blob/master/sensors/ev3_uart_sensor_ld.c#L552

and check if it preceeds `bad message size`

``` C
/*
         * Sometimes we get 0xFF after switching baud rates, so just
         * ignore it.
         */
        if ((u8)cb->buf[cb->tail] == 0xFF) {
            printk("ignoring 0xFF after baudrate change");
            cb->tail++;
            if (cb->tail >= EV3_UART_BUFFER_SIZE)
                cb->tail = 0;
            count--;
            continue;
}
```

Or other idea, rearange:
``` C
static void ev3_uart_send_ack(struct work_struct *work
{
 Send(EV3_UART_SYS_ACK)
 ChangeBitrate()
 RegisterDevice()
 WaitWhateverIsNeeded()
 ScheduleTheStuffFromEndOfChangeBitrate() // keep_alive_timer, set_mode
}
```

Is there some technical reason why we couldn't change the bitrate directly from ev3_uart_send_ack after sending? (I simply don't know the code enough to tell) 

But those all are just guesses.

bmegli, 2016-06-26:
I think I may have found possible cause for Brickman notification "Error reading from file: No such device or address".

The sensor is synced, gets mode info, ACK, sends ACK.

In line https://github.com/ev3dev/lego-linux-drivers/blob/master/sensors/ev3_uart_sensor_ld.c#L399
``` C
err = register_lego_sensor(&port->sensor, port->tty->dev);
```

The device is registered and from now on it's available through sysfs.

Now follows the baudrate change and later (reason still unknown), message with incorrect size.
The sensor sync flag is reset and sync procedure is repeated.

In sync lines https://github.com/ev3dev/lego-linux-drivers/blob/master/sensors/ev3_uart_sensor_ld.c#L519
`mode_info` is reset effectively setting both `num_values` and `data_sets` to 0.
 
``` C
for (i = 0; i <= EV3_UART_MODE_MAX; i++)
			port->mode_info[i] = ev3_uart_default_mode_info;
 ```

Now, if Brickman through vala tries to read the sysfs value0 before getting new mode_info on uart the following code is executed in lego sensor `value_show`:
``` C
	if (index < 0 || index >= lego_sensor_get_num_values(mode_info))
		return -ENXIO;
```
https://github.com/ev3dev/lego-linux-drivers/blob/036f41ab854c3cdc21da3b6451a94fa57d2d606a/sensors/lego_sensor_class.c#L421 

Now, index is clearly 0 but also `num_values` and `data_sets` are 0 because they were reset for resync. So lego_sensor_get_num_values returns 0.

``` C
static inline int lego_sensor_get_num_values(struct lego_sensor_mode_info *mode_info)
{
	return mode_info->num_values ? mode_info->num_values : mode_info->data_sets;
}
```
and sysfs read on value0 will return -ENXIO which in human readable form should translate to something like "no such device or address", the error we get in Brickman.

So summing up - if the sensor resyncs we have a possible race condition between user space trying to read the sensor through sysfs and sensor resetting mode_info to defaults and getting new mode_info from UART. This is all non fatal and errors are reported to users.

-------------------------------------------------

I didn't really think what to do with it. I am trying to find the cause for `bad message size` which seems to be the original problem. The above just happens to be side effect of resync which we may fix in some way.

dlech, 2016-06-26:
Sounds like you should open a new issue for the problem of ENXIO.

bmegli, 2016-06-27:
@dlech I think that maybe you are treating `bad_message_size` in too harsh way.

Original EV3 LMS2012 simply ignores messages with bad size when it's in `UART_DATA` mode:
https://github.com/mindboards/ev3sources/blob/master/lms2012/d_uart/Linuxmod_AM1808/d_uart_mod.c#L1117

Precisely it skips those bytes until it gets valid length data byte and then starts collecting the message.

What do you think about treating `bad_message_size` like CRC failures? E.g. like here:

``` C
		if (chksum != message[msg_size - 1]
			    && port->type_id != EV3_UART_TYPE_ID_COLOR
			    && message[0] != 0xDC)
               {
			port->last_err = "Bad checksum.";
				if (port->info_done) {
					port->num_data_err++;
					goto err_bad_data_msg_checksum;
				} else
goto err_invalid_state;
            }
```

or counting them on their own. I mean - if it happens after info_done ignore a few before complete resync.

bmegli, 2016-06-27:
@kortschak 

If you happen to build your own kernel could you enable some additional printks and check with 9DOF:

The ones that @dlech wanted:
> Can you enable the debug prints here and here (replace debug_pr with printk) so we can see that the bad data looks like.

- https://github.com/ev3dev/lego-linux-drivers/blob/a5219d6036851ea8654bcb007f682bf70b375806/sensors/ev3_uart_sensor_ld.c#L561
- https://github.com/ev3dev/lego-linux-drivers/blob/a5219d6036851ea8654bcb007f682bf70b375806/sensors/ev3_uart_sensor_ld.c#L563

And add one:
https://github.com/ev3dev/lego-linux-drivers/blob/master/sensors/ev3_uart_sensor_ld.c#L552
``` C
/*
         * Sometimes we get 0xFF after switching baud rates, so just
         * ignore it.
         */
        if ((u8)cb->buf[cb->tail] == 0xFF) {

            cb->tail++;
            if (cb->tail >= EV3_UART_BUFFER_SIZE)
                cb->tail = 0;
            count--;
            printk("Ignoring 0xFF (baudrate change?)\n");
            continue;
}
```

dlech, 2016-06-27:
> What do you think about treating bad_message_size like CRC failures?

I think that would be and OK thing to do since it seems to be a common occurrence.

dlech, 2016-10-22:
The buffer overrun seems to be fixed (using `16-ev3dev` kernel). Please re-open if `buffer overrun` is still an issue or open a new issue for `Reconnect due to bad message size`.

laurensvalk, 2017-10-30:
> Please re-open if buffer overrun is still an issue

Several overruns with the LEGO EV3 Gyro [here](https://github.com/laurensvalk/segway/issues/10#issuecomment-340375025), possibly related.

dlech, 2017-10-30:
Kernel version?

gregcowell, 2017-10-30:
**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-ev3-generic-2017-09-30
Kernel version:     4.9.49-ev3dev-1.5.1-ev3
Brickman:           0.10.0
ev3devKit:          0.5.2
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016534FB669
BOARD_INFO_TYPE=main
```

dlech, 2017-10-30:
Hmm... fix for this issue is also in the stretch/4.9 kernel https://github.com/ev3dev/lego-linux-drivers/commit/57fca3bb8d01b8b02bec0d803de2862d0297fed0

laurensvalk, 2017-10-30:
Just adding that I see the same on the latest image.

```
Image file:         ev3dev-stretch-ev3-generic-2017-10-24
Kernel version:     4.9.49-ev3dev-1.5.1-ev3
Brickman:           0.10.0
ev3devKit:          0.5.2
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=001653420A17
BOARD_INFO_TYPE=main
```

dlech, 2018-07-20:
Anyone tried this with a 4.14 kernel yet? I expect that it is still a problem, but you never know...

kortschak, 2018-07-20:
I will have a look later today.

kortschak, 2018-07-21:
```
robot@fatcat:~$ ev3dev-sysinfo 
Image file:         ev3dev-stretch-bone-generic-2017-09-30
Kernel version:     4.14.54-ev3dev-2.2.0-bb.org
Brickman:           0.10.0
BogoMIPS:           995.32
Board:              board0
BOARD_INFO_HW_REV=00C0
BOARD_INFO_MODEL=A335BNLT
BOARD_INFO_SERIAL_NUM=1116BBBK1052
BOARD_INFO_TYPE=main
Board:              board1
BOARD_INFO_MODEL=FatcatLab EVB
BOARD_INFO_TYPE=aux
```

dmesg output:
```
[   53.511796] Registered EV3 UART sensor line discipline. (29)
[   55.553847] lego-sensor sensor0: Registered 'fcl-9dof' on 'evb-ports:in2'.
[   57.865696] tty ttyS2: Reconnected due to: Bad message size
[   71.899120] lego-port port5: Removed device 'evb-ports:in2:ev3-uart-host'
[   71.925348] lego-sensor sensor0: Unregistered 'fcl-9dof' on 'evb-ports:in2'.
[   78.705873] lego-port port5: Added new device 'evb-ports:in2:ev3-uart-host'
[   79.625836] lego-sensor sensor1: Registered 'fcl-9dof' on 'evb-ports:in2'.
[   79.669773] tty ttyS2: Reconnected due to: No data since last keep-alive.
```

So the bad message size persists.

----
I did not have an easy time doing this - wifi appears to be broken on the EB with stretch. After scanning fine and showing services and then performing a connect, connmanctl gives me "Agent request cancelled by ConnMan" and "Error /net/connman/service/wifi_<stuff>_managed_psk: Operation aborted", and dmesg shows:

```
[  290.537615] R8188EU: indicate disassoc
```

Wi-Fi is fine on jessie and the relevant module is loaded in stretch.
```
robot@fatcat:~$ lsmod | grep 8188
r8188eu               454656  0
cfg80211              622592  1 r8188eu
```

dlech, 2018-07-21:
Please open a new issue for the Wi-Fi problem.

dlech, 2018-07-21:
> [ 57.865696] tty ttyS2: Reconnected due to: Bad message size

This seems to be a different issue as well (https://github.com/ev3dev/ev3dev/issues/679#issuecomment-255559047). I don't see anything in your output about "buffer overrun", which is the topic of this issue.

kortschak, 2018-07-22:
OK, I'm not seeing buffer overruns on the EV3.
```
[  177.649351] lego-port port0: Added new device 'ev3-ports:in1:ev3-uart-host'
[  178.621505] Registered EV3 UART sensor line discipline. (29)
[  179.907276] lego-sensor sensor0: Registered 'fcl-9dof' on 'ev3-ports:in1'.
[  237.138936] lego-port port0: Removed device 'ev3-ports:in1:ev3-uart-host'
[  237.270828] lego-sensor sensor0: Unregistered 'fcl-9dof' on 'ev3-ports:in1'.
[  247.254570] lego-port port0: Added new device 'ev3-ports:in1:ev3-uart-host'
[  249.514231] lego-sensor sensor1: Registered 'fcl-9dof' on 'ev3-ports:in1'.
[  249.542461] tty ttyS1: Reconnected due to: Bad message size
[  251.811180] tty ttyS1: Reconnected due to: Bad message size
```

dlech, 2018-07-22:
OK, so let's open a new issue for "Bad message size" to keep them separate.

dlech, 2018-07-22:
Or re-open #671

dlech, 2018-07-22:
Leaving this open because of https://github.com/ev3dev/ev3dev/issues/679#issuecomment-340587990

dlech, 2018-07-26:
The circular buffer has been removed from the EV3 UART line discipline driver in kernel `ev3dev-2.2.1`, so it is no longer possible to get the "buffer overrun" error message.

It is expected that there will still be issue with UART sensors, especially the 9DOF, but the info here is no longer relevant. 

Please check the kernel log for error messages `dmesg | grep tty` and search for existing issue or open a new issue if needed.

closed 2018-07-26 by dlech