ev3dev/ev3dev #680: FATCATLAB 9DOF ACC mode is off by a factor of 2

kortschak, 2016-06-16:
### System information:

* I am using: LEGO MINDSTORMS EV3 and FATCATLAB EVB

* I installed ev3dev using this image file: ev3-ev3dev-jessie-2015-12-30.img and evb-ev3dev-jessie-2016-01-25.img.xz

* My kernel version is: 4.4.13-12-ev3dev-ev3 also the release 12 kernel on the EVB.

* My host computer is running: Linux

* I am connecting using: Wi-Fi

### About my issue:

#### What led up to the situation?
Recent fix of 9DOF kernel crash allows a hidden bug to be identified.

#### What exactly did you do (or not do) that was effective (or ineffective)?
Plug in FATCATLAB 9DOF to EV3 or EVB.
Then navigate to device browser set mode to ACC and click watch values.

#### What was the outcome of this action?
See either ~0.5g or ~-0.5g.

#### What outcome did you expect instead?
Standard gravity (EV-G gives this on both devices, EV3 and EVB).

dlech, 2016-06-16:
Can you enable the debug prints [here](https://github.com/ev3dev/lego-linux-drivers/blob/a5219d6036851ea8654bcb007f682bf70b375806/sensors/ev3_uart_sensor_ld.c#L730), [here](https://github.com/ev3dev/lego-linux-drivers/blob/a5219d6036851ea8654bcb007f682bf70b375806/sensors/ev3_uart_sensor_ld.c#L747) and [here](https://github.com/ev3dev/lego-linux-drivers/blob/a5219d6036851ea8654bcb007f682bf70b375806/sensors/ev3_uart_sensor_ld.c#L764) (replace `debug_pr` with `printk`)? Also the 4 debug prints starting [here](https://github.com/ev3dev/lego-linux-drivers/blob/a5219d6036851ea8654bcb007f682bf70b375806/sensors/ev3_uart_sensor_ld.c#L859).

dlech, 2016-06-16:
It could be that this value is scaled in the EV3-G software.

kortschak, 2016-06-16:
I can do those later this weekend.

I supposed it could be scaled in the EV3-G software, but Occam's razor suggests not; both EV3 and EVB do this and they would both have to know about the scaling factor for this sensor without LEGO having known that it would exist. Since the EV3 does it, the source for its behaviour is available though.

bmegli, 2016-06-18:
@dlech I think that we may have possible value overflow in lego_sensor_ftoi
I will try to elaborate later, now  I have to leave.

UART sensors report bounds in float so this is called on the way.

``` C
int lego_sensor_ftoi(uint32_t f, unsigned dp)
 {
	int s = (f & 0x80000000) ? -1 : 1;
	unsigned char e = (f & 0x7F800000) >> 23;
	unsigned long i = f & 0x007FFFFFL;
	unsigned long m;

	/* handle special cases for zero, +/- infinity and NaN */
	if (!e)
		return 0;
	if (e == 255)
		return s == 1 ? INT_MAX : INT_MIN;

	i += 1 << 23;
	while (dp--)
	{	
		i *= 10;
                //THIS MAY OVERFLOW if dp supplied to the funciton is >=3 for example and f was 16 float
	}
	if (e < 150) {
		m = i % (1L << (150 - e));
		i += m >> 1;
		i >>= 150 - e;
	}
	else
		i <<= e - 150;

	return s * i;
}
```

e.g., reproduce with

``` C
	float x=16;
	uint32_t ux=*(uint32_t *)(&x);
	int y=lego_sensor_ftoi(ux, 3);
	printf("%d\n", y);
```

dlech, 2016-06-18:
I think that when I first started working on ev3dev, I mistakenly believed `long` was 64 bit.

    s32 s = (f & 0x80000000) ? -1 : 1;
    u8 e = (f & 0x7F800000) >> 23;
    u64 i = f & 0x007FFFFFL;
    u64 m;

should be better.

bmegli, 2016-06-18:
@dlech That should be safe.

If I am correctly guessing the mode info supplied by 9DOF this should also solve 9DOF acc mode scaling problem.

e.g. SI_MIN=-16, SI_MAX=16 and 3 decimals would give:
-around 1000 for RAW_MAX/16 with fix
-around 500 for RAW_MAX/16 without fix (overflow magic)

kortschak, 2016-06-19:
With the above decl changes, I get

```
ERROR: "__aeabi_uldivmod" [drivers/lego/sensors/lego_sensor_class.ko] undefined!
```

```
$ gcc --version
gcc (Ubuntu 4.8.4-2ubuntu1~14.04.3) 4.8.4
```

kortschak, 2016-06-19:
dmesg output after debugging lines changed.

```
[  645.681657] lego-port port0: Added new device 'in1:ev3-uart-host'
[  647.351272] mode 2 raw_min:00000000, raw_max:000003ff
[  647.384740] mode 2 si_min:00000000, si_max:00000001
[  647.451682] mode 2 - data_sets:3, data_type:6, figures:4, decimals:0
[  647.451739] raw_min: -4800, raw_max: 4800
[  647.451767] pct_min: 0, pct_max: 100
[  647.451788] si_min: -4800, si_max: 4800
[  647.568524] mode 1 raw_min:00000000, raw_max:000003ff
[  647.601978] mode 1 si_min:00000000, si_max:00000001
[  647.685357] mode 1 - data_sets:3, data_type:6, figures:6, decimals:3
[  647.685416] raw_min: -16000, raw_max: 16000
[  647.685446] pct_min: 0, pct_max: 100
[  647.685470] si_min: -7808, si_max: 7808
[  647.785626] mode 0 raw_min:00000000, raw_max:000003ff
[  647.819092] mode 0 si_min:00000000, si_max:00000001
[  647.916349] mode 0 - data_sets:3, data_type:6, figures:6, decimals:1
[  647.916408] raw_min: -20000, raw_max: 20000
[  647.916438] pct_min: 0, pct_max: 100
[  647.916461] si_min: -20000, si_max: 20000
[  647.946116] lego-sensor sensor1: Registered 'ev3-uart-76' on 'in1'.
```

bmegli, 2016-06-19:
@kortschak 

> With the above decl changes, I get

> ``` bash
> ERROR: "__aeabi_uldivmod" [drivers/lego/sensors/lego_sensor_class.ko] undefined!
> $ gcc --version
> gcc (Ubuntu 4.8.4-2ubuntu1~14.04.3) 4.8.4
> ```

Kernel code is not linked against libgcc. On kernel side you can use 64 bit math functions like those here:
https://github.com/ev3dev/ev3-kernel/blob/b23fdaca737d55757360027be39c9ae1fadb0b72/include/linux/math64.h calling them explicitly.

Or if it's perfomance critical you can try to get away with some binary shifts

@dlech used a mind shortcut there to just show the math

bmegli, 2016-06-19:
> Can you enable the debug prints here, here and here (replace debug_pr with printk)? Also the 4 debug prints starting here.

@dlech
I think the debug prints here, here and here like this:

```C
port->raw_min = *(u32 *)(message + 2);
				port->raw_max = *(u32 *)(message + 6);
				debug_pr("mode %d raw_min:%08x, raw_max:%08x\n",
				       mode, port->mode_info[mode].raw_min,
				       port->mode_info[mode].raw_max);
```
Will not work - they print from mode_info which wasn't assigned (default values) instead of from uart message.

But fortunately the forth @kortschak debug print worked:

``` bash
mode 1 - data_sets:3, data_type:6, figures:6, decimals:3
raw_min: -16000, raw_max: 16000
pct_min: 0, pct_max: 100
si_min: -7808, si_max: 7808
```

And this tells as the following:
- raw_min is correctly ftoi'ed because raws have 0 cecimals
- raw_max likewise
- si_min is float 16 with 3 decimals which overflows to -7808 instead of -16000
- si_max is float 16 with 3 decimals which overflows to 7808 instead of 16000

So thanks to @kortschak help the problem is identified now and moving the ftoi math to 64 bits is one way of fixing it.

dlech, 2016-06-19:
Please have a look at/test https://github.com/ev3dev/lego-linux-drivers/pull/46

kortschak, 2016-06-20:
Fixed. Thank you.

dlech, 2016-07-24:
Fix has been released in the `13-ev3dev` kernels.

closed 2016-07-24 by dlech