ev3dev/ev3dev #681: WiFi Not available

pambrose, 2016-06-18:
### System information:

* I am using: LEGO MINDSTORMS EV3

* I installed ev3dev using this image file: ev3-ev3dev-jessie-2015-12-30.img.zip

* My kernel version is: 4.4.13-12-ev3dev-ev3

* My host computer is running: OS X

* I am connecting using: Wi-Fi

### About my issue:

I upgraded to the newest distro this evening.  It worked fine on a BrickPi and an EVB.
It did not go so well on a couple EV3s.  After doing a *dist-upgrade* I rebooted and
Brickman shows Wi-Fi as "Not Available" and the dongle had no blue LED.  It failed 
with an Edimax dongle and a dongle from Dexter Industries -- both had previously worked. 
I then tried a NetGear N150 dongle and that worked fine.

bmegli, 2016-06-18:
Hi @pambrose and thanks for reporting

If it's any of those https://wiki.debian.org/rtl819x#supported-rtl8192cu, based on RTL8188CUS and RTL8192CU chips (e.g. Edimax EW 7811un) then it is already fixed by @dlech and waiting for new kernel release.

- If it's not any of those chipsets, could you give more details on the devices (this Edimax and Dexter)

- If it's one of those chipsets than the reason is that:
https://github.com/ev3dev/ev3dev/issues/669

For short - ev3dev changed the driver but the kernel module wasn't shipped with v4.4.13-12-ev3dev-ev3. You will have to wait for new kernel release to get those working.

bmegli, 2016-06-18:
The Dexter Industries Mini Wi-Fi Dongle for Raspberry Pi (SKU: 855947002552) is EDUP NANO Card - EP-N8508GS which uses RTL8188CUS chipset. 

If it is that one, it would be affected (information in post above) and it's fixed but waiting for next kernel release.

dlech, 2016-06-18:
Yes, duplicate of #669. It uses the same chipset.

pambrose, 2016-06-18:
Yes, my Edimax dongle is a EW-7811Un device.  One thing to note though: as descried in [#699](https://github.com/ev3dev/ev3dev/issues/669), wi-fi connectivity was intermittent in the previous release. In the new release, it is flat out not available -- ever.  Thus if one does not have a NetGear N150 dongle, you might want to hold off on the new dist-upgrade until @dlech works his magic.

bmegli, 2016-06-18:
@pambrose in the coming release, for EW-7811Un you may want to pass options to the driver (if they are not passed by ev3dev):

Create file:
``` bash
sudo nano /etc/modprobe.d/8192cu.conf
```

With those lines:

``` bash
# Disable power management
options 8192cu rtw_power_mgnt=0 rtw_enusbss=0
```

Otherwise EW-7811un is kind of "sleepy" .

tprk77, 2016-07-24:
Hi, I'm here to +1 this. 

WiFi is "Not available" after upgrading the kernel. The Edimax EW-7811Un is kind of special because it's the [only other WiFi dongle that works with the built-in firmware (almost)](http://thetechnicgear.com/2014/04/howto-configure-wifi-edimax-ew-7811un-mindstorms-ev3/). That's why I have this adapter, anyway. It would be nice if it was supported by `ev3dev`. I can send you one, if you want!

I've tried three other adapters, and seem to be having some kind of issue with all but one. And of course it's the largest one which doesn't really fit on my robot. Oh well.

tprk77, 2016-07-24:
By the way, one of the adapters is popping up this message when I try to connect:

```
GDBus.Error: net.connman.Error.OperationAboarted: Operation aborted
```

Not sure if that's interesting or not.

closed 2016-07-24 by dlech